esp8266/Arduino #6963: missing documentation about WiFi.config() for dns  servers

dsyleixa, 2019-12-29:
missing documentation about WiFi.config() for dns  servers:
which one is prefered? Which is depricated?
WiFi.config(local_IP, gateway, subnet, dns1, dns2)
or
WiFi.config(local_IP, gateway, dns, subnet)
where can a documentation be found?

earlephilhower, 2019-12-29:
The documents seem clear to me, I just went over them, as do the 2 versions of the API.

One sets up DNS.  One does not (and obviously so as no DNS is specified in the 4-param version).  You can run just fine w/o DNS as long as you don't do hostname resolution (lots of embedded intranet stuff).  OTW, you better specify a DNS server or expect host resolution failure.

closed 2019-12-29 by earlephilhower

dsyleixa, 2019-12-29:
of course it's for hostname resolution.
so WiFi.config(local_IP, gateway, dns, subnet) does NOT work???

earlephilhower, 2019-12-29:
What you called doesn't exist.  There is no `WiFi.config(ip,gw,dns,sub)`.  Please read the existing documents, it is explained clearly.  You either have NO DNS (3-param) or DNS1,DNS2 (optional) as the call.

https://github.com/esp8266/Arduino/blob/5d5200af412396dc0c6c716a3945f9063a0bafac/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h#L47

https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/station-class.rst

dsyleixa, 2019-12-29:
The Arduino API also allows 4-parameter call
WiFi.config(local_IP, dns, gateway, subnet)

and you said that it does not workv or esp?

jandrassy, 2019-12-30:
https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/station-class.html#config

dsyleixa, 2019-12-30:
ok, but then the esp API does not apply to the general Arduino API, that's a huge drawback and that should be urgently fixed !!!
https://www.arduino.cc/en/Reference/WiFiConfig

d-a-v, 2019-12-30:
We are compatible with arduino api.
Arduino api is not natural that's the reason why, probably, the other one exists.
Did you try something that does not work ?

jandrassy, 2019-12-30:
@dsyleixa, the `config` function is implemented to detect Arduino order of parameters  based on the net mask value position (net mask has always 255 as first byte in IP)

dsyleixa, 2019-12-30:
to the Arduino philosophy, cross-platform- and backwards-compatibility is inevitible.
Arduino provides a 4-param- Wifi.config:
WiFi.config(local_IP, dns, gateway, subnet)
https://www.arduino.cc/en/Reference/WiFiConfig
so th esp APIs should provide that, too!

jandrassy, 2019-12-30:
if something like
```
  IPAddress ip(192, 168, 1, 6);
  IPAddress dns(192, 168, 1, 1);
  IPAddress gw(192, 168, 1, 1);
  IPAddress sn(255, 255, 255, 0);
  WiFi.config(ip, dns, gw, sn);
```
doesn't work, then it is an issue

dsyleixa, 2019-12-30:
WiFi.config(ip, dns, gateway, subnet);  does not work
WiFi.config(ip, gateway, subnet, dns, dns);  does!

jandrassy, 2019-12-30:
so if the Arduino order really doesn't work, you can open a new issue with all formalities and refer this comment  from ESP8266WiFiSTA.h

> The argument order for ESP is not the same as for Arduino. However, there is compatibility code under the hood to detect Arduino arg order, and handle it correctly.