esp8266/Arduino #6964: Avast HNS crashes Nodemcu in SoftAP

KilianB, 2019-12-30:
The Avast [**Home Network Security (HNS)**]( https://blog.avast.com/2014/11/04/avast-2015-new-feature-home-network-security-scanning/) checks for router vulnerabilities which leads to a  crash of my Nodemcu running in SOFT_AP mode. When reading the issue description please keep in mind that I am not well versed in network related topics. 

Avast queries vulnerable files and protocols. e.g:

- request/ and /HNAP1/ on port 80
- perform a tcp port scan

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ESP8266WebServer.h>

ESP8266WebServer server(80);

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_AP);
  WiFi.softAP("Foo");
}

void loop() {
  server.handleClient();
  delay(50);
}
```

#### Wireshark 

[image]

While this looks like a port scan it bascially works as a ddos leading to tcp retransmissions down the line due to the server not being able to handle all requests

[image]

#### tcpdump -k NP

Resolving for process:
```
10:30:52.206343 (proc com.avast.hns:11459) IP 192.168.4.2.54501 > 192.168.4.1.domain: 55106+ A? wpad.local. (28)
10:30:52.206355 (proc com.avast.hns:11459) IP 192.168.4.2.51300 > 192.168.4.1.domain: 7271+ NS? _msdcs.local. (30)
```

### Stack trace

https://github.com/esp8266/Arduino/blob/5e4c2e9750a0b765270d9fc880539cb4018dfdf3/tools/sdk/lwip/src/app/dhcpserver.c#L1102

```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
Decoding 53 results
0x4023dea8: node_remove_from_list at ?? line ?
0x4022f00c: parse_msg at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-lwip/esp-dhcpserver.c line 626
:  (inlined by) handle_dhcp at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-lwip/esp-dhcpserver.c line 717
0x4022efcc: handle_dhcp at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-lwip/esp-dhcpserver.c line 679
0x401008cb: free at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_malloc.cpp line 362
0x40212aa4: udp_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/udp.c line 404
0x402178cc: ip4_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 1461
0x401008cb: free at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_malloc.cpp line 362
0x40229ab4: ppRecycleRxPkt at ?? line ?
0x4020e93d: ethernet_input_LWIP2 at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 188
0x4020e760: esp2glue_ethernet_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 469
0x4022e7ce: ethernet_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 365
0x4022e7df: ethernet_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 373
0x402296df: ppPeocessRxPktHdr at ?? line ?
0x402305c3: ets_snprintf at ?? line ?
0x40104781: call_user_start_local at ?? line ?
0x40104787: call_user_start_local at ?? line ?
0x4010000d: call_user_start at ?? line ?
0x402285c0: cont_ret at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/cont.S line 142
0x40228579: cont_continue at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/cont.S line 51
0x40100154: ets_post at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x401013ec: pp_post at ?? line ?
0x40104683: lmacTxFrame at ?? line ?
0x4010385e: lmacRecycleMPDU at ?? line ?
0x40103ccf: lmacRecycleMPDU at ?? line ?
0x40100640: umm_free_core at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_malloc.cpp line 316
0x40103797: lmacProcessTxSuccess at ?? line ?
0x40100154: ets_post at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40100154: ets_post at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40104683: lmacTxFrame at ?? line ?
0x401013ec: pp_post at ?? line ?
0x4010469b: lmacRxDone at ?? line ?
0x40101f87: rcReachRetryLimit at ?? line ?
0x40102164: rcReachRetryLimit at ?? line ?
0x40102626: wDev_ProcessFiq at ?? line ?
0x40102348: wDev_ProcessFiq at ?? line ?
0x4020634c: loop_wrapper() at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 180
0x4020634c: loop_wrapper() at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 180
0x40206266: esp_yield_within_cont at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 100
:  (inlined by) __yield at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 118
0x4020628b: optimistic_yield at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 128
0x40204c86: WiFiServer::available(unsigned char*) at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi/src/WiFiServer.cpp line 120
0x40105065: ets_timer_arm_new at ?? line ?
0x40206234: esp_yield_within_cont at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 100
:  (inlined by) __esp_yield at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 105
0x402067ee: __delay at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring.cpp line 54
0x4020312a: loop at /Users/Kilian/Documents/Arduino/APCrashTest/APCrashTest.ino line 14
0x4020634c: loop_wrapper() at /Users/Kilian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 180
```

I can't tell if this is simply a memory leak or an issue that will get fixed with https://github.com/esp8266/Arduino/issues/5380. Do we need to acknowledge requests we receive of ports we are not listening to (due to the tcp specs?) or can we implement a basic firewall to silently drop those packages? The devices crashes periodically within several minutes when the computer connects to it. It doesn't matter if the access point it open or uses wpa2.

#### Platform
- Core Version:			[latest release via board manager]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[NodeMCU 1.0 ESP12-E Module]
- Flash Size: 		[4MB]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]

d-a-v, 2020-03-05:
@KilianB Can you retry with latest [beta release](https://d-a-v.github.io) (v0.0.1) ?

d-a-v, 2022-06-11:
Closing as not updated for two years.

closed 2022-06-11 by d-a-v