esp8266/Arduino #6968: Load application address other than 0x1000

ymallik, 2019-12-31:
Hi Team,

Need a clarification on how to load the application other than 0x1000 address. By default, bootloader loads the application starting at 0x1000.

Now, we want to load the application at 0x2000. For this, we modified the bootloader code. We made below changes in flash.h file, rebuilt the bootloader code.

#define APP_START_OFFSET 0x2000    //0x1000

Now, we flashed the application code at 0x2000, but the application doesn't start.

Do we need to make any other changes in the bootloader code?

Best regards
Malli

devyte, 2019-12-31:
> flashed the application at 0x2000

how? why? our build system makes a unified binary that includes the bootloader at 0x0. If you flash the binary at 0x2000, it won't work.
I think you also have to change linker settings to link your modified bootloader at 0x0 and the app at 0x2000, and flash as usual at 0x0.

Closing because non compliant, but I'm interested in knowing the details if you manage this. Our own boot sector is almost full, but we might have to put more functionality into it, which means moving the app to 0x2000 if it all doesn't fit.

closed 2019-12-31 by devyte

ymallik, 2019-12-31:
Hi team,

Thank you for the information.

We want to do OTA in Arduino environment and we need to maintain both old and new firmware in the flash. Based on the configuration parameters, we will switch to either old or new firmware.

In examples, httpupdate, new firmware overwrites the old firmware.

Any ideas, how to implement this in arduino?

Best regards
Malli

devyte, 2019-12-31:
I think that has been discussed in #905

earlephilhower, 2019-12-31:
rboot is probably your best bet.  We don't use it, but it has support for app slots.

https://github.com/raburton/rboot