esp8266/Arduino #6971: HTTP_UPDATE_FAILD Error (-104): Wrong HTTP Code

vee73, 2020-01-01:
Hi All
I am using ESP8266  to test httpUpdate.  Tried both of these update methods
 //t_httpUpdate_return ret = ESPhttpUpdate.update(client, "http://rohini.ultimatefreehost.in/httpUpdate.bin");   -->Did not work

 t_httpUpdate_return ret = ESPhttpUpdate.update(client, "http://rohini.ultimatefreehost.in", 80, "httpUpdate.bin");  ->> getting HTTP_UPDATE_FAILD Error (-104): Wrong HTTP Code

Debug messages
[SETUP] WAIT 1...
[WIFI][APlistAdd] add SSID: mySSID
[HTTP-Client][sendRequest] type: 'GET' redirCount: 0
[hostByName] request IP for: http://rohini.ultimatefreehost.in
[hostByName] Host: http://rohini.ultimatefreehost.in IP: 199.59.242.153
:ref 1
[HTTP-Client] connected to http://rohini.ultimatefreehost.in:80
[HTTP-Client] sending request header
-----
GET httpUpdate.bin HTTP/1.0
Host: http://rohini.ultimatefreehost.in
User-Agent: ESP8266-http-Update
Connection: close
x-ESP8266-Chip-ID: 7223731
x-ESP8266-STA-MAC: 80:7D:3A:6E:39:B3
x-ESP8266-AP-MAC: 82:7D:3A:6E:39:B3
x-ESP8266-free-space: 1781760
x-ESP8266-sketch-size: 312832
x-ESP8266-sketch-md5: 54acdd6530aed0cfb5679d4daa83348c
x-ESP8266-chip-size: 4194304
x-ESP8266-sdk-version: 2.2.2-dev(38a443e)
x-ESP8266-mode: sketch
Content-Length: 0

-----
:wr 461 0
:wrc 461 461 0
:ack 461
:rn 319
:rcl
:abort
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 400 Bad Request'
[HTTP-Client][handleHeaderResponse] RX: 'Server: openresty'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Wed, 01 Jan 2020 06:14:18 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 170'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 400
[HTTP-Client][handleHeaderResponse] size: 170
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 400
[httpUpdate]  - len: 170
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 1781760
[httpUpdate]  - current Sketch Size: 312832
CALLBACK:  HTTP update fatal error code -104
[httpUpdate] HTTP Code is (400)
[HTTP-Client][end] still data in buffer (170), clean up.
:c0 1, 319
[HTTP-Client][end] tcp stop
HTTP_UPDATE_FAILD Error (-104): Wrong HTTP Code
:ur 1
:del

I have placed httpUpdate.bin under to root .

vee73, 2020-01-01:
HTTP Code is (400)  Indicate Bad Request..  
Anything wrong here   
    **t_httpUpdate_return ret = ESPhttpUpdate.update(client, "http://rohini.ultimatefreehost.in", 80, "httpUpdate.bin");**

----- code  --
  ESPhttpUpdate.setLedPin(LED_BUILTIN, LOW);

    // Add optional callback notifiers
    ESPhttpUpdate.onStart(update_started);
    ESPhttpUpdate.onEnd(update_finished);
    ESPhttpUpdate.onProgress(update_progress);
    ESPhttpUpdate.onError(update_error);

    //t_httpUpdate_return ret = ESPhttpUpdate.update(client, "http://rohini.ultimatefreehost.in/httpUpdate.bin");
    // Or:
    //t_httpUpdate_return ret = ESPhttpUpdate.update(client, "server", 80, "file.bin");
   // t_httpUpdate_return ret = ESPhttpUpdate.update(client, "http://rohini.ultimatefreehost.in", 80, "", "httpUpdate.bin");

micuat, 2020-03-05:
I had a same error with ESP32 and then I found that this code works
https://github.com/espressif/arduino-esp32/tree/master/libraries/Update/examples/AWS_S3_OTA_Update

(originally found in https://github.com/suculent/esp32-http-update )