esp8266/Arduino #6974: Difference in result from dir.fileName() between SPIFFS and LittleFS.

NicoZuid, 2020-01-02:
### Problem Description
Difference in result from dir.fileName() between SPIFFS and LittleFS.
dir.fileName() SPIFFS gives a file name with a / added in front.
For example: **/status.txt**
dir.fileName() LittleFS answers with a file name WITHOUT starting with a slash.
For example: **status.txt**

A source that must be suitable for both LittleFS and SPIFFS must take this into account.
Removing the / in LittleFS gives a file name **without the first letter.**
Adding an / in SPIFSS gives an **invalid URL**.

KilianB, 2020-01-02:
I am not sure if this is easily fixable due to the different nature of how Spiff and LittleFs handle directories. 

Additional information can be found here: https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html

**LittleFS**: Filenames are assumed to be in the root directory if no initial “/” is present.
**Spiff**: First, behind the scenes, SPIFFS does not support directories, it just stores a “flat” list of files. But contrary to traditional filesystems, the slash character '/' is allowed in filenames,

> Because there are directories, the openDir method behaves differently than SPIFFS. Whereas SPIFFS will return files in “subdirectories” when you traverse a Dir::next() (because they really aren’t subdirs but simply files with “/”s in their names), LittleFS will only return files in the specific subdirectory. This mimics the POSIX behavior for directory traversal most C programmers are used to.

NicoZuid, 2020-01-02:
Including the slash in **fileName()** is almost always unnecessary.
Only for deleting a file can that name with slash be used in the sketch.
Adjusting **SPIFFS** now will probably not be an option, I think.
Including the slight differences in the documentation is also an alternative.

earlephilhower, 2020-01-02:
As @KilianB said, for SPIFFS, "/" is just another character in the filename.  There are no directories.  So "happy.txt" and "/happy.txt" are different files.  There's no documentation that needs to be updated as far as I can see.  If your app does a `SPIFFS.open("happy.txt")`, that's perfectly fine.  IT just can't expect to open it at `SPIFFS.open("/happy.txt")`...

For your specific case, @NicoZuid, just don't blindly trash the first character.  Only do so if it's a path separator.

At this point, with SPIFFS being unsupported and showing some serious problems when heavily used/fragmented, I'd really suggest not even looking at starting a new app using it.

If you have a specific example that needs fixing that does the blind delete-first-char, please do open a bug and fill in the info so we can look at it.  But for this specific issue, there's no specific issue so I'll close it for now.

closed 2020-01-02 by earlephilhower