esp8266/Arduino #6975: OTA port not seen (2.6.3)

dagnall53, 2020-01-02:
-Sorry to open a subject that seems to have been solved previously.. But I have read the other issues threads and nothing suggested worked.
I am hoping someone can assist me to track down an issue that started only with an update from 2.5.2 to 2.6.3

In summary:
Some esp8266 Code I am working with worked fine with 2.5.2, but a recent update to 2.6.3 has stopped the OTA working. I cannot see the esp WiFi port in bonjour browser, or through ping or in a restarted Arduino. Rebooting the router has no effect.

If I upload the code using 2.5.2, it works fine, the OTA port appears in Arduino etc..
If I update to 2.6.2 the OTA port (in the esp programmed with 2.5.2) is still visible , (and I can upload using the OTA) but as soon as I do, the port disappears again when Arduino is reloaded.
With 2.6.3 I can use the ArduinoOTA example and it works - I have even added things like SPIFFs and ftp to the example and it still works.
The code I am working with is very complex and uses AP, Webserver, SPIFFs and OTA.
From two days worth of experiments I can say definitely that It is just the (very complex) code I am working with that has the incompatibility between 2.5.2 and 2.6.3 and also that I have not managed to make the ArduinoOTA example break by adding SPIFFs or ftp. (but I have not tried adding a webserver or AP. ) .. SO I am asking for help tracking down the incompatibility.. I have tried setting OTA debug via serial, and got these messages: To me it looks like the OTA should be working, but nothing sees the port..
Perhaps the debug messages may assist? Obviously there is OTA debug stuff interspersed with my own serial print comments..
They look VERY similar to me, but perhaps they may hint at the problem ..
Are there other debug settings that can give me more information?
Many thanks
Dagnall

(with non OTA working 2.6.3 )
SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af

scandone
Connecting to [linksys-25]
WiFi waiting to connect...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt
WiFi waiting to connect...
WiFi waiting to connect...

connected with linksys-25, channel 11
dhcp client start...
WiFi waiting to connect...
WiFi waiting to connect...
WiFi waiting to connect...
ip:192.168.0.75,mask:255.255.255.0,gw:192.168.0.1
WiFi connected RSSI=-35
-----Starting OTA------
My ip:192.168.0.75
My OTA name:WIO33
OTA server at: WIO33.local:8266
-----OTA started------
options 80
I2C device found at address 0x20
I2C device found at address 0x3C
I2C found 2 devices (4,5)
BMP280 status: 0x76=not found 0x77=not found
SSD1306 status: 0x3C=OK 0x3D=not found
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
AP IP address: 192.168.4.1
Connecting to [192.168.0.25]...
server connected
input status retry :33:0 1

With the working 2.5.2 version..
<SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020

scandone
Connecting to [linksys-25]
WiFi waiting to connect...
WiFi waiting to connect...

scandone
state: 0 -> 2 (b0)
WiFi waiting to connect...
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt
WiFi waiting to connect...
WiFi waiting to connect...

connected with linksys-25, channel 11
dhcp client start...
WiFi waiting to connect...
WiFi waiting to connect...
WiFi waiting to connect...
WiFi waiting to connect...
ip:192.168.0.75,mask:255.255.255.0,gw:192.168.0.1
WiFi connected RSSI=-36
-----Starting OTA------
My ip:192.168.0.75
My OTA name:WIO33
OTA server at: WIO33.local:8266
-----OTA started------
options 80
I2C device found at address 0x20
I2C device found at address 0x3C
I2C found 2 devices (4,5)
BMP280 status: 0x76=not found 0x77=not found
SSD1306 status: 0x3C=OK 0x3D=not found
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
AP IP address: 192.168.4.1
Connecting to [192.168.0.25]...
server connected
input status retry :33:0 1

d-a-v, 2020-01-02:
Do you call `MDNS.update()` in your main loop ?

dagnall53, 2020-01-02:
Yes, i tried that and it had no effect. 
There is a small web server in the code and i plan to switch that off today to see if that is part of the issue. 
I need to double check, but i think i could ping the ip address under 2.6.3, but could not ping using the device name.. That makes me think it is a mdns issue, but the update thing in the main loop did not work for me.

d-a-v, 2020-01-02:
To get reliable mDNS in my apps, I copied `hostProbeResult()` from [an example](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/examples/LEAmDNS/mDNS_ServiceMonitor/mDNS_ServiceMonitor.ino#L258).

jandrassy, 2020-01-02:
> To get reliable mDNS in my apps, I copied `hostProbeResult()` from [an example](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/examples/LEAmDNS/mDNS_ServiceMonitor/mDNS_ServiceMonitor.ino#L258).

doesn't ArduinoOTA library do this?

dagnall53, 2020-01-02:
I have added this,.. in 2.5.2 it works..  
In 2.6.3 it does not in my code. 
I have OTA debug switched on and there may be a hint about the error in that  with 2.5.2, after "MDNS responder started, I get the bcn0 etc debug lines, 
but with 2.6.3 these are missing, implying that the code does not  "get there".. 

Full text of serial monitor for 2.5.2 and 2.6.3 below.. 

Note: I can successfully ping the IP address directly, but cannot get a ping response using the expected names of WIO33 or esp8266

-------------------SERIAL OUTPUT using debug ..--------------------
2.5.2
RESULT*******************************************
SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020

options 80
I2C device found at address 0x20
I2C device found at address 0x3C
I2C found 2 devices (4,5)
BMP280 status: 0x76=not found 0x77=not found
SSD1306 status: 0x3C=OK 0x3D=not found
scandone
Connecting to [linksys-25]
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 10
cnt 

connected with linksys-25, channel 11
dhcp client start...
ip:192.168.0.75,mask:255.255.255.0,gw:192.168.0.1
WiFi connected RSSI=-32
MDNS responder started
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
AP IP address: 192.168.4.1
OTA server at: WIO33.local.local:8266
Connecting to [192.168.0.25]...
server connected
MDNSHostProbeResultCallback: Host domain 'esp8266.local' is free
setStationHostname: Station hostname is set to 'esp8266'
MDNSProbeResultCallback: Service query for 'http.tcp' services installed.
Answer ServiceDomain 4A13._http._tcp.local Modified
Answer HostDomainAndPort DVR-4A13.local:80 Modified
Answer TXT path=/index.html;swversion=20.8.1.RC9-VMS-2-C8A;platform=tcd/VM8685DVB;TSN=C8AC62001054A13
kv : TSN : C8AC62001054A13
kv : path : /index.html
kv : platform : tcd/VM8685DVB
kv : swversion : 20.8.1.RC9-VMS-2-C8A Modified
Answer IP4Address - 192.168.0.6 Modified
Answer ServiceDomain CEOL piccolo._http._tcp.local Modified
Answer HostDomainAndPort CEOL-piccolo.local:80 Modified
Answer IP4Address - 192.168.0.23 Modified
Answer ServiceDomain dLAN 500 AV Wireless+ (Shed Wireless)._http._tcp.local Modified
Answer HostDomainAndPort dLANwireless.local:80 Modified
Answer IP4Address - 192.168.0.38 Modified
Answer IP4Address - 169.254.248.132- 192.168.0.38 Modified
Answer TXT path=/
kv : path : / Modified
MDNSServiceProbeResultCallback: Service esp8266 probe succeeded.
Answer ServiceDomain Officejet 4500 G510n-z [E32B66]._http._tcp.local Modified
Answer HostDomainAndPort HPE32B66.local:80 Modified
Answer IP4Address - 192.168.0.15 Modified
*************************************************

-------2.6.3--------------
SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af

options 80
I2C device found at address 0x20
I2C device found at address 0x3C
I2C found 2 devices (4,5)
BMP280 status: 0x76=not found 0x77=not found
SSD1306 status: 0x3C=OK 0x3D=not found
scandone
Connecting to [linksys-25]
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 10
cnt 

connected with linksys-25, channel 11
dhcp client start...
ip:192.168.0.75,mask:255.255.255.0,gw:192.168.0.1
WiFi connected RSSI=-36
**  WIO hostname is <WIO33>  apname is <WIO33.local> 
MDNS responder started
AP IP address: 192.168.4.1
OTA server at: WIO33.local.local:8266
Connecting to [192.168.0.25]...
server connected
MDNSHostProbeResultCallback: Host domain 'esp8266.local' is free
setStationHostname: Station hostname is set to 'esp8266'
MDNSProbeResultCallback: Service query for 'http.tcp' services installed.
MDNSServiceProbeResultCallback: Service esp8266 probe succeeded.

d-a-v, 2020-01-02:
You are using both STA and AP interface.
A single mDNS instance was not properly working over several interfaces.
Now an independant mDNS instance needs to be started per interface.
When both AP and STA are up, the default interface (when not specifying it) is AP.
So your mDNS is not visible on STA side.
After STA is up (with its IP address), you need to start mDNS  with
`MDNS.begin(name, WiFi.localIP())`

If you need also mDNS on AP side, you need to start another instance.

d-a-v, 2020-01-02:
Please also do enable MDNS debug option

dagnall53, 2020-01-02:
Debug now set.. loads of output.. !
To help me understand where these Mdns instances are.. 
, .. I have an AP setup, 
`WiFi.softAP(ms_APName, AP_PWD);` 
then then an "OTA" setup, 
`ArduinoOTA.setHostname(ms_HostName);
    ArduinoOTA.setPassword(AP_PWD);
    ArduinoOTA.begin();`
Question : 
Where should I place the `MDNS.begin(name, WiFi.localIP())`

(I will try some stuff and post the debug if that helps) 
D

dagnall53, 2020-01-02:
Debug text attached.. before I try to add the MDNS.begin(name, WiFi.localIP())

Could one problem be that both the AP and OTA have similar names? 

[MDNSDebug.txt](https://github.com/esp8266/Arduino/files/4016470/MDNSDebug.txt)

dagnall53, 2020-01-02:
Feeling very ignorant.. 
How do I (simply) set up  separate MDNS instances.. so that  
`MDNS.begin(name, WiFi.localIP())`
will work. ? 

Added to my code by its-self it gives `"'MDNS' was not declared in this scope"..`
(I have taken out the " hostProbeResult() " stuff mentioned earlier)

cheers and thanks
d

rocrail, 2020-01-02:
Did you include the library?
`#include <ESP8266mDNS.h>        // Include the mDNS library
`

d-a-v, 2020-01-02:
if you are using OTA and not specifically mDNS, it's in the OTA library that you'll need to modify to test if that's the issue: https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/ArduinoOTA.cpp#L138

If it works, something has to be fixed in the API.

jandrassy, 2020-01-02:
ArduinoOTA.begin() calls MDNS.begin() with provided hostname. try to call  ArduinoOTA.begin(false);. then you can setup MDNS in sketch

rocrail, 2020-01-02:
Since it was no issue with version 2.5.2, the 2.6.3 is in this context IMHO not an improvement.
If the STA and AP cannot be used both at the same time, or need special code, it would be helpful if this is somewhere documented in the release notes of the board library.

d-a-v, 2020-01-02:
> Since it was no issue with version 2.5.2, the 2.6.3 is in this context IMHO not an improvement.

Before discussing of what is an improvement and what is not, please tell us if your issue is fixed by specifically starting mDNS on STA interface by one of the two ways described above.

Then we will have to think of the best improvement path.

dagnall53, 2020-01-02:
Many thanks all. 

- I had omitted to (re) include the `#include <ESP8266mDNS.h>`
- I then added `MDNS.begin("APname", WiFi.localIP());` before OTA started 

I now see the board as a network port "APname" in the Arduino ports list. !!

(jandrassy) Just starting OTA with `ArduinoOTA.begin(false);` (and without the `MDNS.begin("APname", WiFi.localIP());` )   had no effect.. I could not see the port. It is only the explicit MDNS begin that actually seems to work. 

 So to me it looks as though the OTA library now affects any previous MDNS setup? I think rocrail is correct and something in the API has been broken between 2.5.2 and 2.6.3 ..

Dagnall

dagnall53, 2020-01-02:
Many thanks all for the help.. 
I would like to modify the ArduinoOTA example to demonstrate (simply) the bug, but I am running out of talent!.  
I think that "all" you need to do is add an AP /webserver function before the OTA is begun.. 

But writing a simple version of ArduinoOTA that demonstrates this (and triggers the bug) is not something I can manage now.. Perhaps someone else with better skills could modify the example to demonstrate the bug? 
D

d-a-v, 2020-01-02:
Running the example the MDNS debug will display on which interface mDNS is started,
I was thinking that would be sufficient to give hints.
It is however true mDNS is very verbose in debug mode.

This issue is marked for release 2.7.0 (or will be pushed to 3.0.0) so we will eventually come back to it and see how we should update the API, the documentation, or both.

On top of that, a new version of LEA mDNS will come up anytime, updated by its original author @LaborEtArs (with a similar interface-aware API update - because mDNS *cannot* work on more than one interface at the same time, and this interface-aware API change is a fix regarding previous versions).

jandrassy, 2020-01-02:
> (jandrassy) Just starting OTA with `ArduinoOTA.begin(false);` (and without the `MDNS.begin("APname", WiFi.localIP());` ) had no effect.. I could not see the port. It is only the explicit MDNS begin that actually seems to work.

you didn't read carefully.`begin(false)` only disables starting of MDNS by the ArduinoOTA library. because if SoftAP is on it starts MDNS on SoftAP interface (and not on STA (this the issue here)). so add that false, if you don't want waste resources for MDNS on SoftAP.

glyndon, 2020-01-09:
FWIW, in case it's a help:
I'm having the same prob here as the OP with my code which includes a few servers.
Trial-and-error elimination led me to find that OTA works again if I do not poll 
WiFi.status() == WL_CONNECTED 
in the loop.
I had been doing that call so I could light a LED to indicate loss of WiFi, should it occur.
Commenting that call out seems to have enabled OTA for me in this case.

tbdltee, 2020-01-28:
I also have the same problem. Arduino IDE cannot see OTA port when use esp8266 in WIFI_AP_STA. I try with BasicOTA.no. WIFI_STA mode has no problem.
I try both solution, ie.  MDNS.begin()+ArduinoOTA.begin(false);  and use as @glyndon mentioned. Both don't work for me.

At the end, I goto ESP8266mDNS.h and use legacy mDNS instead of new one. Everything works like charm.

Wait for the permanent fixed in next release.

d-a-v, 2020-01-28:
> Arduino IDE cannot see OTA port when use esp8266 in WIFI_AP_STA. I try with BasicOTA.no. WIFI_STA mode has no problem.

That is expected.
mDNS currently works only on AP *by default* when both AP and STA are enabled.
Use `MDNS.begin("name", WiFi.localIP())` to always make it work on STA even if AP is enabled.
@devyte should I change this behaviour now ?

tbdltee, 2020-01-28:
> That is expected.
> mDNS currently works only on AP _by default_ when both AP and STA are enabled.
> Use `MDNS.begin("name", WiFi.localIP())` to always make it work on STA even if AP is enabled.
> @devyte [...]

i'm using STA_AP mode. AP side is using espnow while STA side is used to connect to the router. With my specific setup, using MDNS.begin("name", WiFi.localIP()) after connected to the router, won't solve the problem. OTA port still disappear.
i suggest to change mDNS to work on STA by default rather than AP or do you have any reason why mDNS is on AP side by default on WIFI_AP_STA?

devyte, 2020-01-28:
@d-a-v users keep getting tripped by this. Let's change it now and announce it as breaking for 2.7.0.

tbdltee, 2020-02-04:
> @d-a-v users keep getting tripped by this. Let's change it now and announce it as breaking for 2.7.0.

back to confirm that with lastast git version, problem resolved.

PR #7042 (2020-02-05): BREAKING: mDNS: use STA as default interface
*Edit by @devyte:*
The old behavior is as follows:
-In STA mode, MDNS works on the STA interface
-In AP mode, MDNS works on the AP interface
-In AP+STA mode, MDNS works on the AP interface and not on the STA interface

This sounds reasonable, but it causes all sorts of confusion (and opened issues and comments) when switching WiFi mode on the fly. In particular, when in STA mode, the responder works when accessing from another device on the network, but when switching to AP+STA mode, the ESP suddenly disappears, because MDNS switches to the AP inter [... cut]
closed 2020-02-05 by d-a-v