esp8266/Arduino #6981: Timestamp SDFS does not work correctly with 2GB SD card.

NicoZuid, 2020-01-04:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.6.3]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Wemos D1 r2 or  mini]
- Flash Mode:			[?]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[?}
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200)

### Problem Description

Timestamp SDFS does not work correctly with 2GB SD card.
(I currently have no hardware available for larger SD cards)

dir.next also gives the **"Volume information entry"** 
Timestamp not set:
dir.fileTime () returns 1970-01-01 00:00:00 **always**
file.getLastWrite () returns 2000-01-01 1:00:00 **or correct time if set.**

fsInfo.pageSize **returns 0 (zero)**

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

#include <FS.h>
#include <SDFS.h>
#include <time.h>
#include <ESP8266WiFi.h>

#ifndef STASSID
#define STASSID "groen4ZUID"
#define STAPSK  "pasword"
#endif

const char *ssid = STASSID;
const char *pass = STAPSK;

long timezone = 1;
byte daysavetime = 0;

bool getLocalTime(struct tm * info, uint32_t ms) {
  uint32_t count = ms / 10;
  time_t now;

  time(&now);
  localtime_r(&now, info);

  if (info->tm_year > (2016 - 1900)) {
[... 234 lines cut ...]
Now mount it
Reading file: /hello.txt
Read from file: Hello World!
Listing directory: /
  FILE: System Volume Information  SIZE: 0  SIZE: 0  LAST WRITE: 2020-01-03 18:45:00
  FILE: System Volume Information  SIZE: 0  SIZE: 0  FILE TIME: 1970-01-01 00:00:00
  FILE: hello.txt  SIZE: 13  SIZE: 13  LAST WRITE: 2000-01-01 01:00:00
  FILE: hello.txt  SIZE: 13  SIZE: 13  FILE TIME: 1970-01-01 00:00:00
  FILE: Windows.txt  SIZE: 13  SIZE: 13  LAST WRITE: 2020-01-04 06:29:58
  FILE: Windows.txt  SIZE: 13  SIZE: 13  FILE TIME: 1970-01-01 00:00:00
Wait a momentReading file: /hello.txt
Read from file: Hello World!
Listing directory: /
  FILE: System Volume Information  SIZE: 0  SIZE: 0  LAST WRITE: 2020-01-03 18:45:00
  FILE: System Volume Information  SIZE: 0  SIZE: 0  FILE TIME: 1970-01-01 00:00:00
  FILE: hello.txt  SIZE: 13  SIZE: 13  LAST WRITE: 2000-01-01 01:00:00
  FILE: hello.txt  SIZE: 13  SIZE: 13  FILE TIME: 1970-01-01 00:00:00
  FILE: Windows.txt  SIZE: 13  SIZE: 13  LAST WRITE: 2020-01-04 06:29:58
  FILE: Windows.txt  SIZE: 13  SIZE: 13  FILE TIME: 1970-01-01 00:00:00
FS Bytes used: 160.00 KB
FS Bytes total: 1.84 GB
FS block size: 32768
FS page size: 0

```

earlephilhower, 2020-01-04:
Thanks for the detailed report.  It's funny, I can't even find new uSD cards under 4GB (but have used a 128*MB* one leftover from a Palm PDA for testing).

The only difference between 2GB and larger cards would be the FAT, not the directory/LFN stuff, so I think that's immaterial here.

The "System Volume Information" really is on the card, it's not making it up.  Looks like the card was used on a Windows PC at one point.  SDFS isn't making that up, so I don't see a problem there.  The attribute should be HIDDEN,SYSTEM, but there's no concept of attributes in the Arduino `File` so it's kind of a black box.

fs page size is 0 because it doesn't really make any sense.  There are no pages on a SD card, it has 512b accessible sectors.  Are you suggesting it should be set to the block size (FAT allocation size)?

File access time, I'll have to check.  I don't think the SD would want to update it (wear out the flash on repeated reads causing multiple directory rewrites), but I'll take a look.

earlephilhower, 2020-01-04:
Well, pretty obvious.  I forgot to provide the Dir->fileTime callback.  I was even reading out the file time and caching it in the Dir structure...just never returning it.  PR #6985 should fix this.

PR #6985 (2020-01-04): Add SDFS::Dir::fileTime callback
Forgot to add a Dir->fileTime override, resulting in it always returning
(time_t)0, or Jan 1, 1970.

Add the override, returning the proper lastWriteTime.

Fixes #6981
commit d4d89244 (2020-01-04): Add SDFS::Dir::fileTime callback (#6985)

Forgot to add a Dir->fileTime override, resulting in it always returning
(time_t)0, or Jan 1, 1970.

Add the override, returning the proper lastWriteTime.

Fixes #6981
closed 2020-01-04 by earlephilhower