esp8266/Arduino #6983: Logic operators no precedence over boolean

tedani, 2020-01-04:
- Hardware:					[ESP-12]
- Core Version:			[2.6.3]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

Unlike arithmetic operators, logic operators (&, |) in conditional clause take no precedence over boolean (==, !=).  The following example code explains it all.

#include <Arduino.h>

void setup(void)
{
    uint8_t i, j, x, y;

    Serial.begin(115200);
    i = 3;
    j = 1;
    x = 2;
    y = 2;
    Serial.print("(i + j == x + y) is ");
    if (i + j == x + y)
        Serial.println("true");
    else
        Serial.println("false");

    Serial.print("((i + j) == (x + y)) is ");
    if ((i + j) == (x + y))
        Serial.println("true but tedious.");
    else
        Serial.println("false and stupid.");

    i = 0x04;
    j = 0x07;
    x = 0x04;
    y = 0x0f;
    Serial.print("(i & j == x & y) is ");
    if (i & j == x & y)
        Serial.println("clever.");
    else
        Serial.println("stupid.");

    Serial.print("(i & j) == (x & y) is ");
    if ((i & j) == (x & y))
        Serial.println("not clever.");
    else
        Serial.println("utter idiot.");
}

void loop(void)
{
}

### Debug Messages
The output gives the following results:
(i + j == x + y) is true
((i + j) == (x + y)) is true but tedious.
(i & j == x & y) is stupid.
(i & j) == (x & y) is not clever.

liebman, 2020-01-04:
This is not a bug.  Please read this on operator precedence: https://en.cppreference.com/w/c/language/operator_precedence

closed 2020-01-04 by earlephilhower

tedani, 2020-01-05:
> This is not a bug. Please read this on operator precedence: https://en.cppreference.com/w/c/language/operator_precedence Note that `==` has a higher precedence than `&`.

You're wrong, "&" is precendence 2 while "==" is 7.  "+" and "&" are in the same precedence, but "+" works and "&" not.  Please take a look at my example program again:

(i + j == x + y) --> result both side are equal
(i & j == x & y) --> result both side are not equal

liebman, 2020-01-05:
The `&` in 2 is `address of`, look at 8 `bitwise and`.