esp8266/Arduino #6984: WebServer.serveStatic: index.html.gz as default doesn't work (while index.html does)

JasperHorn, 2020-01-04:
Using `index.html` instead of `index.htm` for the "folder" is supported since https://github.com/esp8266/Arduino/commit/9f2cfb82184176a0930303be05791d0aaac1e918

At that time, the lookup order became:

1. index.htm
2. index.html
3. index.html.gz

As https://github.com/esp8266/Arduino/issues/6792 pointed out, this broke support for `index.htm.gz` and this was changed in https://github.com/esp8266/Arduino/commit/6768116b8c738e3fdee5b80d23f8cdc33241bbc3 . The lookup order now is:

1. index.htm
2. index.html
3. index.htm.gz

However, this has removed support for using `index.html.gz` completely. I think the lookup order should include four items:

1. index.htm
2. index.html
3. index.htm.gz
4. index.html.gz

(Perhaps with 2 & 3 reversed.)

devyte, 2020-01-27:
2 and 3 should be reversed.
Can you please make a PR with your suggestion?

earlephilhower, 2020-02-09:
>2 and 3 should be reversed

I would humbly suggest that anyone evil enough to have both a `index.htm.gz` and a `index.html` and expect for a sane solution gets the randomness they deserve.

closed 2020-02-09 by devyte

JasperHorn, 2020-02-09:
Thanks for picking this up during my radio silence. I was still planning to get back to this, but had since moved on to non-esp projects and didn't get around to this.

Now there's just my other (accepted) pull request which is in need of a follow-up...

PR #7069 (2020-02-09): Fix serach order for index.htm(l)(.gz) files
Fixes #6984

When a directory index is requested with an explicit index.html, follow
the original webserver order and check for: index.htm, index.htm.gz,
index.html, index.html.gz, in order.

Fixes the regressions introduced in 9f2cfb8 and 6768116
commit dbd7b821 (2020-02-09): Fix serach order for index.htm(l)(.gz) files (#7069)

Fixes #6984

When a directory index is requested with an explicit index.html, follow
the original webserver order and check for: index.htm, index.htm.gz,
index.html, index.html.gz, in order.

Fixes the regressions introduced in 9f2cfb8 and 6768116