esp8266/Arduino #6990: IPv6 is not supported by mDNS?

xunni1000, 2020-01-06:


devyte, 2020-01-06:
Not yet.

closed 2020-01-06 by devyte

CRCinAU, 2020-03-07:
I wonder how far this is off - seeing as IPv6 is supported on ESP8266 now... I can connect via PubSubClient to an IPv6 MQTT server - as long as you call it by a hostname - not an IP address.

Compile errors when enabling IPv6 suport in LEAmDNS.h show:
```
*** [.pio/build/d1_mini/libbea/ESP8266mDNS/ESP8266mDNS_Legacy.cpp.o] Error 1
In file included from ~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266mDNS/src/LEAmDNS_Priv.h:179:0,
                 from ~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266mDNS/src/LEAmDNS_Helpers.cpp:28:
~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266mDNS/src/LEAmDNS.h:1052:17: error: 'IP6Address' does not name a type
                 IP6Address      m_IPAddress;
                 ^
~.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266mDNS/src/LEAmDNS_Helpers.cpp: In member function 'bool esp8266::MDNSImplementation::MDNSResponder::_allocUDPContext()':
~/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/ESP8266mDNS/src/LEAmDNS_Helpers.cpp:222:20: error: 'ip_addr_t' has no member named 'addr'
     multicast_addr.addr = DNS_MQUERY_IPV6_GROUP_INIT;
                    ^
*** [.pio/build/d1_mini/libbea/ESP8266mDNS/LEAmDNS_Helpers.cpp.o] Error 1

```

myfreeweb, 2020-06-24:
Looks like work on this is happening in #7335

d-a-v, 2020-06-25:
Please head to https://github.com/esp8266/Arduino/issues/7210#issuecomment-647149448