esp8266/Arduino #6992: SD file system, request to add file.CreationDate ().

NicoZuid, 2020-01-06:
In version 2.5.0 was a directory of utilities included.
A few years ago I found an example on the internet that made it possible to make the creation date of an SD file visible.
In version 2.6.3, Arduino SdFat Library by William Greiman is still used.  (2009)

**..\2.6.3\librarie\ESP8266\SdFat\src\Fatlib\FatFile.cpp**

This file contains code to process the SD file creation date / time.
Is it possible to add this to the options of the SDFS library?

earlephilhower, 2020-01-06:
I tried to mirror the ESP32's `File` interface when I rejiggered things.  Do they already have a call like that?  It's almost trivial to add (requires a minor change to mklittlefs to add a new attribute, but it's binary backwards compatible w/existing FSes), but I want to keep things as closely tied as possible.

One option in the meantime is to use the SdFat library directly and bypass the Arduino core File stuff altogether.  I haven't tried it, but no changes were made to the SDFat lib from Greiman except for compatibility changes to the low-level interfaces in my fork.

NicoZuid, 2020-01-06:
My knowledge of the C ++ programming language is too limited to come up with a solution myself.

I have now added the code from my program which shows the date.
Unfortunately I am unable to get version 2.5.0 up and running again at the moment.
It seems that the IDE version change to 1.8.10 makes it impossible to go back to the old version.

NicoZuid, 2020-01-07:
Just probably found the solution in the sdFat examples of the ESP8266SdFat library.
In the example "Timestamp" the following code is used:
**f.printFatTime (d.creationDate)**; / **f.printFatTime (d.creationTime);**
and also
f.printFatDate (d.lastAccessDate);
I can continue again.
mvg Nico

earlephilhower, 2020-01-07:
Yes, the actual addition isn't a big deal.  We already have enough info to get it with the existing code.  I checked the ESP32 and they don't have such a call, but that's not to say we can't.  Let me see if I can throw something together tonight.

earlephilhower, 2020-01-08:
@NicoZuid, can you give PR #7000 (!) a try on your SD setup and report back?  Mine is presently buried somewhere on my desk so all I was able to verify personally was the LittleFS changes.

NicoZuid, 2020-01-08:
I'm going to see if I succeed.

NicoZuid, 2020-01-08:
Unfortunately this would be my first time that I would have to replace a cpp file in a library with a test version.
As I have done now I get compile errors.
Is there somewhere described the correct way to include the test version of SDFS.cpp in my project?

earlephilhower, 2020-01-08:
https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/checking-out-pull-requests-locally

NicoZuid, 2020-01-08:
Instruction is not yet in line with what I see on my screen.
Do I have sufficient rights to do this?

NicoZuid, 2020-01-09:
I manually transferred the source codes of 7 files to my PC via PR # 7000.
I get the (old) creationDateTime printed correctly (SDFS adapted example LittleFS)
Unfortunately not the changes to # PR6985.
So no NEW dates (from today) are written on the SD.
I cannot judge whether that is due to the way the code was transferred to my PC

### Debug Messages

```

Connecting to groen4ZUID
...............WiFi connected
IP address: 
192.168.2.4
Contacting Time Server

Now is : 2020-01-09 20:06:43

Mount SDFS
Listing directory: /
  FILE: System Volume Information  SIZE: 0    CREATION: 2020-01-03 18:44:58
  LAST WRITE: 2020-01-03 18:45:00
  FILE: hello.txt  SIZE: 61    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: temptab.txt  SIZE: 317    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: styl1.css  SIZE: 1023    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: FSBrowser.ino  SIZE: 8564    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: Windows.txt  SIZE: 13    CREATION: 2020-01-04 06:29:18
  LAST WRITE: 2020-01-04 06:29:58
  FILE: temp_grafiek.html  SIZE: 7218    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: example.txt  SIZE: 0    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: datalog.txt  SIZE: 22180    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: default.txt  SIZE: 0    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: callback.txt  SIZE: 1    CREATION: 2014-10-01 20:30:40
  LAST WRITE: 2014-10-02 20:30:42
  FILE: stamp.txt  SIZE: 0    CREATION: 2014-11-10 01:02:02
  LAST WRITE: 2014-11-11 04:05:06
Deleting file: /hello.txt
File deleted
Writing file: /hello.txt
File written
Appending to file: /hello.txt
Message appended
Listing directory: /
  FILE: System Volume Information  SIZE: 0    CREATION: 2020-01-03 18:44:58
  LAST WRITE: 2020-01-03 18:45:00
  FILE: hello.txt  SIZE: 13    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: temptab.txt  SIZE: 317    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: styl1.css  SIZE: 1023    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: FSBrowser.ino  SIZE: 8564    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: Windows.txt  SIZE: 13    CREATION: 2020-01-04 06:29:18
  LAST WRITE: 2020-01-04 06:29:58
  FILE: temp_grafiek.html  SIZE: 7218    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: example.txt  SIZE: 0    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: datalog.txt  SIZE: 22180    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: default.txt  SIZE: 0    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: callback.txt  SIZE: 1    CREATION: 2014-10-01 20:30:40
  LAST WRITE: 2014-10-02 20:30:42
  FILE: stamp.txt  SIZE: 0    CREATION: 2014-11-10 01:02:02
  LAST WRITE: 2014-11-11 04:05:06
The timestamp should be valid above
Now unmount and remount and perform the same operation.
Timestamp should be valid, data should be good.
Now mount it
Reading file: /hello.txt
Read from file: Hello World!
Listing directory: /
  FILE: System Volume Information  SIZE: 0    CREATION: 2020-01-03 18:44:58
  LAST WRITE: 2020-01-03 18:45:00
  FILE: hello.txt  SIZE: 13    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: temptab.txt  SIZE: 317    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: styl1.css  SIZE: 1023    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: FSBrowser.ino  SIZE: 8564    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: Windows.txt  SIZE: 13    CREATION: 2020-01-04 06:29:18
  LAST WRITE: 2020-01-04 06:29:58
  FILE: temp_grafiek.html  SIZE: 7218    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: example.txt  SIZE: 0    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: datalog.txt  SIZE: 22180    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: default.txt  SIZE: 0    CREATION: 2000-01-01 01:00:00
  LAST WRITE: 2000-01-01 01:00:00
  FILE: callback.txt  SIZE: 1    CREATION: 2014-10-01 20:30:40
  LAST WRITE: 2014-10-02 20:30:42
  FILE: stamp.txt  SIZE: 0    CREATION: 2014-11-10 01:02:02
  LAST WRITE: 2014-11-11 04:05:06

```

earlephilhower, 2020-01-11:
I tried on my own system and SD create/write stamps are looking good and matching what my Linux box reports for the same SD card, so I think we're good.

NicoZuid, 2020-01-11:
Do you mean that the dates written on the sd via your Linux computer are also visible via the arduino?

In my opinion, the problem is not to show the written date stamps.
New creation and/or write date stamps via SDFS are **not correct.**

WRONG:
FILE: hello.txt SIZE: 13 CREATION: **2000-01-01 1:00:00**
  LAST WRITE: **2000-01-01 1:00:00**
CORRECT:
 FILE: Windows.txt SIZE: 13 CREATION: **2020-01-04 06:29:18**
  LAST WRITE: **2020-01-04 06:29:58**

(LittleFS with the same program gives correct date/time stamps)

earlephilhower, 2020-01-12:
I took my 128MB card with old photos and ran the modified listfiles example.  Compared the create/write times to the ones returned from `stat` on my machine and they matched.

NicoZuid, 2020-01-12:
Sorry but I think a modified list files example does NOT change a creation date or lastWrite date.

I used the example LittleFS_Timestamp and changed all code LittleFS to SDFS.
This example creates and changes the hello.txt file.
Date and time stamps  of **hello,txt** remain unchanged 2000-01-01 01:00:00

**Don't forget to remove SDFS.format ();**

earlephilhower, 2020-01-19:
@NicoZuid , I just pushed an update to #7000 which I was able to test.  I was setting a callback for date/time, but never telling the `SdFat` library to use it, so create/write times never got updated in the device.

Just make sure that NTP is configured and running before you do file ops (same as other filesystems, of course!) and then you should get the proper dates on all files made on the device, now.

NicoZuid, 2020-01-20:
@earlephilhower , This morning (CET 7:25 AM) I tested the new version, I can confirm that it now works correctly.
```cpp
_Listing directory: /
  FILE: hello.txt  SIZE: 13    CREATION: 2020-01-20 07:27:24
  LAST WRITE: 2020-01-20 07:27:28_
```
I always use TimeLib.h in my programs (instead of time.h)
TimeLib.h makes it easier, for example, to check whether NTP has started correctly.
```cpp
while (year(time(NULL)) < 2020) {
    delay(500);
    Serial.print(".");
  }
```

PR #7000 (2020-02-22): Add fileCreation/getCreation create-time accessors
For SDFS and LittleFS, enable a creation time accessor for files and Dir
iterators, similar to the existing fileTime/getLastWrite calls.

Remove spurious Dir::getLastWrite method (the proper and only documented
way is really Dir::fileTime).

Update json to point to new mklittlefs which copies the creation date of
files to the image.

Fixes #6992
commit bea64dfa (2020-02-22): Add fileCreation/getCreation create-time accessors (#7000)

* Add fileCreation/getCreation create-time accessors

For SDFS and LittleFS, enable a creation time accessor for files and Dir
iterators, similar to the existing fileTime/getLastWrite calls.

Remove spurious Dir::getLastWrite method (the proper and only documented
way is really Dir::fileTime).

Update json to point to new mklittlefs which copies the creation date of
files to the image.

Fixes #6992

* Remove malloc(), use stack vars for temp names

LFS filenames are limited in size and generally very small.  Use a stack
variable inste [... cut]
closed 2020-02-22 by earlephilhower