esp8266/Arduino #6995: SPIFFSImpl::open: fd=-10001 path=`/timer.txt` openMode=5 accessMode=2 err=-10001

rybertm, 2020-01-07:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.6.3]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:		[dio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[??? (probably nodemcu)]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I have some static files in the filesystem and I'm writing to SPIFFS once every 3 seconds some data (around 29 bytes), after some random amount of time, usually when the file has exceeded 20Kb
	but no fixed size (can happen in 25Kb, 50Kb, 200Kb...), ```print``` returns bytes written but the filesize doesn't usually change, sometimes it changes randomly from its "real size", and other times it
	make multiple duplicates of said file corrupting other files, specifically a small text file (around 1 bytes), changing its size to 0.
	
Worth noting that it never exceeds the filesystem size limit (1Mb) except when it makes duplicates.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
String measurements= "";
char[20] date;
size_t bytesWritten = 0;
...

void getMeasurements()
{
  ...

  File tmplog = SPIFFS.open("/measurements.csv", "a");
  bytesWritten += tmplog.print(measurements);
  bytesWritten += tmplog.print(',');
  bytesWritten += tmplog.println(date); // date is NEVER NULL
  tmplog.close();

  showFiles();  // List files in SPIFFS and set bytesWritten to 0
}

void loop() {
...

if (sec % freq == 0) // Every 3 seconds
      getMeasurements(); // Write data to SPIFFS

// Write seconds passed since it first started running to SPIFFS so it can reset counter after 24h
    File file = SPIFFS.open("/timer.txt", "w"); 
    file.print(++sec);
    file.close();
}

```

### Debug Messages

**Output before error:**

```
07/01/20 11:56:39 #####################
07/01/20 11:56:39 
07/01/20 11:56:39 SPIFFS directory {/} :

07/01/20 11:56:39 	/timer.txt	size: 0
07/01/20 11:56:39 	/measurements.csv	size: 681007
07/01/20 11:56:39 	/historico.html.gz	size: 7337
07/01/20 11:56:39 	/logo-header.png	size: 24178
07/01/20 11:56:39 	/medidores.html.gz	size: 7365
07/01/20 11:56:39 Bytes written to /medidas.csv: 29
07/01/20 11:56:39 #####################
07/01/20 11:56:39 
07/01/20 11:56:41 SPIFFS_close: fd=1

07/01/20 11:56:41 SPIFFS_close: fd=1

07/01/20 11:56:44 SPIFFS_close: fd=1

07/01/20 11:56:44 SPIFFS_close: fd=1

07/01/20 11:56:47 SPIFFS_close: fd=1

07/01/20 11:56:47 SPIFFS_close: fd=1

07/01/20 11:56:48 SPIFFS_close: fd=2

07/01/20 11:56:48 SPIFFS_close: fd=2
```

**Output after error:**

```
07/01/20 11:56:48 #####################
07/01/20 11:56:48 
07/01/20 11:56:48 SPIFFS directory {/} :

07/01/20 11:56:48 	/logo-header.png	size: 24178
07/01/20 11:56:48 	/timer.txt	size: 5
07/01/20 11:56:48 	/historico.html.gz	size: 7337
07/01/20 11:56:48 	/measurements.csv	size: 681036
07/01/20 11:56:48 	/medidores.html.gz	size: 7365
07/01/20 11:56:48 Bytes written to /medidas.csv: 29
07/01/20 11:56:48 #####################
07/01/20 11:56:48 
07/01/20 11:56:51 SPIFFS_close: fd=1

07/01/20 11:56:51 SPIFFS_close: fd=1

07/01/20 11:56:52 SPIFFSImpl::open: fd=-10001 path=`/timer.txt` openMode=5 accessMode=2 err=-10001

07/01/20 11:56:53 SPIFFSImpl::open: fd=-10001 path=`/timer.txt` openMode=5 accessMode=2 err=-10001

07/01/20 11:56:54 SPIFFS_close: fd=2

07/01/20 11:56:54 SPIFFS_close: fd=2
```

earlephilhower, 2020-01-07:
SPIFFS is not supported by the author anymore and has some strange behavior when a FS gets fragmented from lots of small writes.  You could try adding SPIFFS.gc() calls as part of a periodic process, but there are just some cases when it gets hung and does odd stuff like this.

If your app is doing lots of appends to a file, I would suggest going w/LittleFS.  It's actively being developed by the ARMbed guys and seems to be much more stable.  It does, however, have not so good performance when you do updates to the beginning of a large file, so there are tradeoffs.  In other cases, though, it is significantly faster and more robust.

rybertm, 2020-01-07:
@earlephilhower thanks for the input, I will definitely try LittleFS, thank you very much.

rybertm, 2020-02-10:
Ever since @earlephilhower suggested I try with LittleFS I started using it since I was(and still am) doing lots of small appends to a single file. It was apparently working normally, but as of a couple of days go I found an issue where instead of the size not changing, the file itself gets deleted(?) and any attempts to open the file for appending doesn't work. Issue still persists after board reset, need to build and upload a new LittleFS bin for it to operate normally again up until it doesn't.

[MCVE](https://stackoverflow.com/help/mcve) sketch above is still valid.

Any help or directions would be appreciated.

### Debug Messages

**Output before error:**

```
10/02/20 15:51:37 #####################
10/02/20 15:51:37 
10/02/20 15:51:38 LittleFS directory {/} :

10/02/20 15:51:38 	emails.txt	size: 3
10/02/20 15:51:38 	historico.html.gz	size: 54806
10/02/20 15:51:38 	logo-header.png	size: 24178
10/02/20 15:51:38 	medidas.csv	size: 8432
10/02/20 15:51:38 	medidores.html.gz	size: 8713
10/02/20 15:51:38 	reles.txt	size: 5
10/02/20 15:51:38 	timer.txt	size: 3
10/02/20 15:51:38 
10/02/20 15:51:38 #####################
10/02/20 15:51:38 
10/02/20 15:51:38 lfs_file_close: fd=0x3fff4a34
10/02/20 15:51:38 lfs_file_close: fd=0x3fff3e2c
// This is where it tries to open "/medidas.csv" to append to it
10/02/20 15:51:38 LittleFSDirImpl::openFile: rc=-2 fd=0x3fff4894 path=`/medidas.csv` openMode=3 accessMode=2 err=-2
10/02/20 15:51:38 Failed to open "medidas.csv" - Function getMedidas()
```

**Output after error:**

```
// "medidas.csv" no longer exists and 'LittleFS.open("medidas.csv", "a")' doesn't open neither create new file
10/02/20 15:51:38 #####################
10/02/20 15:51:38 
10/02/20 15:51:38 LittleFS directory {/} :

10/02/20 15:51:38 	emails.txt	size: 3
10/02/20 15:51:38 	historico.html.gz	size: 54806
10/02/20 15:51:38 	logo-header.png	size: 24178
10/02/20 15:51:38 	medidores.html.gz	size: 8713
10/02/20 15:51:38 	reles.txt	size: 5
10/02/20 15:51:38 	timer.txt	size: 3
10/02/20 15:51:38 
10/02/20 15:51:38 #####################
10/02/20 15:51:39 
10/02/20 15:51:39 lfs_file_close: fd=0x3fff4894
10/02/20 15:51:39 LittleFSDirImpl::openFile: rc=-2 fd=0x3fff4894 path=`/medidas.csv` openMode=3 accessMode=2 err=-2
10/02/20 15:51:39 Failed to open "medidas.csv" - Function getMedidas()
```

rybertm-old, 2020-02-11:
Today some other file in the file system got truncated to 0 size as well. Not sure why.

earlephilhower, 2020-05-21:
Above is, unfortunately, not an MCVE.  We need to be able to take the MCVE and run it to reproduce things locally, but what's shown is a snippet.

If you have a single complete sketch that just has the file create and then repeated append and it shows the failure, we can take a look at it.  Just strip everything out from the app except the file updates (throw in random data), and get a single complete `sketch.ino` which we can run ourselves on our boards or using the Host emulation (which has things like `valgrind` and other cool tools).

OTW, there's nothing we can do with this report.

earlephilhower, 2020-05-26:
Closing as no feedback or MCVE.  I did my own simple test and it's working like a champ up to ~80KB of 29-byte writes without incident.  I stopped at that point since it went way beyond your own results and I didn't want to use my up flash lifetime.

If you can provide an MCVE that shows the failure, without your special HW, please do open another bug and post it so we can look at it.

````
#include <LittleFS.h>

void setup() {
  Serial.begin(115200);
  Serial.printf("\n\nBEGIN!\n");
  LittleFS.format();
  LittleFS.begin();
}
size_t b = 0;

void listFiles() {
  Dir dir = LittleFS.openDir("/");
  while (dir.next()) {
    Serial.printf("'%s' %d\n", dir.fileName().c_str(), dir.fileSize());
  }
}

void loop() {
   File file = LittleFS.open("/xxx.csv", "a"); 
   b += file.print("1234567890123456789012345678\n");
   file.close();
   Serial.printf("bytes = %d\n", b);
   listFiles();
}
````

````
....
bytes = 72268
'xxx.csv' 72268
bytes = 72297
'xxx.csv' 72297
bytes = 72326
'xxx.csv' 72326
bytes = 72355
'xxx.csv' 72355
bytes = 72384
'xxx.csv' 72384
bytes = 72413
'xxx.csv' 72413
bytes = 72442
'xxx.csv' 72442
bytes = 72471
'xxx.csv' 72471
bytes = 72500
'xxx.csv' 72500
....
````

closed 2020-05-26 by earlephilhower