esp8266/Arduino #7006: Missing information from boards file

jake-code, 2020-01-09:
@----------------------------- Delete below -----------------------------
File https://arduino.esp8266.com/stable/package_esp8266com_index.json is missing the tools download information block for mklittlefs for i686-linux-gnu (32bit) for release 2.6.3 (maybe others in 2.6.x as I was at version 2.5.2 when I noticed there was a problem updating)
----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

d-a-v, 2020-01-09:
I can see it, toolsDependencies[2] version 2.5.0-4-69bd9e6
url: https://github.com/earlephilhower/esp-quick-toolchain/releases/download/2.5.0-4/i686-w64-mingw32-mklittlefs-69bd9e6.zip

jake-code, 2020-01-10:
Not correct, the file you refer to is for use with a 32bit mingw environment running within 64bit Windows, not for native 32bit linux. For this the file would be "i686-linux-gnu.mklittlefs-7f77f2b.1563313032.tar.gz"

d-a-v, 2020-01-10:
What I see is:
```
$ file mklittlefs/mklittlefs.exe
mklittlefs/mklittlefs.exe: PE32 executable (console) Intel 80386 (stripped to external PDB), for MS Windows
```
The architecture file name is "i686-mingw32" in the json file (tools[11].systems[2])

> is for use with a 32bit mingw environment running within 64bit Windows

How is this distinction made ?
Can you please update OP with all the details of your setup ? It currently states that you run on macOS and Linux.
Can you also copy-paste all the error messages you can see ?

d-a-v, 2020-01-10:
@earlephilhower It seems linux/686 entry for mklittlefs has disappeared in [this commit](https://github.com/esp8266/Arduino/commit/72dd589599e0b7d2e208eafeeca01a83b2d4cb94#diff-0ce9364537238452a0ba4bfdb88bba2cL323)

PR #7011 (2020-01-11): Add back mklittlefs for linux32 target
Fixes #7006

mklittlefs for 32-bit Linux disappeared from the packages.json.  Looks
like a transient build problem on the older eqt release (later builds
look fine).  Add it back, pointing to the first successful mklittlefs
build for lin32.
commit 0d38ea73 (2020-01-11): Add back mklittlefs for linux32 target (#7011)

Fixes #7006

mklittlefs for 32-bit Linux disappeared from the packages.json.  Looks
like a transient build problem on the older eqt release (later builds
look fine).  Add it back, pointing to the first successful mklittlefs
build for lin32.
closed 2020-01-11 by earlephilhower