esp8266/Arduino #7007: UDP not receiving data after sometimes

komputerboy, 2020-01-10:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[qio]
- Flash Size: 		        [4MB]
- lwip Variant:                   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:               [115200]

### Problem Description
I have problem with WiFi UDP program with UDP Example. In the beginning Wemos is receiving UDP data. But after some minutes ( > 5 minutes ), Wemos is not receiving UDP data.
Any solution?

Thank You.

### Code
```
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

const char* ssid = "********";
const char* password = "********";

WiFiUDP Udp;
unsigned int localUdpPort = 4210;  // local port to listen on
char incomingPacket[255];  // buffer for incoming packets
char  replyPacket[] = "Hi there! Got the message :-)";  // a reply string to send back

void setup()
{
  Serial.begin(115200);
  Serial.println();

  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected");

  Udp.begin(localUdpPort);
  Serial.printf("Now listening at IP %s, UDP port %d\n", WiFi.localIP().toString().c_str(), localUdpPort);
}

void loop()
{
  int packetSize = Udp.parsePacket();
  if (packetSize)
  {
    // receive incoming UDP packets
    Serial.printf("Received %d bytes from %s, port %d\n", packetSize, Udp.remoteIP().toString().c_str(), Udp.remotePort());
    int len = Udp.read(incomingPacket, 255);
    if (len > 0)
    {
      incomingPacket[len] = 0;
    }
    Serial.printf("UDP packet contents: %s\n", incomingPacket);

    // send back a reply, to the IP address and port we got the packet from
    Udp.beginPacket(Udp.remoteIP(), Udp.remotePort());
    Udp.write(replyPacket);
    Udp.endPacket();
  }
}
```

d-a-v, 2020-02-24:
@komputerboy Please can you retry with latest git master version ?

reynolds087, 2020-02-28:
I have the same problem. I updated today to 2.6.3 and uploaded the sketch and still having the same issue. I have tried everything but after somewhere between 5 and 20 minutes the connection is lost and ESP8266 no longer receives UDP packets. If I ping the ESP8266 from my router, after about 1 minute it will come back online. During the network dropout, the ESP8266 is unreachable, but it's returning WL_CONNECTED, and using event handling, it will never register an event under OnStationModeDisconnected, despite the fact that it is no longer responding on the network.

d-a-v, 2020-02-28:
@komputerboy @reynolds087 
Can you try latest git master, or [beta release 0.0.1](https://d-a-v.github.io) ?

reynolds087, 2020-02-28:
I am willing to try it, but I don't know how to install. Can you send me a link to instructions or just quickly explain? My experience with updating my ESP8266 is just updating the libraries in the Arduino IDE through the board manager. That is how I installed 2.6.3 yesterday.

reynolds087, 2020-02-28:
Nevermind, I see the link for the board manager on the page. I will try it.

reynolds087, 2020-02-29:
I'm not sure if I did it correctly. I added a link to the repository in the preferences under "additional board manager URLs". I left the existing link in there as well so it has this string:

 http://arduino.esp8266.com/stable/package_esp8266com_index.json,https://d-a-v.github.io/esp8266/Arduino/package_esp8266com_index.json

. I then updated esp8266 by ESP8266 Community in the board manager to version 0.0.1--git-version--2.6.3-86-gc61b0de to 0.0.1

Restarted Arduino IDE

ESP8266 Boards (2.6.3-86-gc61b0de) now shows in the Tools => Board context menu.
I opened the same sketch that I have been using with the UDP protocols and flashed it into the ESP8266 without making any changes. It still has the same problem. 

Do I need to delete the old library or take other steps in order to ensure that I am using the correct library?

Not sure if you need to know this, but I am including the following libraries in my sketch: 

#include <stdio.h>
#include <Arduino.h>
#include <Ticker.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <ESP8266WiFiMulti.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

d-a-v, 2020-02-29:
I am running your example without issue for more than 15 minutes.
A message is sent and received every 0.05 sec
```
$ while true; do echo Hello | nc -u 10.0.1.101 4210& pid=$!; sleep 0.05; kill $pid; done
...
[12994] 7179
Hi there! Got the message :-)[12993]-  Terminated              echo Hello | nc -u 10.0.1.101 4210
[12995] 7183
Hi there! Got the message :-)[12994]-  Terminated              echo Hello | nc -u 10.0.1.101 4210
...
```
```
...
18:13:12.106 -> UDP packet contents: Hello
18:13:12.106 -> 
18:13:12.139 -> :urn 6
18:13:12.139 -> Received 6 bytes from 10.0.1.7, port 48019
18:13:12.139 -> :urd 6, 6, 0
18:13:12.139 -> UDP packet contents: Hello
18:13:12.139 -> 
18:13:12.206 -> :urn 6
...
```

reynolds087, 2020-02-29:
Do you see anything in my sketch that would be causing this problem? I don't understand why it's dropping out. No other device on my network has connection instability.

#include <stdio.h>
#include <Arduino.h>
#include <Ticker.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <ESP8266WiFiMulti.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

const char* routerIP = "";
const char* ssid     = "";
const char* password = "";
WiFiUDP UDPTestServer;
ESP8266WiFiMulti WiFiMulti;
unsigned int UDPPort = 6666;
const int packetSize = 255;
char packetBuffer[packetSize];
char keepAlivePacket[] = "Test_Keep_Alive";
Ticker keepAliveTicker;

WiFiEventHandler connectedEventHandler, disconnectedEventHandler, gotIpEventHandler;

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
String strButtonPress = ""; //Console input to initiate IR signal send
IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

void setup() 
{
  Serial.begin(115200);
  delay(20);

  connectedEventHandler = WiFi.onStationModeConnected([](const WiFiEventStationModeConnected& event)
    {
      Serial.println("");
      Serial.println("Station Connected");
    });

    gotIpEventHandler = WiFi.onStationModeGotIP([](const WiFiEventStationModeGotIP& event)
  {
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  });

  disconnectedEventHandler = WiFi.onStationModeDisconnected([](const WiFiEventStationModeDisconnected& event)
    {
      Serial.println("Station disconnected");
    });
  
  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(ssid, password);

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  keepAliveTicker.attach(60, KeepAlive);
  
  while (WiFiMulti.run() != WL_CONNECTED) 
  {
    Serial.print(".");
    delay(500);
  }
  
  irsend.begin();
  UDPTestServer.begin(UDPPort);
}

int value = 0;

void loop() 
{
  if (WiFi.status() != WL_CONNECTED) 
  {
    delay(1);
    WiFiMulti.run();
    return;
  }
    handleUDPServer();
    strButtonPress = "";
    delay(1); 
}

void handleUDPServer() {
  int cb = UDPTestServer.parsePacket();
  if (cb) {
    int len = UDPTestServer.read(packetBuffer, packetSize);
    if (len > 0)
    {
      packetBuffer[len] = 0;
    }
    for(int i = 0; i < strlen(packetBuffer); i++) {
      strButtonPress += packetBuffer[i];
    }
    strButtonPress.trim();
    Serial.println(strButtonPress);
    
    if(strButtonPress.indexOf("NEC_Power") >= 0)
    {
      Serial.println("Sending NEC_Power...");
      irsend.sendNEC(0x20DF10EF);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Input") >= 0)
    {
      Serial.println("Sending NEC_Input...");
      irsend.sendNEC(0x20DFF40B);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Volume_Up") >= 0)
    {
      Serial.println("Sending NEC_Volume_Up...");
      irsend.sendNEC(0x20DF40BF);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Volume_Down") >= 0)
    {
      Serial.println("Sending NEC_Volume_Down...");
      irsend.sendNEC(0x20DFC03F);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Channel_Up") >= 0)
    {
      Serial.println("Sending NEC_Channel_Up...");
      irsend.sendNEC(0x20DF00FF);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Channel_Down") >= 0)
    {
      Serial.println("Sending NEC_Channel_Down...");
      irsend.sendNEC(0x20DF807F);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Mute") >= 0)
    {
      Serial.println("Sending NEC_Mute...");
      irsend.sendNEC(0x20DF906F);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Menu") >= 0)
    {
      Serial.println("Sending NEC_Menu...");
      irsend.sendNEC(0x20DFF20D);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Exit") >= 0)
    {
      Serial.println("Sending NEC_Exit...");
      irsend.sendNEC(0x20DF926D);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Ok") >= 0)
    {
      Serial.println("Sending NEC_Ok...");
      irsend.sendNEC(0x20DF22DD);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Up") >= 0)
    {
      Serial.println("Sending NEC_Up...");
      irsend.sendNEC(0x20DFA25D);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Down") >= 0)
    {
      Serial.println("Sending NEC_Down...");
      irsend.sendNEC(0x20DF629D);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Left") >= 0)
    {
      Serial.println("Sending NEC_Left...");
      irsend.sendNEC(0x20DFE21D);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Right") >= 0)
    {
      Serial.println("Sending NEC_Right...");
      irsend.sendNEC(0x20DF12ED);
      UDPTestServer.flush();
    }
    else if(strButtonPress.indexOf("NEC_Back") >= 0)
    {
      Serial.println("Sending NEC_Back...");
      irsend.sendNEC(0x20DF52AD);
      UDPTestServer.flush();
    } 
    else {}
  }
}

void KeepAlive()
{
  Serial.println("Sending Packet");
  UDPTestServer.beginPacket(routerIP, UDPPort);
  UDPTestServer.write(keepAlivePacket);
  UDPTestServer.endPacket();
  Serial.println("Packet Sent");
}

d-a-v, 2020-03-04:
Did you try to disable ticker ?
Or to use a scheduled function instead of a immediate function with it ?
To do that, use `ticker.attach_scheduled()` instead of `ticker.attach()`

reynolds087, 2020-03-04:
@d-a-v I can try that, but I only added the ticker to try and correct the timeout issue. It sends a UDP message to another computer on the network. I was hoping it would keep the device connected, but it didn't help. For some reason, the device will become unreachable after a period of time and not receive UDP packets. I can't find any rhyme or reason because it will sporadically reconnect, but it's not reliable. The two things that seem to force it start sending or receiving again are when I reboot it or ping it for a few minutes. When I first start pinging it will time out, but then eventually I get a response, and then I can send the UDP.

reynolds087, 2020-03-04:
Ok, I tried 'ticker.attach_scheduled()', but it didn't help.

d-a-v, 2020-03-04:
Did you try `WiFi.setSleepMode(WIFI_NONE_SLEEP);` ?

reynolds087, 2020-03-05:
I added that line in, but it just did the same thing. Not responding to pings or UDP commands after a while.

d-a-v, 2020-03-05:
Magical solution for that kind of issue may include one or several tests from the following list:
(please don't try to read #2330 & al)
- change/improve your power supply
- check DTIM parameter on your AP router (reduce its value)
- try with an external permanent ping to your esp
- start arp gratuitous (unmerged #6889, available in [alpha release](https://d-a-v.github.io) (0.0.2) too)

reynolds087, 2020-03-05:
@d-a-v 
Power supply is probably not an issue, but I will consider it if the other changes don't help.
DTIM is already set to 1 in my wireless access point.
I will add a script to my router to continuously ping the device.
Can you tell me how to enable gratuitous ARP? Would it be 
`ESP8266WiFiMulti.stationKeepAliveEnabled` ??

d-a-v, 2020-03-05:
It's unofficial API, it currently is 
```
WiFi.stationKeepAliveSetIntervalMs(5000); // to be called once, active every 5secs
```
You can use it anytime, it will send gratuitous ARP packets only when interface is up at the given interval.
(This API is supposed to evolve)

reynolds087, 2020-03-06:
That seems to have fixed the issue for me. Sometimes I have to send the command 4 or 5 times if I haven't connected to it in a while, but I no longer have to ping the device for 2 minutes before it will respond.

d-a-v, 2020-03-06:
Thanks for testing !
Let's keep this issue open until the API is merged

d-a-v, 2021-01-28:
#6889 is merged, closing

closed 2021-01-28 by d-a-v

plcdnl, 2021-06-19:
How can i use this function ? 
`WiFi.stationKeepAliveSetIntervalMs(5000);`

The IDE give me this error when i try to compile
`
C:\Users\User\Downloads\project\project.ino: In function 'void setup()':
project:37:8: error: 'class ESP8266WiFiClass' has no member named 'stationKeepAliveSetIntervalMs'
   WiFi.stationKeepAliveSetIntervalMs(5000);
        ^
exit status 1
'class ESP8266WiFiClass' has no member named 'stationKeepAliveSetIntervalMs'
`

d-a-v, 2021-06-19:
Try 
```cpp
#include <ESP8266WiFiGratuitous.h>
…

    ESP8266WiFiGratuitous::stationKeepAliveSetIntervalMs(5000);`
```