esp8266/Arduino #7008: Enhancement request: support for Server Sent Events aka EventSource

ewaldc, 2020-01-10:
This is a request for support of HTML5 Server Sent Events aka [Event Source](https://html.spec.whatwg.org/multipage/server-sent-events.html).
It allows e.g. to implement an event bus over HTML.

The required changes to the ESP8266WebServer library are absolutely minimal: zero impact on existing code and increase in code size/memory footprint of just few bytes (basically sizeof bool).

Essentially, the variable _bool keepCurrentClient_ which is currently hardcoded to _false_, becomes a protected class variable in the ESP8266WebServer class that can be modified via a new function called _keepCurrentClient(bool)_.  

To enable Server Sent Events/event subscription, it requires two web server instances, one being changed to keep the client socket/session (an event listener) open until disconnected. 
```	
ESP8266WebServer _server(port);
ESP8266WebServer _SSEserver(port + 1);
_server.begin();
_SSEserver.keepCurrentClient(true);
_SSEserver.begin();
```
Everything else can implemented in user code (event subscription, event bus keepalive messages, event payload etc.).
I have this currently working on a clean fork from master branch with just this one change.
Happy to post a PR (it's ready to post), if this is a reasonable request.
I can also contribute a working example, I suppose in the ESP8266WebServer examples section.

Thanks once again for this brilliant library and for taking the time to consider this request,
Ewald

d-a-v, 2020-01-10:
Thanks !
Please post your PR and its example

ewaldc, 2020-01-12:
@d-a-v,
Example submitted.  It turns out that there is no need for the _keepCurrentClient_ enhancement.
I found another way to make it work: 
1. set the contentLength to _CONTENT_LENGTH_UNKNOWN_ which allows to submit a virtually endless payload (needed for an event bus)
2. add a KeepAlive timer, using the Ticker library, which keeps the socket open by sending e.g. every minute a keep-alive message in case there are no (normal) state update events.  The use of a KeepAlive timer is not a violation of the with SSE - EventSource standard, it just requires the client to handle/ignore the KeepAlive events.

Hence, no need for the enhancement request. Sorry for that, but I hope the example is still useful.  I tried for days to make it work using WebSockets and AsyncWebServer but both solutions are much more complex, consume much more memory and could not be made stable.

The one thing I have not figured out is why either of _client.setNoDelay(true);_ or _client.setSync(true)_; (lines 121/122 in the example) cause a WDT Soft Reset in the KeepAlive function when issuing _client.println()_, while the regular state event updates using a similar _client.printf_P()_ continues to run without any problem!  They both write to the same WiFiClient with IMHO very similar calls.
Setting "no delay" would ideally make the event bus a bit more real time.  Adding yield() calls makes no difference, as expected as there is no blocking user code.
Any idea why this would cause a WDT Soft Reset ?

Hope the example is still useful.

d-a-v, 2020-01-17:
I think you should try to use `ticker.attach_scheduled()` instead of `ticker.attach()` so you can do whatever is needed in the callback (otherwise the callback is treated like an ISR that is to be used with more constraints).

ewaldc, 2020-01-18:
Thanks for the suggestion, this was already top of mind as it is not recommended to do networking stuff in _Ticker.attach()_, even though a "keepalive" packet is less than 100 bytes.  Rather than posting a PR I will post the update code a link to e.g. pastebin.

d-a-v, 2020-01-18:
Just commit changes to your branch, it will update the PR

ewaldc, 2020-01-20:
Thanks for your flexibility, I have posted the updated example [**here**](https://github.com/ewaldc/Arduino/tree/master/libraries/ESP8266WebServer/examples/ServerSentEvents)
What's new:
- code simplified and reduced in # of lines, yet more functional
- removed second web server
- every event subscription now has its own Ticker for keep-alive events
- moved to _Ticker.attach_scheduled()_, which allows for network operation within the callback
- Eventsource socket now set to _setNoDelay(true)_ and _setSync(true)_ (no more crashes)
- multiple subscriptions and listeners allowed (also for each client IP)
- introduction of listener channels
- documentation improved (within code)
- validated with lasted master branch
Essentially the same code within a much larger application has passed 48 hours of stress test

ewaldc, 2020-02-04:
[Example](https://github.com/ewaldc/Arduino/blob/master/libraries/ESP8266WebServer/examples/ServerSentEvents/ServerSentEvents.ino) updated after feedback/review from users

PR #7012 (2020-05-16): Server Sent Events example - issue #7008
Illustrates the use of SSE using ESP8266WebServer

*edit from maintainer:* closes #7008
closed 2020-05-16 by earlephilhower