esp8266/Arduino #7017: How can i decode the IR protocol from these details?

wounder123, 2020-01-16:
From below, response get from IR receiver (using  Arduino library )
could you please help me decode the protocol formate?
thanks.
 
Timestamp : 000657.493
Library   : v2.7.2

Protocol  : UNKNOWN
Code      : 0xB4C0FCA9 (129 Bits)
uint16_t rawData[257] = {462, 1044,  474, 532,  472, 1016,  492, 532,  472, 532,  448, 1038,  
492, 514,  490, 1010,  462, 544,  490, 1014,  450, 558,  416, 1090,  474, 1012,  490, 516,  492, 
1008,  418, 590,  462, 544,  492, 514,  584, 440,  472, 532,  536, 474,  462, 538,  446, 560,  
416, 590,  462, 548,  492, 516,  464, 560,  434, 570,  444, 556,  466, 538,  416, 588,  462, 
544,  490, 1014,  472, 534,  462, 548,  486, 518,  492, 514,  492, 536,  560, 440,  468, 534,  
472, 532,  416, 590,  462, 544,  492, 514,  492, 532,  472, 532,  472, 532,  472, 532,  444, 
560,  416, 590,  486, 520,  490, 516,  490, 534,  470, 538,  446, 554,  470, 534,  416, 590,  
462, 544,  490, 516,  492, 534,  470, 534,  470, 534,  470, 534,  470, 536,  416, 590,  462, 
544,  490, 516,  490, 534,  470, 534,  470, 534,  470, 534,  418, 588,  460, 546,  488, 518,  
490, 536,  470, 532,  472, 534,  470, 534,  468, 536,  416, 590,  462, 548,  486, 520,  492, 
478,  518, 536,  470, 534,  470, 534,  416, 588,  460, 546,  490, 518,  490, 534,  470, 534,  
470, 534,  470, 534,  444, 560,  416, 600,  454, 544,  490, 518,  488, 538,  470, 530,  468, 
536,  470, 536,  416, 590,  434, 578,  482, 520,  490, 534,  470, 536,  474, 528,  470, 538,  
458, 542,  416, 592,  460, 546,  490, 518,  488, 538,  470, 536,  444, 556,  446, 560,  442, 
562,  482, 530,  476, 530,  478, 544,  462, 540,  468, 536,  468, 536,  466, 538,  416, 590,  
436, 570,  488, 518,  488, 536,  468};

Pablo2048, 2020-01-16:
Are you serious? This is your second issue about something with IR protocol/decoding which has **nothing** to do with this core. Why you didn't read the issue template?

d-a-v, 2020-01-16:
@wounder123 
Please ask on [this repository](https://github.com/crankyoldgit/IRremoteESP8266) instead.
We can't help here.

closed 2020-01-16 by d-a-v