esp8266/Arduino #7019: ESP-8266-LED- control // using watchdog timer EEPROM to save to flash memory // error restarting

droidbreaker, 2020-01-17:
Hello all,
            I am creating new project DIY for my home automation and i am new to this esp-8266 why i am using this is to save my cost and also saves space in equipment
            But i am having error in code 
            
"ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
LED1 = 
673LED2 = 
541  "

here LED1- control for led strip 1
LED 2 does for another strip
so i have given values to those LED's via my webpage slider i also have uploaded code;

also i am using watchdog timer and EEPROM but i think there are minor error for using them with my esp-8266 this ESP crashes several times
kindly preview my code and if possible guide me and give me solution for this right now i am using nodeMCU and Huzza feather development board for esp-8266-12E
HEre is my code:
//<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>//
//#include <WiFi.h>
//#include <ESP8266WebServer.h>
#include <ESP8266WiFi.h>

#include <EEPROM.h>
#include <Ticker.h>

Ticker secondTick;

#define debug 1

volatile int watchdogCount = 0;

void ISRwatchdog() {
  watchdogCount++;
  if ( watchdogCount == 5 ) {
     // Only print to serial when debugging
     //(debug) && Serial.println("The dog bites!");
     ESP.reset();
  }
}

// GPIO the LED is attached to
static const int LED1pin = 5;//LED1pin white to MT pwm dimming pin
static const int LED2pin = 13; //LED2pin warm to cct change pwm
//flash memory storage variable 
unsigned long  LED1slide; 
unsigned long  LED2slide; 
unsigned long  LED1Button; 
unsigned long  LED1Butslide;

// Replace with your network credentials
const char* ssid     = "smrttest_1huzza";
const char* password = "12345678";

// Set web server port number to 80
WiFiServer server(80);

// Variable to store the HTTP request
String header;

// Decode HTTP GET value
String valueString = String(5);
String valueStringg = String(5);
//slider 3 button
//String valueStringb = String(5);
[... 195 lines cut ...]
              
               
            }
            
            
            // The HTTP response ends with another blank line
            client.println();
            // Break out of the while loop
            break;
          } else { // if you got a newline, then clear currentLine
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }
      }
    }
    // Clear the header variable
    header = "";
    // Close the connection
    client.stop();
    Serial.println("Client disconnected.");
    Serial.println("");
  }
}

Tech-TX, 2020-01-18:
You need to look at your WHILE loop, as I don't see a delay() or yield() in there anywhere, as far as I can tell.  If you don't periodically return control to the system, the WDT fires within a couple of seconds.

This is not a problem with the core, and should rightfully be addressed on esp8266.com or forums.adafruit.com.

devyte, 2020-01-18:
Previous comment is correct. Also, issue template was ignored.
Closing due to not compliant.

closed 2020-01-18 by devyte

droidbreaker, 2020-01-20:
@Tech-TX and @devyte  Thank you sir i will look into my while() loop and check things one by one.