esp8266/Arduino #7028: 'Stream' has not been declared

stepanurban, 2020-01-21:
Hi,
I'm getting following error when trying to compile ScanNetworksAdvanced example for Arduino nano 33
/home/.../Arduino/libraries/WiFiEspAT/src/utility/EspAtDrvTypes.h:61:30: error: 'Stream' has not been declared
   virtual uint8_t readRxData(Stream* serial, size_t len) = 0;

Where is the problem? How to get it?
Thank you

closed 2020-01-21 by stepanurban

olsonap, 2021-11-15:
Did you figure this out? I'm having the same issue.

d-a-v, 2021-11-15:
Try with `#include <Stream.h>`

olsonap, 2021-11-15:
No luck.

C:\esp\myProjects\a2dp_gatts_coex>idf.py build 
Executing action: all (aliases: build)
Running ninja in directory c:\esp\myprojects\a2dp_gatts_coex\build
Executing "ninja all"...
[1/13] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\esp\myProjects\a2dp_gatts_coex\build\bootloader\esp-idf\esptool_py && C:\Users\aman_\.espressif\python_env\idf4.4_py3.9_env\Scripts\python.exe C:/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/esp/myProjects/a2dp_gatts_coex/build/bootloader/bootloader.bin"
Bootloader binary size 0x6350 bytes. 0xcb0 bytes (11%) free.
[4/13] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/PulseSensorPlayground.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/PulseSensorPlayground.cpp.obj 
ccache C:\Users\aman_\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -IC:/esp/esp-idf/components/newlib/platform_include -IC:/esp/esp-idf/components/freertos/include -IC:/esp/esp-idf/components/freertos/include/esp_additions/freertos -IC:/esp/esp-idf/components/freertos/port/xtensa/include -IC:/esp/esp-idf/components/freertos/include/esp_additions -IC:/esp/esp-idf/components/esp_hw_support/include -IC:/esp/esp-idf/components/esp_hw_support/include/soc -IC:/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -IC:/esp/esp-idf/components/esp_hw_support/port/esp32/. -IC:/esp/esp-idf/components/heap/include -IC:/esp/esp-idf/components/log/include -IC:/esp/esp-idf/components/lwip/include/apps -IC:/esp/esp-idf/components/lwip/include/apps/sntp -IC:/esp/esp-idf/components/lwip/lwip/src/include -IC:/esp/esp-idf/components/lwip/port/esp32/include -IC:/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/esp/esp-idf/components/soc/include -IC:/esp/esp-idf/components/soc/esp32/. -IC:/esp/esp-idf/components/soc/esp32/include -IC:/esp/esp-idf/components/hal/esp32/include -IC:/esp/esp-idf/components/hal/include -IC:/esp/esp-idf/components/hal/platform_port/include -IC:/esp/esp-idf/components/esp_rom/include -IC:/esp/esp-idf/components/esp_rom/include/esp32 -IC:/esp/esp-idf/components/esp_rom/esp32 -IC:/esp/esp-idf/components/esp_common/include -IC:/esp/esp-idf/components/esp_system/include -IC:/esp/esp-idf/components/esp_system/port/soc -IC:/esp/esp-idf/components/esp_system/port/public_compat -IC:/esp/esp-idf/components/esp32/include -IC:/esp/esp-idf/components/xtensa/include -IC:/esp/esp-idf/components/xtensa/esp32/include -IC:/esp/esp-idf/components/driver/include -IC:/esp/esp-idf/components/driver/esp32/include -IC:/esp/esp-idf/components/esp_pm/include -IC:/esp/esp-idf/components/esp_ringbuf/include -IC:/esp/esp-idf/components/efuse/include -IC:/esp/esp-idf/components/efuse/esp32/include -IC:/esp/esp-idf/components/vfs/include -IC:/esp/esp-idf/components/esp_wifi/include -IC:/esp/esp-idf/components/esp_event/include -IC:/esp/esp-idf/components/esp_netif/include -IC:/esp/esp-idf/components/esp_eth/include -IC:/esp/esp-idf/components/tcpip_adapter/include -IC:/esp/esp-idf/components/esp_phy/include -IC:/esp/esp-idf/components/esp_phy/esp32/include -IC:/esp/esp-idf/components/esp_ipc/include -IC:/esp/esp-idf/components/app_trace/include -IC:/esp/esp-idf/components/esp_timer/include -IC:/esp/esp-idf/components/mbedtls/port/include -IC:/esp/esp-idf/components/mbedtls/mbedtls/include -IC:/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -IC:/esp/esp-idf/components/app_update/include -IC:/esp/esp-idf/components/spi_flash/include -IC:/esp/esp-idf/components/bootloader_support/include -IC:/esp/esp-idf/components/nvs_flash/include -IC:/esp/esp-idf/components/pthread/include -IC:/esp/esp-idf/components/esp_gdbstub/include -IC:/esp/esp-idf/components/esp_gdbstub/xtensa -IC:/esp/esp-idf/components/esp_gdbstub/esp32 -IC:/esp/esp-idf/components/espcoredump/include -IC:/esp/esp-idf/components/espcoredump/include/port/xtensa -IC:/esp/esp-idf/components/wpa_supplicant/include -IC:/esp/esp-idf/components/wpa_supplicant/port/include -IC:/esp/esp-idf/components/wpa_supplicant/esp_supplicant/include -IC:/esp/esp-idf/components/ieee802154/include -IC:/esp/esp-idf/components/console -IC:/esp/esp-idf/components/mdns/include -IC:/esp/esp-idf/components/esp_adc_cal/include -IC:/esp/esp-idf/components/bt/common/osi/include -IC:/esp/esp-idf/components/bt/include/esp32/include -IC:/esp/esp-idf/components/bt/common/api/include/api -IC:/esp/esp-idf/components/bt/common/btc/profile/esp/blufi/include -IC:/esp/esp-idf/components/bt/common/btc/profile/esp/include -IC:/esp/esp-idf/components/bt/host/bluedroid/api/include/api -IC:/esp/esp-idf/components/protobuf-c/protobuf-c -IC:/esp/esp-idf/components/nghttp/port/include -IC:/esp/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/esp/esp-idf/components/esp_http_server/include -IC:/esp/esp-idf/components/protocomm/include/common -IC:/esp/esp-idf/components/protocomm/include/security -IC:/esp/esp-idf/components/protocomm/include/transports -IC:/esp/esp-idf/components/json/cJSON -IC:/esp/esp-idf/components/wifi_provisioning/include -IC:/esp/esp-idf/components/wear_levelling/include -IC:/esp/esp-idf/components/sdmmc/include -IC:/esp/esp-idf/components/fatfs/diskio -IC:/esp/esp-idf/components/fatfs/vfs -IC:/esp/esp-idf/components/fatfs/src -IC:/esp/esp-idf/components/spiffs/include -IC:/esp/esp-idf/components/openssl/include -IC:/esp/e [... cut ...]
      ^~~~~~~~~~~~~~~~~~~~~
../main/PulseSensorPlayground.cpp:199:6: note: no functions named 'void PulseSensorPlayground::setSerial(Stream&)'
In file included from ../main/PulseSensorPlayground.cpp:17:
../main/PulseSensorPlayground.h:126:7: note: 'class PulseSensorPlayground' defined here
 class PulseSensorPlayground {
       ^~~~~~~~~~~~~~~~~~~~~
[7/13] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/utility/PulseSensor.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

olsonap, 2021-11-15:
Nevermind. User error. Accidentally deleted an entire function somehow.