esp8266/Arduino #7032: ESP-8266 websocket-EEPROM interface issue

droidbreaker, 2020-01-24:
Hello Sir,
               I am at beginner level and i wanted to implement EEPROM write after receiving data from websocket if websocket doenst receive any data within 5 seconds then the value which i have sent via websocket to turn on and off led here LED_STATE want to write it to EEPROM after 5 seconds if there will be no incoming data from websocket :
Here is my code please if you can help me out to solve this problem thank you.
I have modified as per my need.
but require EEPROM for websocket if possible.

#include <ESP8266WiFi.h>
#include <WebSocketsServer.h>
#include <Hash.h>
#include <ESP8266WebServer.h>
#include <EEPROM.h>

WiFiServer server(80);
WebSocketsServer webSocket = WebSocketsServer(81);
 
const int pin_led = 5;
unsigned int value;
String data;
const char *msg_toggle_led = "toggleLED";
const char *msg_get_led = "getLEDState";
char msg_buf[10];
int led_state = 0;
int counter = 0; // for EEPROM testing

char* ssid = "esp8266websocket_1";
char* password = "123456789";
 
void setup()
{
 // EEPROM.begin(512);
  pinMode(pin_led, OUTPUT);
  digitalWrite(pin_led,LOW);
 
  Serial.begin(115200);
  Serial.println();
  Serial.println("Serial started at 115200");
  Serial.println();
 
  // Connect to a WiFi network
  Serial.print(F("Connecting to "));  Serial.println(ssid);
  WiFi.mode(WIFI_AP);
  IPAddress apLocalIp(10,50,1,1);
  IPAddress apSubnetMask(255,255,255,0);
  WiFi.softAPConfig(apLocalIp,apLocalIp,apSubnetMask);
  WiFi.softAP(ssid, password);
  WiFi.setSleepMode(WIFI_NONE_SLEEP);

 
  Serial.println("");
  Serial.println(F("[CONNECTED]"));   Serial.print("[IP ");  Serial.print(WiFi.localIP()); 
  Serial.println("]");
 
  // start a server
  server.begin();
  Serial.println("Server started");
  webSocket.begin();
  webSocket.onEvent(onWebSocketEvent);
  //Serial.print(webSocket.localIP());

}
 
void loop()
{
    webSocket.loop();
 
  
}
 

// Called when receiving any WebSocket message
void onWebSocketEvent(uint8_t client_num,
                      WStype_t type,
                      uint8_t * payload,
                      size_t length) {
 
  
  
  // Figure out the type of WebSocket event
  switch(type) {
 
    // Client has disconnected
    case WStype_DISCONNECTED:
      Serial.printf("[%u] Disconnected!\n", client_num);
      break;
 
    // New client has connected
    case WStype_CONNECTED:
      {
        IPAddress ip = webSocket.remoteIP(client_num);
        Serial.printf("[%u] Connection from ", client_num);
        Serial.println(ip.toString());
      }
      break;
 
    // Echo text message back to client
    case WStype_TEXT:
      Serial.printf("[%u] Text: %s\n", client_num, payload);
      webSocket.sendTXT(client_num, payload);
     // data = webSocket.getData(payload);
     
      Serial.printf("[%u] Received text: %s\n", client_num, payload);
 
      // Toggle LED
      if ( strcmp((char *)payload, "toggleLED") == 0 ) {
        counter++;
        led_state = led_state ? 0 : 1;
        //sprintf(msg_buf, "%d", led_state);
        Serial.printf("Toggling LED to %u\n", led_state);
        digitalWrite(pin_led, led_state);
       // webSocket.sendTXT(client_num, msg_buf);
    
      // Report the state of the LED
      } else if ( strcmp((char *)payload, "getLEDState") == 0 ) {
        sprintf(msg_buf, "%d", led_state);
        Serial.printf("Sending to [%u]: %s\n", client_num, msg_buf);
        webSocket.sendTXT(client_num, msg_buf);
        
      // Message not recognized
      } else {
        Serial.println("[%u] Message not recognized");
      }
     
    
     
     
    
      
      
      
     
      
      
      
      
      
      break;
 
    // For everything else: do nothing
    case WStype_BIN:
    case WStype_ERROR:
    case WStype_FRAGMENT_TEXT_START:
    case WStype_FRAGMENT_BIN_START:
    case WStype_FRAGMENT:
    case WStype_FRAGMENT_FIN:
    default:
      break;
  }
 
}

I am not using SPIFFS i am calling webpage in HTML which is 

/////<<<<<<<<<<<<<<<<<<<>>>>>>>>>>>>>>>>>/////
<!DOCTYPE html>
<html>
<style>

input[type="text"]{
  width: 90%;
  height: 3vh;
}

input[type="button"]{
  width: 9%;
  height: 3.6vh;
}

.rxd{
  height: 90vh;
}

textarea {
  width: 99%;
  height: 100%;
  resize: none;
}

</style>

<script>
var Socket;
function start() {
  Socket = new WebSocket('ws://10.50.1.1:81/');
  
  Socket.onmessage = function(evt) {
	document.getElementById("rxConsole").value += evt.data;
  }
  
}

function enterpressed() {
 Socket.send(document.getElementById("txbuff").value);
 document.getElementById("txbuff").value = "";
}
</script>

<body onload="start();">
   <div>		
		<input class="txd" type="text" id="txbuff" onkeydown="if(event.keyCode == 13) enterpressed();">
		<input class="txd" type="button" onclick="enterpressed();" value="Send" >
  </div>
  <br>
  <div class="rxd">
    <textarea id="rxConsole" readonly></textarea>
  </div>
</body>
</html>

<<<<<<<<<<<<<<<<<<<<<<>>>>>>>>>>>>>>>>>>>>>>>
please help to implement EEPROM in WEBSOCKET because if websocket starts it stays there as long as clients are connected. so need help to implement websocket for ToggleLED status save data after 5 second if there is no data coming from html webpage.
please help if possible thank you all.

devyte, 2020-01-24:
This isn't the right place to request help with your particular project. This is an issue tracker, meant for tracking issues in the core. Please request such help at a community forum like esp8266.com or stackoverflow.
Also, issue template was ignored. 
Closing due to not compliant.

closed 2020-01-24 by devyte

droidbreaker, 2020-01-25:
ok sir thank you for the guidance.!!