esp8266/Arduino #7033: File download from https server does not work

nightflyer88, 2020-01-24:
#### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.6.3]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qior]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I want to download this file from an https server (github) with the ESP8266: https://github.com/nightflyer88/test/releases/download/V2.0/index.html.gz

But it doesn't work, the server responds with 404. Is there a certification problem?

I can load the file using the web browser on Mac, only it doesn't work on the ESP8266.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

/**
   BasicHTTPSClient.ino

    Created on: 20.08.2018

*/

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
//#include <WiFiClientSecure.h>
#include <WiFiClientSecureBearSSL.h>

#include <ESP8266HTTPClient.h>

const uint8_t fingerprint[20] = {0xCA, 0x06, 0xF5, 0x6B, 0x25, 0x8B, 0x7A, 0x0D, 0x4F, 0x2B, 0x05, 0x47, 0x09, 0x39, 0x47, 0x86, 0x51, 0x15, 0x19, 0x84};

ESP8266WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(115200);
  Serial.setDebugOutput(true);

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }
[... 166 lines cut ...]
[HTTP-Client][handleHeaderResponse] RX: 'Status: 404 Not Found'
[HTTP-Client][handleHeaderResponse] RX: 'Vary: X-PJAX'
[HTTP-Client][handleHeaderResponse] RX: 'Cache-Control: no-cache'
[HTTP-Client][handleHeaderResponse] RX: 'Set-Cookie: has_recent_activity=1; path=/; expires=Sat, 18 Jan 2020 21:14:32 -0000'
[HTTP-Client][handleHeaderResponse] RX: 'Set-Cookie: _octo=GH1.1.1957086682.1579378472; domain=.github.com; path=/; expires=Tue, 18 Jan 2022 20:14:32 -0000'
[HTTP-Client][handleHeaderResponse] RX: 'Set-Cookie: _gh_sess=RnVtMU4wZGN5OU5sdmFOUWNkb0JwaGlIdEw2MC8zWTFEZXFHdEFDZjlTdGpTMGdrenI4T3ZBSGtyNmhRMDBibU5PWW5vcUlQR2htem9DMmxMYk5DYmpsYUJwL0VnTXNXYytaNHMxeDlJM3crdDc2UldGeWJ5SFI4OHVjZzk3S3ZLOXYzY1N4d3hFQjdudm54c3dzYjRnPT0tLURST1lVaGEzMDVpSHlEaEh4UDEvaEE9PQ%3D%3D--a1c79406044787d94dbc38613f65d152e8bd6b4f; path=/; secure; HttpOnly'
[HTTP-Client][handleHeaderResponse] RX: 'X-Request-Id: 513878d7-f6bd-430d-8426-cbbd3f99d12a'
[HTTP-Client][handleHeaderResponse] RX: 'Strict-Transport-Security: max-age=31536000; includeSubdomains; preload'
[HTTP-Client][handleHeaderResponse] RX: 'X-Frame-Options: deny'
[HTTP-Client][handleHeaderResponse] RX: 'X-Content-Type-Options: nosniff'
[HTTP-Client][handleHeaderResponse] RX: 'X-XSS-Protection: 1; mode=block'
[HTTP-Client][handleHeaderResponse] RX: 'Referrer-Policy: origin-when-cross-origin, strict-origin-when-cross-origin'
[HTTP-Client][handleHeaderResponse] RX: 'Expect-CT: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Security-Policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; connect-src 'self' uploads.github.com www.githubstatus.com collector.githubapp.com api.github.com www.google-analytics.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com wss://live.github.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com; frame-ancestors 'none'; frame-src render.githubusercontent.com; img-src 'self' data: github.githubassets.com identicons.github.com collector.githubapp.com github-cloud.s3.amazonaws.com *.githubusercontent.com; manifest-src 'self'; media-src 'none'; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com'
[HTTP-Client][handleHeaderResponse] RX: 'X-GitHub-Request-Id: E6C6:12AE:1329988:1D134EA:5E236727'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 404
[HTTP-Client][handleHeaderResponse] Transfer-Encoding: chunked
[HTTPS] GET... code: 404
[HTTP-Client][end] still data in buffer (512), clean up.
pm open,type:2 0
[HTTP-Client][end] tcp keep open for reuse
********************************************************** Wait 10s before next round...

```

devyte, 2020-01-27:
@liebman at a glance I see that there are redirects involved. If memory serves, you were involved with httpClient redirects in the past. Could you please help with this one?

liebman, 2020-01-27:
Yes - this looks like a bug.  From the log it's re-using the connection even tho the host changes and it should have closed and opened a new one.  I'll try to look at this when I get home this afternoon.

closed 2020-01-28 by devyte

devyte, 2020-01-28:
@liebman thanks for the help!

PR #7039 (2020-01-28): Fix redirect support in http client.
Looks like this was introduced in #5250 - it missed the need to clear `_canReuse` in `setUrl()` if the new URL has a hostname and is not just a path.  Without it `disconnect()` will assume it can reuse even tho the hostname may be different.

fixes #7033