esp8266/Arduino #7037: my esp8266 function ESP.deepSleepMax() return Negative value

artcheeze, 2020-01-26:
#### Platform

- Hardware:			        ESP-8266
- Core Version:			-
- Development Env:		VS Code
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU v2
- Flash Mode:			 -
- Flash Size: 		         4MB
- lwip Variant:                    -
- Reset Method:		  -
- Flash Frequency:	          -
- CPU Frequency:		160MHz
- Upload Using:		OTA
- Upload Speed:              115200

### Problem Description

[1] I've used ESP.DeepSleep(**max**) that ' **max**'  variable are come from function ESP.deepSleepMax();
First time running **max** is not a problem and the program run correctly and do the ESP.DeepSleep(max) at the end of code but when my NodeMcu wake up in second time '**max**' is negative value ( Ex. max = -6867045959 ) and then my NodeMcu do not wake up again until i have unplugged the USB and plug again.

[2] my NodeMcu `max ` never reach 71 minute max sleep as show in documentation 

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
int FAIL_GET = 0;
void setup(){
    Serial.begin(115200);
}
void loop(){
    int avgCounter = 0;
    long max = ESP.deepSleepMax();
    long time = (max / 1e6) / 60;
    Serial.println("");
    Serial.print("Max Sleep Time : ");
    Serial.print(max);
    Serial.print(" ---> ");
    Serial.println(time);
    Serial.println("");
    WIFI_CONNECT(); 
    WAKE_UP();
    GET(); 
    Serial.println("xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx");
    ESP.deepSleep(max);
}

```

### Debug Messages

```
First time runing 

{lp�N��l��#o�|$��b��on�l���oN$`N{���N#�lrےnb�l�crc��ㄛ ��o�
Max Sleep Time : 69206009 ---> 1

......................Connect!
----------- Waking Up !! -------------
200
----------- GETTT !! -------------
200
false
NO
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
-----------------------------------------------------------------------------------------------
Second time running after woke up

{lp�N��l��#o�|$��b��on�l���oN$`N{���N#�lrےnb�l�crc��ㄛ ��o�
Max Sleep Time : -56206009 ---> -1

......................Connect!
----------- Waking Up !! -------------
200
----------- GETTT !! -------------
200
false
NO
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

```

PS. Sorry for my English skill, I'm weak English skill but i cannot find any solution at my country

jandrassy, 2020-01-26:
`unsigned long max = ESP.deepSleepMax();`

dok-net, 2020-01-27:
@artcheeze and @jandrassy Could you please try this corrected MVCE, both `ESP.deepSleepMax()` and `Esp.deepSleep()`work with unsigned 64 bit integer, not long, float or long unsigned:
```
void setup(){
    Serial.begin(115200);
}
void loop(){
    const auto max = ESP.deepSleepMax();
    long unsigned time = (max / 1000000ULL) / 60;
    Serial.println("");
    Serial.print("Max Sleep Time : ");
    Serial.print(max);
    Serial.print(" ---> ");
    Serial.print(time);
    Serial.println(" seconds");
    Serial.println("xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx");
    ESP.deepSleep(max);
}
```

d-a-v, 2020-01-27:
> `Serial.println("xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx");`

`Serial.flush()` should work instead

artcheeze, 2020-01-27:
> @artcheeze and @jandrassy Could you please try this corrected MVCE, both `ESP.deepSleepMax()` and `Esp.deepSleep()`work with unsigned 64 bit integer, not long, float or long unsigned:
>
> ```
> void [...]

I got error at Serial.println(max) 
[image]

d-a-v, 2020-01-27:
try with `Serial.printf("%llud", max);`

dok-net, 2020-01-27:
I've updated the MVCE, didn't have compiler or target around at the time :-)
@d-a-v thanks for pointing out the format string -why not overload Serial.print for (u)int64_t?

devyte, 2020-01-27:
@dok-net is correct: the return value is unsigned 64bit, but you're assigning to signed 32bit. In case of 32bit overflow you'll get a wrap around to negative value.
For help with types or language basics in general please refer to a community forum like esp8266.com or stackoverflow.
Closing due to user error.

closed 2020-01-27 by devyte

d-a-v, 2020-01-27:
@dok-net Yes such overloads would be useful.

dok-net, 2020-01-27:
@d-a-v A few bytes can be saved by optimizing print.. whether and which base to support comes to mind.