esp8266/Arduino #7038: Smartconfig Status (idle, busy)

paccerdk, 2020-01-27:
Currently, it doesn't seem to be possible to check if SmartConfig is currently in progress or has been inactive. (only if it has finished)

Lets say you'd want SmartConfig to timeout after 60 seconds passed (and fall back to AP mode), if a SmartConfig client "connects" after 50 seconds, it would be helpful to be able to reset the timeout back to 60 seconds (or pause timer), so that in the case that SmartConfig takes more than 10 seconds to finish, it doesn't get interrupted.

Is there any way of detecting this? I've looked around, but could not find one. 
If not, i think it would be a nice addition to the SmartConfig API.

ESP8266WiFiSTAClass::_smartConfigCallback already seems to be keeping track of status, so i guess it at least should be possible to check it

devyte, 2020-01-27:
@paccerdk At a glance: you have WiFi.smartConfigDone() at your disposal. Start smartconfig, then check in loop() whether it's done together with whether wifi has been setup, and timeout after a while. I suggest using polledTimeout for the timeout bit.
For further help with implementation details please refer to a community forum like esp8266.com or stackoverflow.
BTW, issue template was ignored.
Closing due to not compliant.

closed 2020-01-27 by devyte

paccerdk, 2020-01-29:
@devyte the issue template was not ignored, but it didn't seem to apply for this case, since it seems catered towards bugs not not lack of functionality., I read through the documentation in it though.

I believe you've misunderstood the whole purpose of this, **i already have an implementation using smartConfigDone()**, but as i wrote, it does not allow to check if smartconfig is currently in progress with a client "connected".  
I sat down and spent quite a while writing this issue to avoid exactly this misunderstanding. please re-read my examples, i can only check if smartconfig has finished or not, i'd like to check if its **idle**

smartconfig can easily take 30 seconds or more to finish, depending on conditions, i don't want my users to time out while smartconfig is actively working, recieveing/decrypting packages etc., it don't see any way of detecting this currently.

devyte, 2020-01-30:
The issue template is required for every issue, whether bug, enhancement request documentation, or whatever else. In this case, in addition to the required info fields, you could have filled out the sketch part with a sketch using your smart config code, including comments on relevant lines and output msgs where applicable, and avoided confusion.

I don't use smart config, but my suspicion is that your timeout code is wrong. Please read carefully what I wrote in my previous comment. You mentioned a timer, but I mentioned polling. They are very different approaches.

paccerdk, 2020-02-08:
@devyte 
If your suspicion is that my timeout code is wrong, then you haven't understood the issue.

The functionality i am requesting is currently not a part of the API [[1]](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L762) [[2]](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L780) [[3]](https://github.com/esp8266/Arduino/blob/master/tools/sdk/include/smartconfig.h#L35)
As stated a few times above, i am not simply trying to determine if SmartConfig has finished or not. **i am trying to determine if it was attempted recently**.

For this reason, i don't see how i would be able to provide a sketch (without using pseudocode)

Either way, i am going to leave things here if its still not clear what i am trying to say.
Thanks for trying

devyte, 2020-02-08:
@paccerdk said:
> The functionality i am requesting is currently not a part of the AP

Of course not, what you want is specific to your application. And what you want has been pretty clear from your op. What I am telling you is that the pieces to put together what you want are already available. It's up to you to implement it.
You need the smartconfigStart and smartconfigDone calls, the WiFi.status() call (or variation like checking IP or whatever), a polledTimeout object with its API, and some if logic in the loop(). On sc start you reset the timeout object. If a connection is detected before timeout expires you reset the timeout object. If the connection succeeds, you're done. If a new sc attempt is done before timeout, you reset the timeout again. If nothing happens and timeout expires, or you timeout again after a connection start is detected, it failed.