esp8266/Arduino #7049: ESP8266HTTPClient issue in library

byq13, 2020-01-31:
Message bigger than 1460 bytes can't be send.
ESP8266HTTPClient.CPP

Problem is in:
<pre>
Function: int HTTPClient::sendRequest(const char * type, const uint8_t * payload, size_t size)

// send Payload if needed
        if (payload && size > 0) {
            size_t bytesWritten = 0;
            const uint8_t *p = payload;
            while (bytesWritten < size) {
                int written;
                int towrite = std::min((int)size, (int)HTTP_TCP_BUFFER_SIZE);
                written = _client->write(p + bytesWritten, towrite);
                if (written < 0) {
                     return returnError(HTTPC_ERROR_SEND_PAYLOAD_FAILED);
                } else if (written == 0) {
                     return returnError(HTTPC_ERROR_CONNECTION_LOST);
                }
                bytesWritten += written;
                size -= written;
            }
        }
</pre>

Problematic is: 
`size -= written;` - because SIZE is subtracting every loop and then in WHILE is compared to overal bytes sent. Only first chunk 1460 bytes can be send.

Solution:
`size -= written;` - this should be erased.
`bytesWritten += written;` - this is enough for this function.

Proper code is:
<pre>
// send Payload if needed
        if (payload && size > 0) {
            size_t bytesWritten = 0;
            const uint8_t *p = payload;
            while (bytesWritten < size) {
                int written;
                int towrite = std::min((int)size, (int)HTTP_TCP_BUFFER_SIZE);
                written = _client->write(p + bytesWritten, towrite);
                if (written < 0) {
                     return returnError(HTTPC_ERROR_SEND_PAYLOAD_FAILED);
                } else if (written == 0) {
                     return returnError(HTTPC_ERROR_CONNECTION_LOST);
                }
                bytesWritten += written;
                //size -= written;
            }
        }
</pre>

d-a-v, 2020-01-31:
That is true.
Would you like to make the fix ?

byq13, 2020-01-31:
I am not familiar with github. I am using is as a forum :)
Please if it is not big problem for you, could you fix it?

d-a-v, 2020-01-31:
Someone (a maintainer or any other volunteer) will address this issue very soon.
Please leave it opened, it will be closed when fixed.
This is not a github process but a git process, if you are familiar with the `git` tool.

PR #7051 (2020-02-01): Fixes large payload issues for sendRequest
sendRequest has a major problem when sending a big payload, the comparator in the IF loop has its two operators changed at the same time, so the last part of payload is never sent. This is seen when you try to send a payload that needs more than one operation.

The modification has been tested for a 1500 bytes payload.

*edit from maintainer:* fixes #7049
closed 2020-02-01 by devyte