esp8266/Arduino #7050: ESP8266 sketch upload fails with Failed to connect to ESP8266: Timed out waiting for packet header

Twilight-Logic, 2020-01-31:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				NodeMCU ESP8266
- Core Version:			[2.6.3]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu|Mint]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[Arduino IDE]
- Flash Size: 		[4MB]
- lwip Variant:   [don't know]
- Reset Method:		[???]
- Flash Frequency:	[don't know]
- CPU Frequency:		[don't know]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200)

### Problem Description

When I upload a sketch using the Arduino IDE, with library version 2.6.0 or later I get the following error:

esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

The problem does not occur when library version 2.5.2 is used, or when the upload is perfomed manually. I have tested 2.6.0, 2.6.3 and 3.0 with the following command:

./esptool.py --port /dev/ttyUSB2 write_flash 0x000000 mybinary.bin

I also substituted 3.0 for 2.6.3 in the Arduino library directory, but this did not help. The problem has been reported to me by others, including Windows users, trying to upload to an ESP8266, although the errors tend to vary somewhat..

Full output:

```
esptool.py v2.7
Serial port /dev/ttyUSB2
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: cc:50:e3:55:db:52
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
esptool.py v2.7
Serial port /dev/ttyUSB2
Connecting........_____....._____....._____....._____....._____....._____....._____
Traceback (most recent call last):
  File "/home/johnc/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "/home/johnc/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/esptool/esptool.py", line 2841, in main
    esp.connect(args.before)
  File "/home/johnc/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

Warning: file /home/johnc/.arduino15/packages/esp8266/tools/python/3.7.2-post1/python links to an absolute path /usr/bin/python
```

hasenradball, 2020-02-01:
Hello,
I use Core 2.6.3 on Win10, ubuntu and Mac. Everything is working fine.
I Hand in the past similar probs like you and the root cause was a wrong boards.txt file.
If you use a core 2.6.3 you should use the boards.txt from this core.
Maybe you can douple check this.

Twilight-Logic, 2020-02-01:
The boards.txt file will be whatever the Arduino IDE has installed. I have not changed anything, just installed the library via the Boards Manager in Arduino IDE. One assumes the IDE Boards Manager will do whatever is necessary to install the correct resources. besides, how would I as an Arduino IDE end user know whether a particular boards.txt file is "wrong"? On what basis would I determine that?

jandrassy, 2020-02-01:
the first run of esptool is successful. (this is for the erase, but it only prints informations, because you have Sketch only selected). but then the reset takes too long and the chip doesn't go into flashing mode.
is something connected to io 0, io 2, io 15, EN or reset pin?

Twilight-Logic, 2020-02-01:
No, only TX and RX are connected.

After some further investigation I might be able to shed some more light on the matter. Looking at the output in the Arduino IDE after the upload, it shows that the following command line is being used:

$ python upload.py --chip esp8266 --port /dev/ttyUSB0 --baud 115200 --before no_reset --after soft_reset write_flash 0x0 AR488-wifi.bin

Note that for clarity I have removed the full path to the python and tools directory in the esp8266 library hierarchy. Running with the above command line in terminal duplicates the error condition. Since the error output showed and unexpected reset at the start of the upload, I ran the command again but without the  '--before no_reset' parameter:

$ python upload.py --chip esp8266 --port /dev/ttyUSB0 --baud 115200 --before no_reset --after soft_reset write_flash 0x0 AR488-wifi.bin

This time the upload worked. It seems that this parameter is not used in the version 2.5.2 call which looks like this:

python upload.py --chip esp8266 --port /dev/ttyUSB0 --baud 115200 --trace version --end --chip esp8266 --port /dev/ttyUSB0 --baud 115200 --trace write_flash 0x0 /tmp/arduino_build_299222/AR488-wifi.ino.bin --end

The '--before no_reset' parameter appears in both versions of esptool.py, but does not appear in the version 2.5.2 of the boards.txt.py file, but it does appear in version 2.6.3 of boards.txt.py::

```
$ grep -n "no_reset" boards.txt.py
305:            ( '.menu.ResetMethod.v1.upload.resetmethod', '--before no_reset --after soft_reset' ),
1017:        ( '.menu.ResetMethod.ck.upload.resetmethod', '--before no_reset --after soft_reset' ),
1022:        ( '.menu.ResetMethod.nodtr_nosync.upload.resetmethod', '--before no_reset_no_sync --after soft_reset' ),
1028:        ( '.upload.resetmethod', '--before no_reset --after soft_reset' ),
1036:        ( '.upload.resetmethod', '--before no_reset --after soft_reset' ),
1044:        ( '.upload.resetmethod', '--before no_reset_no_sync --after soft_reset' ),

```

Perhaps this might provide some clue?

jandrassy, 2020-02-01:
the upload.py should run esptool only once since 2.6.1. before it run it twice, first for erase then for upload..
it looks like you have a wrong upload.py for 2.6.1+

Twilight-Logic, 2020-02-01:
Sorry, I don't follow you here? How can I have the "wrong upload.py"? It is being installed by the Board Manager from http://arduino.esp8266.com/stable/package_esp8266com_index.json?

devyte, 2020-02-01:
@Twilight-Logic Please try a new install from scratch for 2.6.3. Don't forget to delete the .arduino15 dir.

Twilight-Logic, 2020-02-01:
Well, I must admit to being sceptical about this suggestion, but tried it anyway. I renamed .arduino15 to .arduino15.bak and re-launched the IDE. My preferences were lost (as expected) so had to be set up again, although this takes just a few moments. I installed the ESP8266 library and also had to update the AVR boards library to bring it up to date. After some 5 minutes that was all done and I tried again. I can report that this time the upload was successful. I guess something must have got out of sync during the update process at some point....

The only drawback to this fix is that if someone has other categories of boards installed then they are all going to have to be re-installed but it doesn't take too long so not a big deal I guess.

Thanks for that suggestion.

Twilight-Logic, 2020-02-01:
It seems that it is sufficient to delete .arduino15/packages/esp8266. This would avoid loosing preferences and having to re-install other board libraries.

devyte, 2020-02-01:
When moving back and forth between versions with a core, the IDE can sometimes mess up something somewhere, and you end up with an install that is a mix of files. This is not unique to us, I've had it happen with other cores. Me, when moving from one version to another in either direction, I prefer to uninstall - clean - install the new one, because paranoia.
Closing.

closed 2020-02-01 by devyte

Twilight-Logic, 2020-02-03:
Made no changes between yesterday and today, but its failing again:

```
esptool.py v2.8
Serial port /dev/ttyUSB3
Connecting........_____....._____....._____....._____....._____....._____....._____
Traceback (most recent call last):
  File "/home/johnc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "/home/johnc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool/esptool.py", line 2890, in main
    esp.connect(args.before)
  File "/home/johnc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 145, in apport_excepthook
    os.O_WRONLY | os.O_CREAT | os.O_EXCL, 0o640), 'wb') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_usr_bin_python3.6.1000.crash'

Original exception was:
Traceback (most recent call last):
  File "/home/johnc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "/home/johnc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool/esptool.py", line 2890, in main
    esp.connect(args.before)
  File "/home/johnc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
```
???