esp8266/Arduino #7052: connection Refused : HTTP error code -1

Vishalkargathara, 2020-02-01:
#### Platform

- Hardware:			         ESP8266 Nodemcu V1
- Development Env:		Arduino IDE
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Nodemcu
- Flash Size: 		        4MB
- Reset Method:		nodemcu
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:             115200

### Problem Description

showing connection refused error.
Below is my code

Nodemcu CODE
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
float volt = 230.5;

void setup () {
 
  Serial.begin(115200);
  WiFi.begin("vishal", "vishal123");
 
  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(500);
    Serial.println("Waiting for connection");
   }
 // check if below pin is necessary!!! ***********************************
  delay(100);
}
 
void loop() {
  if (WiFi.status() == WL_CONNECTED) 
  { 
    Serial.printf("Wifi Connected\n");
    HTTPClient http;  //Declare an object of class HTTPClient
 
    http.begin("http://192.168.43.158:80/pzem/update.php?voltage=" + String(volt));  //Specify request destination
    int httpCode = http.GET();                                                                  //Send the request
 
    //if (httpCode > 0) { //Check the returning code
     Serial.printf("[HTTP] GET... code: %d\n", httpCode);
     
     if (httpCode >0) {
      String payload = http.getString();   //Get the request response payload
      Serial.println(payload);                     //Print the response payload
 
    }
    else
    {
     
      Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
    }
       // http.writeToStream(&Serial);
    http.end();   //Close connection
 
  }
 
  delay(1000);   
}
### Debug image

[https://user-images.githubusercontent.com/35455643/73595076-71ed0580-453a-11ea-84a8-b11362d61355.png]

devyte, 2020-02-01:
The error is pretty clear: the server is refusing your connection for some reason. Either the IP is wrong, or the port is wrong, or the url endpoint is wrong, or the url parameters are wrong, or the http method is wrong, or your server set up (e.g.: firewall) is wrong, or your webserver application (e.g.: webserver config) is wrong, or your php script is wrong. Or something else I'm not thinking of.
I don't see a core issue here. Please seek further support from a community forum like esp8266.com or stackoverflow. If you do reach the conclusion that there is something wrong in the core, please open a new issue and post your investigation results as to why you think so.
Closing.

closed 2020-02-01 by devyte

bkrajendra, 2022-12-04:
I am facing the same issue. with similar code.
It works for the first time, but the subsequent time throws the error:
```
[HTTPS] GET... failed, error: connection failed
```
I am running this code in a timescheduler every 10 seconds.
it only runs one time but for the next iteration, it throws the error.

The exact URL works from the browser. The server is a common PHP server.

Akshaykushawaha, 2023-04-18:
> The exact URL works from the browser. The server is a common PHP server.

Exactly my issue, the website works just fine on my browser but not on my esp8266

bkrajendra, 2023-04-19:
I resolved the issue by using standard coding practice.
In my case, it was due to improperly using a variable buffer in a loop.
using a global variable for the URL and updating it by reference, resolved the issue.

Akshaykushawaha, 2023-04-19:
> Thanks a lot for the quick reply!

codegrue, 2023-12-06:
Wanted to add that I battled this same issue for days while doing SSL connections. What worked for me was adding these properties to the WiFiClientSecure and HTTPClient objects:

```
WiFiClientSecure client;
client.setInsecure(); 
client.setHandshakeTimeout(10);

HTTPClient http;
http.setTimeout(30000);
http.setConnectTimeout(30000);
http.useHTTP10(true);
http.setReuse(false);
http.setFollowRedirects(HTTPC_STRICT_FOLLOW_REDIRECTS);
...
http.begin(client, host.c_str(), port, apiUri.c_str(), useSSL);
```

Akshaykushawaha, 2023-12-07:
Hey, thanks a lot! 
Will try that next time for sure!!