esp8266/Arduino #7056: SNTP: configTime server cannot reference temporary char pointer

mcspr, 2020-02-02:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP8285
- Core Version:			[63b73bf181bfd495ed8d8c45742ed1a117842220]
- Development Env:		PlatformIO
- Operating System:		Linux

### Settings in IDE

- Defaults

### Problem Description
With this piece of code below:
https://github.com/esp8266/Arduino/blob/63b73bf181bfd495ed8d8c45742ed1a117842220/cores/esp8266/time.cpp#L61
If user does something like:
```c++
const String server = retrieveMyServerSettings();
configTime("UTC0", server.c_str());
```
Because lwip only stores a pointer
https://git.savannah.nongnu.org/cgit/lwip.git/tree/src/apps/sntp/sntp.c?h=STABLE-2_1_x#n876
```
void
sntp_setservername(u8_t idx, const char *server)
{
  LWIP_ASSERT_CORE_LOCKED();
  if (idx < SNTP_MAX_SERVERS) {
    sntp_servers[idx].name = server;
  }
}
```
Pointer becomes garbage after some time.

Examples are note affected ofc, since they only use raw strings

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <Arduino.h>
#include <sntp.h>
#include <PolledTimeout.h>

static esp8266::polledTimeout::periodicFastMs poll(1000);

void setup() {
    Serial.begin(115200);
    String value("testtesttest");
    configTime("UTC0", value.c_str(), "domainname");
    Serial.printf("before: 0 -> %s, 1 -> %s\n", sntp_getservername(0), sntp_getservername(1));
    WiFi.persistent(false);
    WiFi.mode(WIFI_STA);
    WiFi.begin("blah", "blah");
}

void loop() {
    if (poll) {
        Serial.printf("1 -> %s\n", sntp_getservername(1));
        Serial.flush();
        Serial.printf("0 -> %s\n", sntp_getservername(0));
        Serial.flush();
    }
}
```

### Debug Messages

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
before: testtesttest domainname
1 -> domainname
0 -> ␔��?�:
1 -> domainname
0 -> ␔��?�:
1 -> domainname
0 -> ␔��?�:

Exception (28):
epc1=0x40207c2e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
```

devyte, 2020-02-08:
You're right, configTime saves a pointer to the received string. It was meant for const char strings, and not for temporary strings. For the rare cases where a configurable server is needed, the user can maintain the lifetime of the string.
Your case is a classic C code gotcha, it's mentioned in books even. I was inclined to just close this, but the internals can be enhanced without too much complexity to cover the use case, and thereby reduce the onus on the user, so marking as enhancement.
How about implementing it yourself and making a pr? Given the current repo workload, and low priority of this, I don't know when we'll get to it, but if you make a pr we can review it and merge sooner.

mcspr, 2020-02-09:
I think my main problem was that it is never stated anywhere, `const char *` as a hint does not really say much since we have other functions for persistent services that accept it but copy internally.
https://github.com/esp8266/Arduino/blob/50a491b7fd7219973c16e6bc3ef5215150bba001/libraries/ArduinoOTA/ArduinoOTA.cpp#L79-L81
https://github.com/esp8266/Arduino/blob/50a491b7fd7219973c16e6bc3ef5215150bba001/libraries/ArduinoOTA/ArduinoOTA.h#L74
Which can be the approach here, either having static String inside setServer(), or static array in time.cpp

Or, considering most common approach with static server string that is never changed, and
- add comment to NTP-TZ sketch
- add comment to configTime
- add reference to configTime to documentation
(which I assume cannot be generated from code comment?)

Which is sort-of like better.

d-a-v, 2022-06-13:
This is still opened.
Is it going to be comments, or an array of Strings in the core ?

mcspr, 2022-06-16:
Could be both, just to retain the ability to hold pointers manually.

```cpp
// use this if your char pointers persist for the duration of the program. implementation detail!
void blah(const char* a, const char* b, const char* c);

// use this if your strings are deleted after this block ends
void blah(String a, String b, String c) {
  static String _a;
  static String _b;
  static String _c;
  _a = std::move(a);
  _b = std::move(b);
  _c = std::move(c);
  configTime(_a.c_str(), _b.c_str(), _c.c_str());
}
```

PR #8606 (2022-06-24): Allow to use temporary strings in configTime
resolves #7056 

in addition to the existing function, adds an overload accepting String
adds warning in the comment. not sure about older versions, but in ide 2.0 this will be displayed when hovering over the function call

although... there may be a confusion between F() and PSTR(). F() will be loaded into a static string, while PSTR() is const char ptr and uses the original function. at least with DNS, there's is a strlen call which will obviously fail
closed 2022-06-24 by d-a-v