esp8266/Arduino #7061: ESP8266 2ch relays Smartconfig app not working

nezmanista, 2020-02-04:
Hello,
first, i bought esp8266 1ch relay with esp-01 and found code which worked ( code is here in STEP 3 https://www.instructables.com/id/ESP0101S-RELAY-MODULE-TUTORIAL/ )

Then i bought ESP8266 2ch relays with esp-01 and when i received it I uploaded the same code from above and nothing happend...then started uploading lot of others code from internet but nothing ( for arduino IDE i am using USB programmer for transferring data from pc to esp-01).

When i saw that noting happend million of times i realized that maybe i have problem with app (esp8266 smart config ) and then notices that on SSID name stands UNKNOWN SSID and that i cannot changed. I tried everything, installing many times but app cannot recognize SSID.

After that i started thinking that maybe i deleted preloader code which was on esp-01 originally when i received it and replaced it with first code i used for 1ch relay.

Please for your help and steps what to do

type of relay
Onboard ESP-01 WiFi modules and high performance microprocessor STM8S103 .

Tech-TX, 2020-02-04:
This area is for issues with the core files or included libraries,  not a "how do I do..."

Please ask over at esp8266.com  or arduino.cc for help with the software.

CristianVladescu, 2020-02-04:
esp8266.com is down btw.

devyte, 2020-02-04:
or at stackoverflow.
Off the top of my head, try a complete flash wipe, then reflash your 1ch bin. 
Closing due to not compliant.

closed 2020-02-04 by devyte