esp8266/Arduino #7072: ESP8266 AP reboots when connected by an OS X device

gimme-esp, 2020-02-10:
### Basic Infos

- [ X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X ] I have tested that the issue is present in current master branch (aka latest git).
- [ X ] I have searched the issue tracker for a similar issue.
- [ X ] If there is a stack dump, I have decoded it.
- [ X ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.6.3]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[4MB/2MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] 

### Problem Description

The ESP runs setup, scans available wifi networks to indicate unused channels, and starts the own AP (code in attachement).

When it is connected by Android or IOS devices - all good, and the ping commands to the IP 192.168.4.1 have no errors.

However, if I connect to the AP using the MacBook Pro, OS X El Captain 10.11.6, The AP crashes continuously and module restarts.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>                                          
#include <FS.h>
#include <ESP8266FtpServer.h>

char* password = "Lea111111";
char* myssid = "Snezhinka";
IPAddress local_ip(192,168,4,1);
IPAddress gateway(192,168,4,1);
IPAddress subnet(255,255,255,0);
ESP8266WebServer HTTP(80);
FtpServer ftpSrv;

bool firstRun = true;
bool serachingNetworks = false;
byte wifiChannels[] = {0,0,0,0,0,0,0,0,0,0,0};

int getMin(byte* temp, int size) {
  int minIndex = 0;
  int min = temp[minIndex];
  
  for (int i=1; i<size; i++) {
    if (temp[i] < min) {
      min = temp[i];
      minIndex = i;
    }
  }
  return minIndex;
}

[... 149 lines cut ...]
0x40100f8c: umm_poison_malloc(size_t) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_poison.c line 158
0x40218ac2: new_linkoutput at glue-lwip/lwip-git.c line 260
0x40100c25: check_poison_neighbors(unsigned short) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_local.c line 71
0x40100ced: umm_malloc_core(size_t) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_malloc.cpp line 422
0x40100bd6: check_poison_neighbors(unsigned short) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_local.c line 59
0x40100bd6: check_poison_neighbors(unsigned short) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_local.c line 59
0x40100ced: umm_malloc_core(size_t) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_malloc.cpp line 422
0x40100f58: umm_malloc(size_t) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_malloc.cpp line 511
0x40100fcd: umm_poison_calloc(size_t, size_t) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_poison.c line 176
0x40100154: ets_post(uint8, ETSSignal, ETSParam) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40212375: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 194
0x40100bd6: check_poison_neighbors(unsigned short) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_local.c line 59
0x40100ced: umm_malloc_core(size_t) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/umm_malloc/umm_malloc.cpp line 422
0x40100154: ets_post(uint8, ETSSignal, ETSParam) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40100154: ets_post(uint8, ETSSignal, ETSParam) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40208ed1: run_scheduled_functions() at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/Schedule.cpp line 123
0x4020681d: __loop_end() at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 167
0x40206830: loop_wrapper() at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 172
0x402066e7: loop_task(ETSEvent*) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 188
0x40206830: loop_wrapper() at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 172
0x40100154: ets_post(uint8, ETSSignal, ETSParam) at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40100175: esp_schedule() at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 113
0x40206859: loop_wrapper() at /Users/michael/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 182

```

martinius96, 2020-05-11:
buy Android.

d-a-v, 2020-05-20:
I tried with 10.13.6 with no issue. 
I removed `#include <...ftp...>` though. Can you try to remove it too ?

devyte, 2020-05-20:
Also just tested with 10.15.4, connected/disconnected multiple times while monitoring heap and uptime, no problems detected.
Closing as can't reproduce.

closed 2020-05-20 by devyte

mafe, 2020-06-04:
I have experienced the same exception(3) issue connecting to the module using different types of smartphones including Android based systems.
Reducing the processor clock rate 160 MHz (overclocked) to 80 MHz (default) fixed it.

Although the issue starter mentions, that it is clocked already at 80 MHz, it might be just a bad module and should be replaced.