esp8266/Arduino #7075: ESP8266WebServer::client() should return a reference

bblanchon, 2020-02-11:
Hi,

Currently `ESP8266WebServer::client()` returns a `WiFiClient` by value, which forbids passing the result directly to a function taking a `Client&`.

For example, it causes a compilation error with [ArduinoJson](https://arduinojson.org):

```c++
serializeJson(doc, server.client());
```

Because [`serializeJson()`](https://arduinojson.org/v6/api/json/serializejson/)  takes a `Print&`, the line above produces the following error:

```
no instance of overloaded function "serializeJson" matches the argument list -- argument types are: (ArduinoJson6141_0000010::StaticJsonDocument<768U>, WiFiServer::ClientType)
```

To fix this problem, `ESP8266WebServer::client()` must return a reference (and not a copy), just as `HTTPClient::getStream()` does.

(see also [this question on StackOverflow](https://arduino.stackexchange.com/q/72433/3430))

Best regards,
Benoit

earlephilhower, 2020-02-12:
Looks like this is a popular request.  At first glance, though, it looks like a breaking API change.  Would you like to do a PR with the change and see if it breaks any of the existing examples, @bblanchon ?  Seems like it would be a simple tweak...

If it does end up breaking things, then it probably need to be pushed to 3.0.  If not, might make it into 2.7...

bblanchon, 2020-02-13:
@earlephilhower , I opened #7077 .
I don't think it should break anything: since `WiFiClient` is copy-constructible, a copy will be implicitly created if needed.

devyte, 2020-02-14:
@earlephilhower @d-a-v I seem to remember this was already tried, and it did break something. Or maybe it was with WiFiServer::available()?

PR #7080 (2020-07-17): Made ESP8266WebServer::client() return a reference (fixes #7075)
*edit from maintainer:* fixes #7075
commit c18f7cb1 (2020-07-17): Made ESP8266WebServer::client() return a reference (fixes #7075) (#7080)
closed 2020-07-17 by d-a-v