esp8266/Arduino #7076: Dreaded ssl->need_bytes error (again)

mischmerz, 2020-02-11:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc]
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			[TYWE3S]
- Core Version:		[2019]
- Development Env:		[Arduino IDE|]
- Operating System:		[Linux]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

We have a number of older devices hard wired. We tested them so that we knew OTA worked before they were installed. A recent attempt to upgrade failed because of ssl->need_bytes error. We tried to install the "old" version from backups to try to figure out what changed, but were unsuccessful .  After reading as much as I could, I understand that this error might be related to the size of the certificate. However - as you see below -  SSL negotiation has already ended as the header of the request is displayed. Only after starting the download are we seeing the SSL error. Additionally - all communications to the devices work well - it's "just" the OTA that gives us trouble. I am asking for possible clues that would help me to flash those hard wired devices as pulling them and serial-flash them would be a lot of work.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Serial.println("Starting UPDATE: " + url);
result = ESPhttpUpdate.update(lumosur_local,8080,url,String(""),String(otaprint));
Serial.print("Update failed - Reason:");
switch(result) {
         case HTTP_UPDATE_FAILED:
          Serial.println(ESPhttpUpdate.getLastErrorString());
           reboot=0;
           break;
case HTTP_UPDATE_NO_UPDATES:
          Serial.println("No Updates available");
           reboot=0;
           break;
case HTTP_UPDATE_OK:
           Serial.println("Unexpected - device should be rebooting");
           break;
default:
           Serial.println("Something weird");

### Debug Messages

```
Debug messages go here
```
ET /ota/1.25/plug-hw8012-14-0-2-13.bin HTTP/1.0
Host: lumosur.local:8080
User-Agent: ESP8266-http-Update
Connection: close
x-ESP8266-STA-MAC: DC:4F:22:4D:BB:62
x-ESP8266-AP-MAC: DE:4F:22:4D:BB:62
x-ESP8266-free-space: 638976
x-ESP8266-sketch-size: 387728
x-ESP8266-sketch-md5: 6dd312f2e9c80143580adad78ad42efd
x-ESP8266-chip-size: 1048576
x-ESP8266-sdk-version: 2.2.1(cfd48f3)
x-ESP8266-mode: sketch

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.0 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Tue, 11 Feb 2020 04:50:38 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Server: Lumosur HA V0.1'
[HTTP-Client][handleHeaderResponse] RX: 'Cache-Control: no-cache, must-revalidate'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: application/binary'
[HTTP-Client][handleHeaderResponse] RX: 'Expires: Sat, 28 Nov 2009 05:36:25 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 383712'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 383712
ssl->need_bytes=8240 > 6859
pm open,type:2 0

earlephilhower, 2020-02-11:
A MCVE needs to be *complete* (the C), and unfortunately your snippet is not compilable.

That said, you are out of memory and using obsolete axtls.  Nothing to be done other than adjust your app to increase free RAM and move to non-obsolete BearSSL.

closed 2020-02-11 by earlephilhower

mischmerz, 2020-02-12:
Though this ticket is closed - I worked all day to install the latest libraries (2.63) including BearSSL. The chip has enough room, but guess what: Same error.
Starting UPDATE: /ota/1.9/plug-hw8012-14-0-2-13.bin

[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 537072
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 1556480
[httpUpdate]  - current Sketch Size: 540096
[httpUpdate] runUpdate flash...
pm open,type:2 0
ssl->need_bytes=8240 > 6859

earlephilhower, 2020-02-12:
`ssl->need_bytes=8240 > 6859` cannot be emitted by code running BearSSL.  The binary that printed your cut-n-paste is running axtls, not BSSL.

mischmerz, 2020-02-12:
You are right oc course, wrong cut&paste :( Sorry.

BUT:
result = ESPhttpUpdate.update(UpdateClient,"https://192.168.1.93:8080/ota/1.9/plug-hw8012-14-0-2-13.bin",otaprint);

Gives me: peekBytes magic header failed
... other overloads result in Permission denied.

wget has no problems:

wget -S --no-check-certificate https://192.168.1.93:8080/ota/1.9/plug-hw8012-14-0-2-13.bin
--2020-02-11 22:36:33--  https://192.168.1.93:8080/ota/1.9/plug-hw8012-14-0-2-13.bin
Connecting to 192.168.1.93:8080... connected.
WARNING: The certificate of ‘192.168.1.93’ is not trusted.
WARNING: The certificate of ‘192.168.1.93’ hasn't got a known issuer.
The certificate's owner does not match hostname ‘192.168.1.93’
HTTP request sent, awaiting response... 
  HTTP/1.0 200 OK
  Connection: close
  Date: Wed, 12 Feb 2020 04:36:33 GMT
  Server: Lumosur HA V0.1
  Cache-Control: no-cache, must-revalidate
  Content-Type: application/binary
  x-MD5: ed3b1669c795b096ac4681dab26abdd8
  Expires: Sat, 28 Nov 2009 05:36:25 GMT
  Content-Length: 537072
Length: 537072 (524K) [application/binary]
Saving to: ‘plug-hw8012-14-0-2-13.bin.2’

earlephilhower, 2020-02-12:
Sorry, but w/o a MCVE there's nothing we can really do on our end.  We do have lots of OTA update users and downstream projects using it, so there's nothing fundamentally wrong.  If you're hitting a corner case, it needs to be reproducible for us to winnow down...

mischmerz, 2020-02-12:
Ok .. this is a stripped down example. Download verified with wget (as shown in a previous post), this OTA attempt results in 

BSSL: read: failed

Serial.begin(115200);
  delay(1000);
  int result;
  int a;
  
  if (WiFi.status() != WL_CONNECTED) {
       while (WiFi.status() != WL_CONNECTED) {
               Serial.println("**Attempting to connect to network");
               WiFi.disconnect();
               WiFi.persistent(false);
               WiFi.begin("Steyla","xxxxxxxxxx",10);
               delay(10000);
               }
       Serial.println("CONNECTED to WiFi");    
       }
       
 delay(1000);
 BearSSL::WiFiClientSecure UpdateClient;
 UpdateClient.setInsecure();

result = ESPhttpUpdate.update(UpdateClient,"https://192.168.1.93:8080/ota/1.9/xplug-hw8012-14 0-2-13.bin");

 Serial.print("Update failed - Reason:");
 switch(result) {
      case HTTP_UPDATE_FAILED:
      Serial.println(ESPhttpUpdate.getLastErrorString());
      break;
      case HTTP_UPDATE_NO_UPDATES:
      Serial.println("No Updates available");
      break;
      case HTTP_UPDATE_OK:
      Serial.println("Unexpected - device should be rebooting");
      break;
      default:
      Serial.println("Something weird");
      }  
}

DEBUG ===============

[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.0 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Wed, 12 Feb 2020 18:41:18 GMT'
:rch 536, 536
[HTTP-Client][handleHeaderResponse] RX: 'Server: Lumosur HA V0.1'
[HTTP-Client][handleHeaderResponse] RX: 'Cache-Control: no-cache, must-revalidate'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: application/binary'
[HTTP-Client][handleHeaderResponse] RX: 'x-MD5: ed3b1669c795b096ac4681dab26abdd8'
[HTTP-Client][handleHeaderResponse] RX: 'Expires: Sat, 28 Nov 2009 05:36:25 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 537072'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 537072

[...]

[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 537072
[httpUpdate]  - MD5: ed3b1669c795b096ac4681dab26abdd8
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 1671168
[httpUpdate]  - current Sketch Size: 424784
[httpUpdate] runUpdate flash...
sleep disable
[begin] roundedSize:       0x00084000 (540672)
[begin] updateEndAddress:  0x00200000 (2097152)
[begin] currentSketchSize: 0x00068000 (425984)
[begin] _startAddress:     0x0017C000 (1556480)
[begin] _currentAddress:   0x0017C000 (1556480)
[begin] _size:             0x000831F0 (537072)
Header: 0xE9 1 3 20
Set flash mode from 0x3 to 0x2
:rd 5, 536, 205
:rdi 331, 5
:rd 326, 536, 210
:rdi 326, 326
:c0 326, 536

BSSL:read: failed

SERVER SIDE =================================

EC:10 2020-02-12 13:18:40  PID:25200 media: Sending buffer (401689)
EC:10 2020-02-12 13:18:40  PID:25200 media: Still keeping data in buffer (393497)

We are using non-blocking sockets so we're sending only the data that the receiver can handle. In this case we try to send 401689 bytes but the receiver would only take 8192 bytes. In which case we return to our select() network function and wait for the queue to get ready again. This unfortunately never happens.

This all indicates that the whole OTA mechanism is extremely fragile and IMHO not very reliable.

mischmerz, 2020-02-12:
Update: I did manage to get the OTA running .. sort of. I programmed a stand-alone server that can use blocking writes and by adjusting a sleep() in between 2048 byte blocks, I was successful uploading an OTA transfer to the ESP-chip - though not consistently. It seems to me that the BearSSL routines are not using any sort of flow control at all and are therefore easily overwhelmed by too much data. However .. sending too few data will also result in an error. 

This is not really useful .. unfortunately.