esp8266/Arduino #7079: 2.6.0 and above won't connect to hidden SSID on channels 12 and 13

sezeryalcin, 2020-02-12:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266EX  (wemos d1 mini)
- Core Version:			Arduino Core 2.6.0, 2.6.1, 2.6.2 and 2.6.3
- Development Env:		Arduino IDE
- Operating System:		Debian 10
### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I am using 1 ESP as AP and another as client to sent signal to the other. This has been working fine on Core 2.5.2 and earlier versions. However when I decided to upgrade to recent 2.6.0 and above, I noticed ESP won't connect to AP ESP when channels 12 and 13 are selected. Also in my case SSID is hidden.

I have an example code on AP ESP where I set the channel to 13. So in this case, client ESP cannot connect to this AP using 2.6.0 and above. When I downgrade to 2.5.2, it works fine.

I also manually tested AP and I was able to connect to ESP on channel 13 via my laptop and ESP worked fine. So issue seems like a restriction on client side.

This caused backwards incompatible code and should be fixed ASAP. Ch 12 and 13 are valid and should be allowed, especially not to cause any compatibility issues.

```cpp
Here is how I set up AP ESP (wemos d1 mini)

AP SIDE code:
----------------------
  IPAddress _apip(192, 168, 50, 1);
  IPAddress _subnet(255, 255, 255, 0);

  WiFi.mode(WIFI_AP);
  WiFi.setSleepMode(WIFI_NONE_SLEEP);
  WiFi.softAPConfig(_apip, _apip, _subnet);

  WiFi.softAP("SOMESSID", "somepsk123", 13, true, 2);   // 12 and 13 doesn't work

Client SIDE code:
----------------------
  WiFi.begin("SOMESSID", "somepsk123");

```

### Debug Messages

```
Debug output shows SSID not found message for ch 12 and 13.
```

devyte, 2020-02-13:
Whether channels 12 and 13 are valid or not depends on your country's laws. There is an api in the sdk that I think is relevant, you set the country code and that sets a policy that sets or restricts things, like the channels allowed. As far as I know, the settings haven't been changed by our core, and when the sdk api was introduced, the default was set to China, with all channels available. However, the setting is also adaptive, and wifi stack will adopt the country code policy of the ap the sta connects to. Please see the nonos api reference doc 2b sections 3.5.74 and 3.5.75 for further info. 
Whatever the case for the settings or restrictions, all of it is handled by the sdk and not by our code. If you want a specific setting that is not the sdk default, I suggest looking into the sdk api usage and setting to your specific needs. However, be aware that if your country does not allow channels 12 and/or 13 and you enable them forcibly, you'll likely be violating your local laws. 
Closing.

closed 2020-02-13 by devyte

devyte, 2020-02-13:
A last note, in case it wasn't clear from the above: previous releases did not enforce the use of country code policy, and so the esp could by default be used in violation of local laws. When the sdk introduced the api and country code policy use, it started to enforce it.

sezeryalcin, 2020-02-13:
@devyte While your comments are helpful, I could guess it was what's going on behind the scenes. So in this case, we need a way to set country code in Arduino Core.

I think document you refer to is 
https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf

And I see referenced function:
bool	wifi_set_country(wifi_country_t	*country)

Also related: https://github.com/esp8266/Arduino/issues/5031

sezeryalcin, 2020-02-13:
My attempt to make 2.6.0+ backwards compatible didn't yield much.
I have following code to revert back to default 13ch config as most of us are not in US and won't be restricted by 11 channels.

-----------------------
wifi_country_t mycountry = 
                           {
                            .cc = "CN",
                            .schan = 1,
                            .nchan = 13,
                            .policy = WIFI_COUNTRY_POLICY_MANUAL,
                           };
  wifi_set_country(&mycountry);
-----------------------

However this did not solve the problem and had to go back to 2.5.2 again. Anybody having a fix here?

This issue should be noted in backward compatibility checklist.