esp8266/Arduino #7082: Can't re-assign an AP address

Twilight-Logic, 2020-02-13:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			2.6.3
- Development Env:		Arduino IDE
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			[qio|dio|other]
- Flash Size: 			1MB
- lwip Variant:   		[v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu] ? probably NodeMCU
- Flash Frequency:		[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   		115200

### Problem Description

The softAP can easily be assigned a default IP address and a HTTP configuration page provided, but if that IP address is subsequently changed (e.g. by the user to suit their own needs) the SSID is no longer broadcast and clients can no longer connect. Switching between softAP and Station mode and works seamlessly as does switching between Station static and DHCP addressing modes. The only issue seems to be changing an IP address when in softAP mode. A power cycle is required to reload the default config after which the SSID is broadcast again and clients can connect.

It may be possible to work around this by saving the config to flash and on boot-up checking whether a previous config has been saved or else loading defaults. Saving the user config is naturally planned for development, but was hoping there would be a way to change IP address "on the fly" so to speak and merely having to reset the softAP stack rather than power cycling the ESP or doing an ESP.reset() which essentially does the same.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

This is the section of my code that sets up the WiFi:

  // Stop webserver
  AR488srv->stop();
  // Disconnect stations (AP mode) or from network (Station mode)
    if (AP.wclient) {
#ifdef DEBUG_1
      Serial.println(F("Station mode>"));
#endif
      if (WiFi.getMode()==2) {
        // Disconnect clients and stop AP mode
        WiFi.softAPdisconnect(false);
        WiFi.enableAP(false);
        delay(500);
      }
      // Station mode
      WiFi.mode(WIFI_STA);
      WiFi.setSleepMode(WIFI_NONE_SLEEP);
      // Static IP address mode
      if (AP.dhcp) {
        // DHCP mode
        WiFi.config(0u, 0u, 0u);
      }else{
        // Static IP mode
        WiFi.config(addr, gate, mask, gate);
      }
      // Connect to AP
      WiFi.begin(AR488srv->arg("essid"), AR488srv->arg("pword")); // Connect to an AP with SSID and password (DHCP address)
//      stat = hasWifiStarted();
    }else{
#ifdef DEBUG_1
      Serial.println(F("SoftAP mode>"));
#endif
      
      if (WiFi.getMode()==1) {
        // Disconnect client from network
        WiFi.disconnect();
      }
//      }else{
        // Disconnect stations from the AP and stop the AP  
//        WiFi.softAPdisconnect(false);
//        WiFi.enableAP(false);
//        delay(500);
//      }
      // Restart and configure the AP      
      WiFi.enableAP(true);
      WiFi.mode(WIFI_AP);
      WiFi.setSleepMode(WIFI_NONE_SLEEP);
      WiFi.softAPConfig(addr, addr, mask); // set IP address
      WiFi.softAP(AR488srv->arg("essid"), AR488srv->arg("pword")); // set SSID and password
      delay(500);
      stat = true;
      WiFi.mode(WIFI_STA);
      WiFi.mode(WIFI_AP);
    }

//      ESP.restart();
startWebServer();

I have attached the full sketch below.

[wifi.ino.zip](https://github.com/esp8266/Arduino/files/4200898/wifi.ino.zip)

### Debug Messages

```
Debug messages go here
```

Twilight-Logic, 2020-02-15:
Looks like I might have found another issue. According to the documentation when calling WiFi.begin():

"Calling it will enable station mode and connect to the last used access point based on configuration saved in flash memory."

While this does work and the cached SSID and password are used, if a static IP address has been configured, the connection defaults to DHCP on re-start (reset button or power cycle).

jandrassy, 2020-03-07:
use WiFi.disconnect() to clear the static IP

devyte, 2020-05-14:
What @jandrassy says is correct. Also, beware reliance on the SDK (re)connection, you can burn your flash write cycles if you don't know what you're doing. It is best practice to use setAutoConnect(false) and persistent(false), and handle the reconnect logic yourself. The only case where you really need a fast startup connect time is in ultra low power apps, such as with coin batteries, in which case you need to do a very thorough and detailed research anyways.

devyte, 2020-05-14:
@Twilight-Logic your sketch above doesn't comply with the "C" in "MCVE". Please edit your post and fix it. 
BTW, the current "M"inimal is ok, please maintain it, i.e.: don't just post your entire application to make it "C"omplete.

earlephilhower, 2020-07-29:
Two months without response for MCVE request.  Closing.

Twilight-Logic, 2020-07-29:
Sorry, but this is the first notification in relation to this issue that I have seen from GitHub in 4-5 months? Now that I have received a notification and have seen the posted updates I will see what I can do to work out a minimal working sketch.

jandrassy, thank you for your suggestions. I will give them a try as well.

devyte, this is not about fast startup time but about being able to change a static IP address when the ESP is in softAP mode. Set a static IP address and then try to change it on the fly. The log will indicate that the new IP address has been applied and the AP attempts a restart but for whatever reason it does not come back up and does not broadcast a SSID. It becomes necessary to re-flash the sketch with a default config at which point it comes up again.

Twilight-Logic, 2020-08-02:
Ok, here is a cut down but complete and working version of the sketch.

`/*
 * SoftAP IP address change issue example
 */

#include <ESP8266WiFi.h>

// Terminators
#define CR 0x13
#define LF 0x10

/***** Parameter variables *****/
#define BAUD 115200		// serial baud rate

/***** Default WiFi AP mode config *****/
const char *dfltSSID = "TestWifi";
const char *dfltPwd = "1234567890";

/***** Parameters saved in EEPROM *****/
union cfgObj {
  struct {
    uint16_t webp;      // Web server port
    uint16_t gpibp;     // GPIB passthrough port
    bool ssl;           // SSL enabled switch state
    bool gpib;          // GPIB passthrough enabled switch state
    bool wclient;       // WiFi station (client) mode enabled
    bool dhcp;          // Station mode DHCP enabled
    uint8_t addr[4];    // IP address
    uint8_t gate[4];    // IP gateway
    uint8_t mask[4];    // Subnet mask
  };
  uint8_t db[20];
};
union cfgObj AP;

/**********************************/
/*****  MAIN PROGRAM  *****/
/**********************************/

void setup() {

  Serial.begin(BAUD);

  Serial.println();
  Serial.println(F("Starting WiFi..."));

  // Set config defaults
  setDefaultCfg();

  // Start WiFi and Web server
  if (startWifi()>-1) {
    Serial.println(F("Started WiFi."));
  }

  Serial.println(F("Press I to restart SoftAP with new IP address"));
  Serial.println(F("Press R to restart the ESP8266"));

}

void loop() {
  uint8_t sByte = 0;

  if (Serial.available()){
    sByte = Serial.read();
    Serial.print(F("Read: ")); Serial.println((char)sByte);
  }

  if ( (sByte==0x49) || (sByte==0x69) ) {  // Option I - restart SoftAP with alternative IP address
    Serial.println(F("Applying address 192.168.1.10 to SoftAP..."));
    restartWiFi();
    Serial.println(F("Alternative IP address applied ?"));
  }

  if ( (sByte==0x52) || (sByte==0x72) ) {  // Option I - restart SoftAP with alternative IP address
    Serial.println(F("Restarting ESP..."));
    ESP.restart();
    Serial.println(F("Shouldn't get here!"));
  }

}

/*****************************/
/***** PROGRAM FUNCTIONS *****/
/*****************************/

/***** Default configuration *****/
void setDefaultCfg() {
#ifdef DISABLE_SSL
  AP = {80,8488,false,false,false,false,{192,168,4,88},{192,168,4,88},{255,255,255,0}};
#else
  AP = {443,8488,true,false,false,false,{192,168,4,88},{192,168,4,88},{255,255,255,0}};
#endif
}

/***** Start WiFi *****/
/*
 * Flag:
 * -1 : fail
 *  0 : fallback mode
 *  1 : Started in AP mode
 *  2 : Started in Client mode
 */
int startWifi() {
  // Start AP mode
  Serial.println(F("Starting AP..."));
  startWifiAP(dfltSSID,dfltPwd,AP.addr,AP.addr,AP.mask);
  return 1;
}

/***** Start WiFi an access point mode ****/
void startWifiAP(String ssid, String psks, uint8_t addr[4], uint8_t gate[4], uint8_t mask[4]) {
  uint8_t gway[4];
  bool gws = true;  // Set gateway to primary address (removes compiler warning)

  // If no SSID or password, then use the currently configured one
  if (ssid == NULL) ssid = WiFi.softAPSSID();
  if (psks == NULL) psks = WiFi.softAPPSK();

  // Make sure station mode is disabled
  if (WiFi.getMode()==WIFI_STA) {
    // Disconnect client from network
    WiFi.disconnect();
    WiFi.enableSTA(false);
  }

  // Gateway is primary address or separate address?
  for (uint8_t i=0; i<4; i++) {
    if (gws) {
      gway[i] = addr[i];  // Gateway set to primary address
    }else{
      gate[i] = gate[i];  // Gateway set to different IP address
    }
  }

  // Restart and configure the AP      
//  if (!WiFi.enableAP()) WiFi.enableAP(true);
  WiFi.enableAP(true);
  WiFi.mode(WIFI_AP);
  WiFi.setSleepMode(WIFI_NONE_SLEEP);
  WiFi.softAPConfig(addr, gway, mask); // set IP address
  WiFi.softAP(ssid, psks); // set SSID and password
  delay(500);
}

/***** Re-start WiFi *****/
void restartWiFi(){
  uint8_t addr[4] = {192,168,1,10};
  uint8_t mask[4] = {255,255,255,0};
  startWifiAP("", "", addr, addr, mask);
}`

I hope this is now compliant with "C" ?

Sending 'i' to change the IP address seems to result in the SoftAP hanging. I am looking for this to restart with the new IP but with the same SSID and password.

devyte, 2020-08-02:
@Twilight-Logic in startWiFiAP() you're receiving the ssid and psk as Strings by value. When you call restartWiFi() you pass in "", "" (i. e. 2 empty strings), yet you compare them against NULL for reassignment. That isn't correct: NULL is essentially a zero uint, so a comparison of a String against it is comparing against "0". Your condition won't be true. 
Because that condition isn't true, you're then calling WiFi.softAP() wih empty Strings, which obviously fails. 
Closing due to user error.

closed 2020-08-02 by devyte

Twilight-Logic, 2020-08-03:
devyte, thank you for taking the time and trouble to respond despite my missing your first request for feedback. A bit embarrassing really as the problem was so obvious I missed it! Changing NULL for "" solves the hanging problem, although when the IP address is changed it still seems to require the wireless connection to the reset, but that's a WiFi/networking issue rather than the SoftAP one. Thanks for casting your eyes over it.