esp8266/Arduino #7083: ESP can't connect on Channel 13 and wifi_set_country() has no effect

sezeryalcin, 2020-02-14:
Basic Infos
[x ] This issue complies with the issue POLICY doc.
[x ] I have read the documentation at readthedocs and the issue is not addressed there.
[x ] I have tested that the issue is present in current master branch (aka latest git).
[x ] I have searched the issue tracker for a similar issue.
[x ] If there is a stack dump, I have decoded it.
[x ] I have filled out all fields below.

Platform
Hardware: ESP8266EX (wemos d1 mini)
Core Version: Arduino Core 2.6.0, 2.6.1, 2.6.2 and 2.6.3
Development Env: Arduino IDE
Operating System: Debian 10
Settings in IDE
Module: [Wemos D1 mini r2]
Flash Mode: [qio|dio|other]
Flash Size: [4MB/1MB]
lwip Variant: [v2 Lower Memory]
Reset Method: [nodemcu]
Flash Frequency: [40Mhz]
CPU Frequency: [160MHz]
Upload Using: [SERIAL]
Upload Speed: [115200]

Problem Description
I upgraded to 2.6.3 from 2.5.2 and now I am not able to connect to APs and other ESPs using wifi channel 13.
Looks like default is restricted to US channels (1-11) but users should be able to change it according to country they are in.

I updated my code to change country and settings but this has no effect.

wifi_country_t mycountry = 
                           {
                            .cc = "CN",
                            .schan = 1,
                            .nchan = 13,
                            .policy = WIFI_COUNTRY_POLICY_MANUAL,
                           };
                              
  wifi_set_country(&mycountry);

Here is how I set up AP ESP (wemos d1 mini)

AP SIDE code:
----------------------
  IPAddress _apip(192, 168, 50, 1);
  IPAddress _subnet(255, 255, 255, 0);

  WiFi.mode(WIFI_AP);
  WiFi.setSleepMode(WIFI_NONE_SLEEP);
  WiFi.softAPConfig(_apip, _apip, _subnet);

  WiFi.softAP("SOMESSID", "somepsk123", 13, true, 2);   // 12 and 13 doesn't work

Client SIDE code:
----------------------
wifi_country_t mycountry = 
                           {
                            .cc = "CN",
                            .schan = 1,
                            .nchan = 13,
                            .policy = WIFI_COUNTRY_POLICY_MANUAL,
                           };
                              
  wifi_set_country(&mycountry);
  WiFi.begin("SOMESSID", "somepsk123");

Debug Messages
Debug output shows SSID not found message for ch 12 and 13.

devyte, 2020-02-14:
@sezeryalcin it seems I wasn't clear enough in #7079. The wifi country policy feature is an Espressif SDK functionality. The API is open and available for use, but we have no control over the internal code. If you found a problem with it, please open an issue at the [Espressif NONOS SDK repo](https://github.com/espressif/ESP8266_NONOS_SDK).
Closing due to not a core issue.

closed 2020-02-14 by devyte

sezeryalcin, 2020-02-14:
yes, it's was clear enough and this is surely a backwards breaking feature which is not mentioned anywhere.
So in that case, what's the use of wifi_set_country() other than calling corresponding SDK? I will check with Espressif repo then.

When you build something on top of something else, underlying issues are still tracked at the upper layer. (e.g gnome bugs on Fedora repo) That doesn't mean you will or should fix the issue. It's a responsibility thing. Other than simple technical support questions, I see a high tendency to close issues in this repo which is not productive. Now you answer says we have this piece of function here, it should do something but if it doesn't, go report to someone else.