esp8266/Arduino #7090: ESP8266WebServerSecure disconnects after the long run

slackwaree, 2020-02-17:
Hello,

Did anyone run into issues with 2.6.3 as well running projects over the long term +1 month that they disconnect from the wifi and will not reconnect?

As I noticed reconnection is handled by the library because if I restart the AP the ESP8266 will reconnect.

devyte, 2020-02-18:
That is most likely due to memory fragmentation resulting in no block available that is big enough for some needed malloc. Try compiling with oom enabled.
BTW, it is never a good idea to have devices without virtual mem running for a long time. I have mine automatically reboot once per night.
This is not the right place to ask such questions. This is an issue tracker, meant for tracking issues in the core. Please refer such questions to a community forum like esp8266.com or stackoverflow. 
Also, issue template ignored. 
Closing due to not compliant.

closed 2020-02-18 by devyte

slackwaree, 2020-02-25:
BTW, it is never a good idea to have devices without virtual mem running for a long time. I have mine automatically reboot once per night.

Hi,

Thanks for your answer. Can you please explain why. This device is for remote control rebooting it once per day is not an option because when rebooted it would turn off the load. Is there anything like a "soft reboot" which does not change the GPIO pins during it's reboot? Even then I don't like the idea rebooting them I have arduinos with 1 year+ uptime working robustly. I do understand that the ESP8266 way more complex but rebooting cannot be a solution.

devyte, 2020-02-25:
I already did: memory fragmentation. The longer the uptime, the more fragmented the heap. It usually plateaus if your app always repeats (i. e. keeps doing the same thing, like reading a sensor and sending to some server), but in general it depends on your application.
Consider e. g. an secure server, where a 16KB contiguous buffer allocation is needed from a e. g. 32KB free heap in an app. That's half of the total heap. Just one alloc passed the heap middle is enough so you can't allocate the buffer, even if the total free mem is enough =>oom. 
Mitigation requires preallocating everything on app start, but that's not how our core works (network stack). 
There are other reasons in addition to fragmentation, such as subtle bugs or subtle mem corruption which are unknown. In a complex device such as this it is beyond difficult to even reproduce them. Example: It took something like two years to figure out the ets_intr_lock thing. 
If your load can't be shut off, then I suggest a latch relay: one pulse for on, another pulse for off. Holding critical devices with level logic is also not a good idea precisely due to reliability concerns.