esp8266/Arduino #7092: After Stage merge #6993 I lost sntp functionality

arendst, 2020-02-19:
Using Tasmota on Arduino core stage after merge #6993 I lost sntp functionality and compilation fails with Undefined References to the once documented SNTP API calls by espressif:

- sntp_get_real_time
- sntp_get_current_timestamp
- sntp_set_timezone

I guess some functionality has been dropped...

For sntp_get_real_time I'll make a workaround saving 1k code space anyway. The others still break compilation

d-a-v, 2020-02-19:
Sorry for that @arendst I focus too much on internals and Arduino API, not espressif API :]
I will re-add these asap.

d-a-v, 2020-02-20:
@arendst `sntp_get_real_time()` returns [this string](https://github.com/esp8266/Arduino/blob/3d128e5c785cbe2096a0def394554d1d8091601d/cores/esp8266/sntp-lwip2.cpp#L371).

`ctime()` would return `Mon Mar  4 21:22:16 2019` so I guess it'd be what you expect, do you confirm ?
I'll propose wrappers mocking espressif API around callers to standard functions.

arendst, 2020-02-20:
@d-a-v confirmed.

PR #7097 (2020-02-22): SNTP: backport espressif sntp api for lwip2
@arendst Can you try it ?

fixes #7092
closed 2020-02-22 by d-a-v