esp8266/Arduino #7095: Deprecate SPIFFS

devyte, 2020-02-20:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

### Problem Description

SPIFFS needs to be deprecated and eventually pulled out. There are known limitations (e.g.: dir and filename) and issues (fs corruption under certain conditions). The original repo hasn't been maintained in a long time, so handling issues can't be done at origin.
We have LittleFS as alternative already implemented. The repo is active maintained, and it supports directories.

TD-er, 2020-02-21:
How well does LittleFS work on really small file systems? For example 64k or 128k filesystem (for 1M nodes)

I only tested it for 16M modules as it was working (a lot) better on larger file systems compared to SPIFFS which is somewhat unusable for > 1M filesystems. (SPIFFS is absolutely unusable for 14M due to taking over 2 sec per file access)

earlephilhower, 2020-02-22:
LittleFS now uses spiffs_blocksize in the boards.txt, which for blocksize=4096 when on 256KB or smaller FSes.  SPIFFS uses spiffs_pagesize = 256b.

So to begin with, you will be able to have significantly fewer individual files.  Performance under load, I have no idea (haven't run anything that small).  Usable file space should not be dramatically different, though, if those 64K are only holding 2-4 configuration files.  It's testable now, just use LittleFS in your app instead of SPIFFS (you'll lose the SPIFFS data in flash, of course).

There's no easy way to adjust this down as LittleFS really wants to be able to manage things in the minimum erase size blocks (4K in our case).

TD-er, 2020-02-22:
> There's no easy way to adjust this down as LittleFS really wants to be able to manage things in the minimum erase size blocks (4K in our case).

I think SPIFFS also needs at least 2 of those blocks with unallocated pages to work properly.
In ESPEasy we normally have very few files (order of 4) so that would not be an issue.

I will make some test builds with LittleFS to see what will happen.
To be honest, I haven't tested it since the last time when I did file some bug reports about it to you :)

earlephilhower, 2020-02-24:
Another potential gotcha is that LittleFS has poor performance when doing update-in-place on a large file.  SPIFFS can do updates in place relatively quickly.

Jeroen88, 2020-02-25:
LittleFS is not (yet) supported on the ESP32. There are people (like me ;)) using the same code for both MCUs. Wouldn't it be logical to deprecate SPIFFS only after LittleFS is supported on the other platform?

TD-er, 2020-02-25:
Hmm that's a valid point. ESPEasy is also running on both platforms.
Although, I could make it selectable at build using defines.

devyte, 2020-02-25:
@Jason2866 @TD-er maintenance on our side for libs that aren't active is a load we're not willing to have.
To be clear, this issue covers deprecation, not pulling out. Think of it as a "first notice". Worst case for you we deprecate for 2.7.0 and pull out for 3.0.0, so even in that case there is time.
You should be using the top level fs class by now, and have very little or no code that uses the low level spiffs api directly. You have three alternatives to pursue, not mutually exclusive:
1. Find a migration strategy. At top level, there is no difference between the twofile systems i. e. same fs class, so api is the same. Then the only question is how to migrate existing spiffs file systems (e. g. copy files somewhere?)
2. Get the "other repos" to implement and support littlefs
3. Find maintainers/developers for the spiffs lib. Having the spiffs code reactivate is the one thing that would stop our deprecation.

Jason2866, 2020-02-25:
@devyte  Project Tasmota does NOT use SPIFFS or LittleFS.

devyte, 2020-02-25:
@Jason2866 sorry, I meant @Jeroen88 :stuck_out_tongue_closed_eyes:

Jeroen88, 2020-02-25:
😰

zekageri, 2020-03-18:
So what's up with esp32 arduino and littlefs? Any news?
I'm currently using my 16mb esp32 with a file system config like 7mb app 7mb ota and 1mb flash.

earlephilhower, 2020-03-18:
@zekageri , suggest you ask that on the ESP 32 repo.  I'm sure they wouldn't mind getting a PR from you for it. :)

PR #7263 (2020-05-04): Deprecate SPIFFS, move examples to LittleFS
SPIFFS has been a great filesystem, but it has significant problems in
many cases (and it's also pretty slow).  Development seems to have
slowed/stopped on the upstream version, and we're not able to provide
support or fix the known issues with it as-is.

Deprecate SPIFFS variable.

Update all examples to use LittleFS instead of SPIFFS.

Also, minor cleanup on very old examples which has obsolete delays
waiting for the Serial port to come up, or which were stuck at 9600 baud
because of their ancient AVR heritage.

Fixes #7095
commit 83166f94 (2020-05-04): Deprecate SPIFFS, move examples to LittleFS (#7263)

* Deprecate SPIFFS, move examples to LittleFS

SPIFFS has been a great filesystem, but it has significant problems in
many cases (and it's also pretty slow).  Development seems to have
slowed/stopped on the upstream version, and we're not able to provide
support or fix the known issues with it as-is.

Deprecate SPIFFS variable.

Update all examples to use LittleFS instead of SPIFFS.

Also, minor cleanup on very old examples which has obsolete delays
waiting for the Serial port to come up, or which were stuck at 9600 baud
because of their anc [... cut]
closed 2020-05-04 by devyte