esp8266/Arduino #7102: Use of Uri in webserver breaks current implementations

TD-er, 2020-02-22:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Breaking commit:			[https://github.com/esp8266/Arduino/commit/4eca62cb530297d98c00f463dba03113811a444e]

### Problem Description

When calling the `WebServer.on(...)` function with a flash string, the compiler refuses to build.

Example error:
```
C:/GitHub/TD-er/ESPEasy/src/WebServer.ino:603:6: error: no matching function for call to 'esp8266webserver::ESP8266WebServerTemplate<WiFiServer>::on(const __FlashStringHelper*, HTTPMethod, WebServerInit()::__lambda4)'
```

Simple work-around is to replace all calls like this with a `String()` wrapper around the flash string.

I guess the best fix would be to add a constructor for Uri with `const __FlashStringHelper*` as parameter.

N.B. some of the derived classes use explicit in the constructor definition, but Uri not. Is this intentional?

devyte, 2020-02-22:
I guess we forgot the forwarder method on(const String & uri,...) that forwards with on(UriGlob(uri),...).
Or we could implement an UriExact class where canHandle() {return true;} and forward to that.

devyte, 2020-02-22:
@Bmooij is there any chance you could help with this? The fix should be easy.

devyte, 2020-02-23:
> N.B. some of the derived classes use explicit in the constructor definition, but Uri not. Is this intentional?

Yes. The explicit in the derived classes is meant to force using them and nlt have an automatic conversion into them, while the lack of it in the base class is meant for exactly that purpose. Which is why the example works.

devyte, 2020-02-23:
@TD-er does #7105 fix your issue?

closed 2020-02-23 by earlephilhower

TD-er, 2020-02-23:
> @TD-er does #7105 fix your issue?

Yes, it does :)

PR #7105 (2020-02-23): Add flash helper constructor to Uri
Fixes #7102