esp8266/Arduino #7103: PR #6887 generates execptions in Project Tasmota

Jason2866, 2020-02-22:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform
-  device                                 |ESP01 ]
- Core Version:			[latest git]
- Development Env:		[Platformio]
- Operating System:		[Windows|Linux]

### Settings in IDE

- Module:  			[|Nodemcu|other]
- Flash Mode:			[DOUT]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Tasmota does generate exceptions when using latest git.

Reverting https://github.com/esp8266/Arduino/pull/6887 does solve the issue.
Without this and using all other PR until to the latest Tasmota does not generate exceptions anymore.

d-a-v, 2020-02-23:
Can you give more details ?
When and what exceptions ?

Jason2866, 2020-02-23:
@d-a-v  exceptions 3 and 9 right after device comes up.

GOOD NEWS. Tried PR #7096 and this seems to solve. Test device now 5 minutes up. 
no exception and responsive as it is with core 2.6.1

So the PR #7096 should be merged ASAP

arendst, 2020-02-23:
Verified too that #7096 fixes this issue.

PR #7096 (2020-02-24): TCP: fixes backlog limit handling, and allows ephemeral TCP connections
When a sketch does not check `server::available()` often enough, short connections are *now* received and stored in background (backlog limit number of connections at most) and can be retrieved and used by subsequent `client = server::available()`. Those clients will be `status() == CLOSED`, `connected() == false` but `available() > 0`.

@kirillandy Could you check this ? I get what's expected by #2569 with these changes.

@jandrassy This is still not what you are waiting for but it would be nice if you checked nothing has [... cut]
closed 2020-02-24 by d-a-v