esp8266/Arduino #7106: ESP8266WebServer args() result too high?

Stefan300381, 2020-02-23:
Hi,

I'd like to understand if the following is a feature or bug:

I pass a few post parameters to a URL to set my WiFi configuration. These are:

- WiFiSSID = a
- WiFiPasswd = b
- DevName = c
- useDHCP =  true

I assumed if I get the number of arguments via httpserver.args() it would return "4", but it returns "5".

The content of the last argument is:
`plain=WiFiSSID=a&WiFiPasswd=a&DevName=a&useDHCP=true`

In the documentation I found:
```
arg - get request argument value, use arg("plain") to get POST body
argName - get request argument name
args - get arguments count
```

So is it expected that the post body gets counted in the args() integer?

best regards

Stefan

d-a-v, 2020-02-23:
Would you have a change to test #6768 ?
(see there for history on that issue)

Stefan300381, 2020-02-24:
well, if you can help me ;-) 
I need to know where I can download the full package and how do install it in my PlatformIO system

d-a-v, 2020-02-24:
* You can try with the [git version](https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version) of this core on which the [PR #6768 must be applied](https://gist.github.com/d-a-v/801cdfcf56d813dca2ff7b23f620c9d3).
* You can try the [alpha release 0.0.2](http://d-a-v.github.io)

devyte, 2020-02-24:
Duplicate of #6435, closing.

closed 2020-02-24 by devyte

Stefan300381, 2020-02-25:
sorry, need some more help. I found this:

https://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

So what do I need to enter under platform_packages =
?

d-a-v, 2020-02-25:
Staging version = git master.
Here I propose to try a pull request which is not yet merged = not in git master yet.
Beside the two methods proposed above (the second is with the official arduino IDE), I can't help much. A PIO specialist may answer (or you'll have to ask them on their forum).

PR #6768 (2020-07-13): webserver: do not count "plain" in argument list
`"plain"` is implicit and is not a real url parameter.
This PR will hide it from the argument counter, but `arg("plain")` is still available.

fixes #6435 

*edit*:

fixes #7106

#5252 added key `"plain"` to the list of arguments. Unfortunately as a side effect, this key is also always counted in the argument list. This PR removes this effect.