esp8266/Arduino #7108: PubSubClient: "Socket error on client <unknown>, disconnecting"

coolcodecding, 2020-02-24:
I have setup mosquitto on docker using certificates to secure the connection. I have no username and password configured in mosquitto. I am using this code to connect my wemos d1 mini to mosquitto with pubsubclient.  
https://github.com/debsahu/ESP_MQTT_Secure/blob/master/ESP8266_MQTT_SSL/Arduino/ESP8266_PubSubClient_SSL/ESP8266_PubSubClient_SSL.ino  
I do not use the secrets.h file, but I tried it before with the same result.  
After inserting my credentials, mqtt data and ca.crt i get the following serial output:  

```
Attempting to connect to SSID: MYSSID ... connected!  
Setting time using SNTP.done!  
Current time: Wed Feb 19 13:53:17 2020 Time: Wed Feb 19 13:53:17 2020  
MQTT connecting ... failed, status code =-2. Try again in 5 seconds.  
MQTT connecting ... failed, status code =-2. Try again in 5 seconds.  
MQTT connecting ... failed, status code =-2. Try again in 5 seconds. 
```

mosquitto gives the following log:  
```
1582138429: New connection from 192.168.0.8 on port 8883.  
1582138434: Socket error on client <unknown>, disconnecting.  
1582138435: New connection from 192.168.0.8 on port 8883. 
1582138440: Socket error on client <unknown>, disconnecting.  
1582138440: New connection from 192.168.0.8 on port 8883.  
1582138443: Client <unknown> has exceeded timeout, disconnecting. 
```
192.168.0.8 is the wemos D1 mini. 
I used the MQTTfx to connect to mosquitto with the ca.crt file before and it worked fine, as well as using the command line with `mosquitto_pub` and `mosquitto_sub` with `ca_cert` successfully.
my mosquitto.conf:  

```
allow_anonymous true  
port 8883  
cafile /ca.crt 
keyfile /server.key 
certfile /server.crt 
tls_version tlsv1.2  
persistence true 
persistence_location /mosquitto/data/ 
```

why does mosquitto not recognise the client name that I have specified in the code? 
Why can I not connect to mosquitto? And how can I solve this?

Tech-TX, 2020-02-24:
MQTT isn't our library,  so I'd ask at esp8266.com or arduino.cc for general how-to stuff.

devyte, 2020-02-25:
Or the pubsub repo.
I suggest looking at the WiFiClientSecure examples as well, it sounds like you have a certificate issue. 
In addition to 3rd party lib, issue template was ignored. 
Closing due to not compliant

closed 2020-02-25 by devyte