esp8266/Arduino #7117: Exceptions in ping callback: using wrong/corrupt stack?

philbowles, 2020-02-27:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			2.6.3
- Development Env:		Arduino IDE 1.8.10
- Operating System:		Windows10

### Settings in IDE

[image]

### Problem Description

Behaviour of ping contradicts espressif docs:  callback parameters incorrect, callback using different stack address from callee, both issues cause crash.

It seems to me that the callback is using a different (or corrupt?) stack frame.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <Ticker.h>
#include <string>

extern "C" {
  #include <ping.h>
}
class pinger {
    std::string ip;
    int   reply;
  public:
    pinger(const char* _ip): ip(_ip){}
    void ping(){
         Serial.printf("ping %s this=%08x @reply=%08x\n",ip.c_str(),this,&reply);
        struct ping_option *ping_opt = NULL;
        ping_opt = (struct ping_option *)os_zalloc(sizeof(struct ping_option));
        ping_opt->count = 1;    //  try to ping how many times
        ping_opt->coarse_time = 1;  // ping interval
        ping_opt->ip = ipaddr_addr(ip.c_str());
        ping_regist_recv(ping_opt,reinterpret_cast<ping_recv_function>(&pinger::reply_recv_cb));
        ping_start(ping_opt);
    }
    void reply_recv_cb(void *arg, void *pdata){ // LINE A
      Serial.printf("reply_recv_cb this=%08x @reply=%08x\n",this,&reply); // LINE B
      uint32_t i=reinterpret_cast<struct ping_resp*>(arg)->ping_err; // LINE C
      Serial.printf("reply=%d\n",i);
    }
  
};

pinger p("192.168.1.31");
Ticker T;

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("Starting WiFi...");

  WiFi.mode(WIFI_STA);
  WiFi.begin("XXXXXXXX", "XXXXXXXX");
[... 129 lines cut ...]
0x40104f29: ets_timer_disarm at ?? line ?
0x40203bd7: run_scheduled_functions() at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/Schedule.cpp line 123
0x40202781: __loop_end at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 167
0x40202794: loop_wrapper() at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 172
0x40100154: ets_post at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 160
0x40104f29: ets_timer_disarm at ?? line ?
0x402027b1: loop_wrapper() at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 179
0x402027a0: loop_wrapper() at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 175
0x40202794: loop_wrapper() at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 172
0x4010506d: ets_timer_arm_new at ?? line ?
0x40100200: millis at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_wiring.cpp line 188
0x40201bca: Ticker::_attach_ms(unsigned int, bool, void (*)(void*), void*) at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\Ticker\src/Ticker.cpp line 49
0x40100908: malloc at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\umm_malloc/umm_malloc.cpp line 511
0x402018bc: ESP8266WiFiSTAClass::status() at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 634
0x402026ce: esp_yield_within_cont at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 100
:  (inlined by) __yield at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 118
0x402011b1: setup at C:\Users\phil\Desktop\pingmcve/pingmcve.ino line 44
0x40100154: ets_post at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 160
0x40100175: esp_schedule at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 113
0x402027bd: loop_wrapper() at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/core_esp8266_main.cpp line 182
```

---

Secondly it can be seen from the output at "LINE B" that the address of the class member "reply" is different inside the callback function from that of the run function. Unsurprisingly, any attempt to access member variables from within the call back causes an exception, since the wrong address is accessed.

earlephilhower, 2020-03-02:
@philbowles, you are passing in a C++ calling ABI method to one expecting a C one.  That's probably your issue.

The callback must be a `static` class member or else it will expect to have the 1st parameter as the `this` pointer.

i.e. `static void reply_recv_cb(void *arg, void *pdata){`

Or, a separate external C (non-class member!) can be passed as the callback fcn, and then you can bounce from the C to a global object pointer->callback.

philbowles, 2020-03-03:
@earlephilhower  That makes a lot of sense. What worries me is that I took the code originally from a "working" library, also using a class method...

closed 2020-03-03 by philbowles

earlephilhower, 2020-03-04:
As long as the method doesn't *use* any object-local variables/methods, then it could work just fine if the microcontroller's ABI doesn't place the `this` ahead of the parameters.  The 8266 uses an ABI where the first few params are passed in registers, but maybe the AVR uses a stack and pushes the this before params, so that extra param doesn't change offsets for the remaining params.

It's still *wrong*, of course.  But it *could* run on that one specific ABI.  That library should really be fixed...