esp8266/Arduino #7120: ESP8266 reboots whenever I run wifi.begin()

spencerkraisler, 2020-02-27:
##Platform

- Hardware:	                        esp8266
- Core Version:			-
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			       Wemos D1 Mini Lite
- Flash Mode:			-
- Flash Size: 		               1MB
- lwip Variant:                       v2 lower memory
- Reset Method:		       -
- Flash Frequency:	       -
- CPU Frequency:		      80 MHz
- Upload Using:		      Serial
- Upload Speed:                  921600

### Problem Description
Hello. I am currently using the Wemos D1 Mini Lite in a camera setup. For the past few weeks, it was working fine. I was simply taking a photo and uploading the photo to the esp8266 server in AP mode.

Now it is not working. Whenever I try to connect to wifi, it constantly reboots. I targeted the method WiFi.begin() as being the main thing that causes the esp to constantly reboot. I've ran simple scripts like Blink, and it works fine. It seems to reboot only when I include the WiFi.begin(). 

I believe I may know what happened. I accidentally uploaded a script to it with the board as Arduino Uno rather than the Wemos D1 Mini Lite. I feel this may have been the cause, since the board was working fine before that fateful accident.

I am a bit inexperienced with things, so I tried to fill in the fields above as best as I can. I am happy to supply any info needed. Thank you.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
#include <ESP8266WiFi.h>        // Include the Wi-Fi library

const char* ssid     = "University of Washington";         // The SSID (name) of the Wi-Fi network you want to connect to
const char* password = "";     // The password of the Wi-Fi network

void setup() {
  Serial.begin(115200);         // Start the Serial communication to send messages to the computer
  delay(10);
  Serial.println('\n');

  WiFi.begin(ssid, password);             // Connect to the network
  Serial.print("Connecting to ");
  Serial.print(ssid); Serial.println(" ...");

  int i = 0;
  while (WiFi.status() != WL_CONNECTED) { // Wait for the Wi-Fi to connect
    delay(1000);
    Serial.print(++i); Serial.print(' ');
  }

  Serial.println('\n');
  Serial.println("Connection established!");  
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());         // Send the IP address of the ESP8266 to the computer
}

void loop() { }

```
### Debug Messages

This is what is printed to serial monitor.
```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld

Exception (3):
epc1=0x40100710 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4006e9a9 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb90 end: 3fffffc0 offset: 01a0
3ffffd30:  feefeffe feefeffe feefeffe feefeffe  
3ffffd40:  feefeffe feefeffe feefeffe 3ffffef0  
3ffffd50:  0000049c 0000049c 00000020 40100900  
3ffffd60:  feefeffe feefeffe feefeffe feefeffe  
3ffffd70:  00000002 400042db 000000fd 40100b58  
3ffffd80:  40004b31 00001000 000000fd 40100274  
3ffffd90:  40105ae0 feefeffe feefeffe 4022db4d  
3ffffda0:  40105c9d 4022dc37 3ffef144 0000049c  
3ffffdb0:  000000fd 3ffffef0 3ffef144 4022dc1a  
3ffffdc0:  ffffff00 55aa55aa 0000000e 00000020  
3ffffdd0:  00000020 00000078 00000012 aa55aa55  
3ffffde0:  000000ff 4022e11a 3ffef144 3ffef144  
3ffffdf0:  000000ff 000000f2 000000f2 40100640  
3ffffe00:  40105c9d 00000001 3ffef154 4022e33a  
3ffffe10:  00000005 3ffef144 000000ff 3ffffef0  
3ffffe20:  3fffff10 3ffef17b 0000000e 00000020  
3ffffe30:  3ffef204 3fffff51 00000001 4022e3ea  
3ffffe40:  3ffffef0 40239920 00000000 00000000  
3ffffe50:  3ffef544 3fffff10 3fff547c 4022e3b9  
3ffffe60:  3ffef144 4022e420 3ffe84d4 3ffe8655  
3ffffe70:  40201976 3ffe8655 3ffe8654 402018cb  
3ffffe80:  76696e55 69737265 6f207974 61572066  
3ffffe90:  6e696873 6e6f7467 3ffffe00 40204cbc  
3ffffea0:  3ffee400 00000000 3ffffe60 40204ef9  
3ffffeb0:  00000f98 000001f3 000001f3 40100640  
3ffffec0:  000006e8 000000dd 00000014 3ffeebc4  
3ffffed0:  007a1200 44e0a632 00000000 401008cb  
3ffffee0:  adb5c801 3f20b34c 00000046 00000100  
3ffffef0:  76696e55 69737265 6f207974 61572066  
3fffff00:  6e696873 6e6f7467 feefef00 feefeffe  
3fffff10:  3ffe8600 00000000 6e000a0d 402032c5  
3fffff20:  3ffe8676 00003d37 00000000 00000000  
3fffff30:  007a1200 9ee97333 00003d00 3ffee3f4  
3fffff40:  40201a8c 3ffee37c 3ffe8674 40201a98  
3fffff50:  40201a00 3ffee37c 3ffe8681 00000000  
3fffff60:  00000004 00000000 3ffee348 00000001  
3fffff70:  40202a21 0000000a 3ffee3f4 3ffee3f4  
3fffff80:  3ffee348 3ffe84d0 3ffee37c 4020107a  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  3fffdad0 00000000 3ffee3b4 40202574  
3fffffb0:  feefeffe feefeffe 3ffe84f0 40100b8d  
<<<stack<<<
```

This script does not crash. But I get that message constantly in the serial monitor.

devyte, 2020-02-28:
This is a "how do I..." (fix it) type question. And the stack trace wasn't decoded, only you can decode your own stack. 
The crash looks to me like a failure to boot. Most likely you have some bad parameter in the IDE, or a corrupted install. I suggest you delete everything (don't forget the arduino15 dir) and do a clean install. 
Also, did you wipe the entire flash?
For general support please refer to a community forum like esp8266.com or stackoverflow.
Closing due to not compliant.

closed 2020-02-28 by devyte

mmillet, 2020-05-13:
> This is a "how do I..." (fix it) type question. And the stack trace wasn't decoded, only you can decode your own stack.
> The crash looks to me like a failure to boot. Most likely you have some bad [...]

I got the same problem, and finally found Crystal Frequency is wrong (should be 26 MHz when using esp01s). Thanks a lot!

TekMaker, 2023-04-23:
I spent a couple of hors figuring this out, have you ported other code from another chip maybe an Arduino? I found I was programming pins that I shouldn't and it was that which was resetting the ESPs.