esp8266/Arduino #7123: Placing interrupt function after setup() -> "error: too many decimal points in number"

MagnusThome, 2020-02-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			Wemos D1 mini
- Core Version:		2.6.3
- Development Env:	Arduino IDE
- Operating System:	Windows

### Settings in IDE

[image]

### Problem Description

If the interrupt function is declared or defined before setup() all works as it should. But if the interrupt function is defined further down in the code (and no declaration at the top) it will not compile and I get this error message:
```
est2:15:61: error: too many decimal points in number
 void ICACHE_RAM_ATTR irqfunc () {
                                                             ^
test2:15:42: error: request for member 'iram' in '""', which is of non-class type 'const char [1]'
 void ICACHE_RAM_ATTR irqfunc () {
                                          ^
test2:15:6: error: section attribute not allowed for 'void irqfunc()'
 void ICACHE_RAM_ATTR irqfunc () {
      ^
exit status 1
too many decimal points in number
```

I can't find any info on this, shouldn't the compile error message rather say function xxx not declared in this scope? It really isn't a very helpful error message, especially since it's quite common in the Arduino environment not to declare functions and just define them further down in the code?

# DOES NOT COMPILE
```cpp
#define GPIO 5
volatile bool irqdetected = false;

void setup(void) {
  Serial.begin(115200);   
  pinMode(GPIO, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(GPIO), irqfunc, FALLING);
} 
void loop(void) {
  if (irqdetected) {
    Serial.println("irq");
    irqdetected = false;
  }
}
void ICACHE_RAM_ATTR irqfunc () {
  irqdetected = true;
}
```

# COMPILES OK
```cpp
#define GPIO 5
volatile bool irqdetected = false;

void ICACHE_RAM_ATTR irqfunc () {
  irqdetected = true;
}
void setup(void) {
  Serial.begin(115200);   
  pinMode(GPIO, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(GPIO), irqfunc, FALLING);
} 
void loop(void) {
  if (irqdetected) {
    Serial.println("irq");
    irqdetected = false;
  }
}
```

# COMPILES OK
```cpp
#define GPIO 5
volatile bool irqdetected = false;

void ICACHE_RAM_ATTR irqfunc ();

void setup(void) {
  Serial.begin(115200);   
  pinMode(GPIO, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(GPIO), irqfunc, FALLING);
} 
void loop(void) {
  if (irqdetected) {
    Serial.println("irq");
    irqdetected = false;
  }
}
void ICACHE_RAM_ATTR irqfunc () {
  irqdetected = true;
}
```

philbowles, 2020-02-28:
This is a function of the way arduino IDE "simplifies" things for newbies, it _almost_ removes the requirement to declare a function before its definition which is a core feature of c/c++.

The simple solution is to include the definitions ahead of time, or include forward references.

I am not connected / affiliated to the project in any way, I just use it a lot, but this isn't a bug or an issue, it is simply a function of the way the IDE works. It is discussed / explained in many places, google "arduino forward reference".

Hope this helps.

earlephilhower, 2020-03-02:
Thanks, @philbowles .  Looks like you've provided a good description of the underlying issue (Arduino's finicky "auto-generate-prototypes" for people who don't know C) and the solution.  Closing.

closed 2020-03-02 by earlephilhower

MagnusThome, 2020-03-02:
I know very well how declaration and definition works, that's how I figured it out. But it still took me quite a lot of time to figure it out since it's an anomaly in the Arduino IDE to work this way (99% of the time users don't need to declare functions and they don't do it). So when instead of getting a helpful "out of scope" error message you get a "too many decimal points" which is straight up a bizarre message for the actual problem for any normal programmer. 

If some links with info on this particular case of esp + irq + Arduino out of scope problem popped up when googling "too many decimal points" I wouldn't have posted here at all....

The error message is simply badly chosen and very unhelpful.

´

earlephilhower, 2020-03-02:
Unfortunately, that's Arduino IDE related.  I suggest you look at their repo here on GitHub and file an issue about this there.  From the 8266 core, there's nothing we can do here.

MagnusThome, 2020-03-02:
Ah, okay, my bad. 

I thought since this was called "ESP8266/Arduino" and not "core"... Easy to get lost on github hehe.

MagnusThome, 2020-03-02:
Now you DO find this thread if you google "esp8266" and "too many decimal points"

Mission accomplished  ;-)