esp8266/Arduino #7124: OTA broken after commit #7036

MarkyAD, 2020-02-28:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					nodemcu
- Core Version:			e6decacd0a03a61f00068264b024e7
- Development Env:		Makefile
- Operating System:		Linux

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		        4M
- lwip Variant:                 v2 Lower Memoru
- Reset Method:		nodemci
- Flash Frequency:	       40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		OTA
- Upload Speed:              115200

### Problem Description

@d-a-v @devyte After commit #7036 OTA no longer works, it just hangs. I traced it back to this specific commit.

d-a-v, 2020-02-28:
I tried the OTALeds (ArduinoOTA lib using UDP to negociate) example sketch and it's still working.
Can you give more details on your setup ?

MarkyAD, 2020-02-28:
I have several nodemcu and d1_mini_pro and they all had the same symptom, all refused authentication. I do all my own dns and dhcp services. Once I reverted my git repo back before that commit, OTA worked again on all of them. Host environment is x86_64 and arm (both 32 and 64).

MarkyAD, 2020-02-29:
Just tried again on a pi3, same thing but this time espota.py spills "No Answer". I was beginning to think it was a linux python3 issue before I tried reverting my repo. This time on the pi with a board currently flashed with a pre #7036 sketch, ota works the first time, then ota hangs after being flashed with a post #7036 sketch.

Build environment is plain Makefiles and/or Makefiles with arduino-builder (1.8.12).

d-a-v, 2020-02-29:
Ah I didn't try with auth.

PR #7132 (2020-03-04): udp: fix again pbuf management
This follows #7036 where udp pbuf chain handling is made hairy because of interleaved information fragment introduced with IPv6.

@szekelyisz Could you please test this change ?
fixes #7124
closed 2020-03-04 by d-a-v