esp8266/Arduino #7125: esptool.FatalError: Failed to connect to ESP8266: invalid head of packet(0xf0)

rajatsaini736, 2020-03-01:
### Basic Info

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

i have an uno board, esp, breadboard, jump wires....and installed esp libraries in arduino ide.
But when i upload the code while selecting tools->Board->generic ESP8266.. it gives me the error ->esptool.FatalError: Failed to connect to ESP8266: Invalid head of packet (0xF0)

MY SKETCH.....

void setup() {

}

void loop() {

}

```

### Debug Messages
Executable segment sizes:

IROM   : 227372          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 26368   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 

DATA   : 1252  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 672   ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 24848 )         - zeroed variables      (global, static) in RAM/HEAP 

Sketch uses 255664 bytes (26%) of program storage space. Maximum is 958448 bytes.
Global variables use 26772 bytes (32%) of dynamic memory, leaving 55148 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM11
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\Rajat Saini\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/Rajat Saini/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2890, in main
    esp.connect(args.before)
  File "C:/Users/Rajat Saini/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Invalid head of packet (0xF0)
esptool.FatalError: Failed to connect to ESP8266: Invalid head of packet (0xF0)

rajatsaini736, 2020-03-01:
I tried this error 1000 times on google but no-one gives me the best solution.

Tech-TX, 2020-03-01:
This isn't a problem with the core, it's a problem between (IDE settings) >> (driver for the module) >>  (your module) for whatever kind of module you have.  The IDE can't talk to the ESP8266 module.  You probably shouldn't use the 'generic' module unless you know what settings you need for that specific module.  You'll have better luck asking that question over at http://esp8266.com

You've completely ignored the template, so there's not enough information to figure out what you've done incorrectly.

devyte, 2020-03-01:
Previous comment is correct. This is an issue tracker meant for tracking issues in the core. For general support questions please refer to a community forum like esp8266.com or stackoverflow.
Closing due to not compliant.

closed 2020-03-01 by devyte

Badeshka1, 2020-04-25:
> ### Basic Info
> * [ ]  This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
> * [ ]  I have read the documentation at [...]

I have the exact same Problem. Have you found a solution?

Rahuldakuri, 2020-04-25:
Check the onboard components of the esp module there might be a possibility of  minor short circuit it happens when we usually jump it with the UNO board due the voltage parameters

Badeshka1, 2020-04-25:
Hmm Im using a voltage reducer regulator to step down the 5V from the Arduino Uno to the 3.3V it needs. 

Im also getting the Same Error but with a (0x08) and not the (0xF0) anymore. Do you know what the difference between these two is?

Thanks for the quick reply!

Badeshka1, 2020-04-25:
Also all the AT+ commands work. I can connect to my wifi network and get the IP ETC. I just cant upload code...?

nitrame, 2020-04-25:
hi guys, please i have this same error of invalid head of packet (0x90). i am using an ESP-12F module on an arduino UNO. i removed the AtMEga I.C from the arduino and connected tx to tx from the esp to UNO. all i see when i open serial monitor is a lot of rubbish. and the blue LED on the ESP keeps blinking fast..... SOMEBODY HELPPP!!!

devyte, 2020-04-25:
This is an issue tracker, meant for tracking issues in the core. It's not meant for general support or help requests. Please seek such help at a community forum like esp8266.com or stackoverflow.

Rahuldakuri, 2020-04-28:
Well this might sound sarcastic but did you check the port declaration on the ide before uploading the code I just had the doubt

tedhts, 2020-04-30:
I had the same error.  Anyone get this to work?

Badeshka1, 2020-04-30:
For the the Problem was caused because I didn't have the reset pin connected to ground the Arduino uno board. Once you connect that the Arduino knows that it just needs to pass through the code.

Hope it helps!

tedhts, 2020-05-01:
Wow, thank you!  That's exactly what I needed.  I uploaded the code with no problem.

teenkevo, 2020-06-03:
> For the the Problem was caused because I didn't have the reset pin connected to ground the Arduino uno board. Once you connect that the Arduino knows that it just needs to pass through the code.
> 
> Hope it helps!
> 
> > 
> > 
> > I had the same error. Anyone get this to work?

Hi, I am trying to understand this comment but i'm failing to yet it seems like the solution lies here. Can you reach out please. Thanks

Badeshka1, 2020-06-03:
The reason I was having the problem is because I didn't have the reset pin on the Arduino uno connected to ground

25b3nk, 2020-08-01:
@Badeshka1 Can you share the PIN configuration you used while uploading the code for generic ESP8266 module ? I am still facing issues.

tedhts, 2020-08-01:
> @Badeshka1 Can you share the PIN configuration you used while uploading the code for generic ESP8266 module ? I am still facing issues.

1. GND -> GND
2. TX -> TX
3. RX -> RX
4. VCC -> VC

It's counterintuitive because you'd assume TX would go to RX on the other side.  It doesn't.  Also, be very sure that you're only applying 3.3V - not 5V.  If you're using Adafruit FTDI Friend, there are solder points on the back to send 3.3V on VCC.

Badeshka1, 2020-08-02:
All the pins you posted are correct. maybe try swapping the TX and RX. Most
importantly you need to have the reset port on the arduino connected to the
Ground on the Arduino. This tells the Arduino to just send the Code on to
the ESP.

Hope it helps

danieleasteggiante, 2020-12-11:
Reset of Arduino connected to ground, 
Rx arduino -> Rx esp
Tx arduino -> Tx esp
gnd arduino -> gnd esp
3.3v arduino -> vcc esp
3.3 v arduino ->chpd esp
gnd arduino -> GPIO 0

while upload the program from Arduino to Eps on UART, you need connect to 3.3 RESET esp for one second and GPIO 0 to ground, after that,  disconnect the reset pin from 3.3 and wait the program finish upload. After that disconnect and reconnect the vcc.

For me it works.

IceWalnut, 2021-11-23:
> Check the onboard components of the esp module there might be a possibility of minor short circuit it happens when we usually jump it with the UNO board due the voltage parameters

Thanks!

ParkpoomR, 2021-12-14:
I try with
1. un supply : ESP-01 from Arduino 3.3V
2. re Supply : ESP-01
3. Jump GPIO-2 to GPIO-0 to GND
4. Upload Code (with out RST)

pau1phi11ips, 2022-02-08:
Ended up here after searching for `esptool.FatalError: Failed to connect to ESP8266: Invalid head of packet (0xF0)`

2 hours later, I've realised the IR hat I had on the D1 mini is the wrong way around 🤦 Worth double checking this peeps.

musapinar, 2022-04-24:
For people coming via search engines:
I had this issue when I was using long Dupont wires from the USB FTDI to the ESP.
The issue vanished as soon as I soldered thicker wires (22 AWG). 
Size does matter.
Hope it helps, in case anything above didn't solve it.

vanisridhar102, 2022-06-14:
> 

same problem AT only comments working for me. Did you solved the problem??

JamieShelley, 2022-10-04:
> 1. GND -> GND
> 2. TX -> TX
> 3. RX -> RX
> 4. VCC -> VC
>
> It's counterintuitive because you'd assume TX would go to RX on the other side. It doesn't. Also, be very sure that you're only applying 3.3V - [...]

I had a similiar issue here. For some strange reason: If I have rx/tx connected to a peripheral. I can't program the board at all.
I'm not sure why.

afaroo01, 2022-10-15:
I am still having the same issue even after following the above solutions. May be its a hardware issue?

Bighoneypot, 2022-10-15:
You use FTDI CH340?

logichex, 2023-03-20:
Hey all, I just created a github account for those facing the issue mentioned. It's definitely not a core issue, as someone mentioned, high quality cabling is required due to interference etc. I tried everything possible with my ESP01 (which is possibly the bare minimum it can get=read as minimum immunity towards interference). 
**I reduced the upload speed to 57600 and it worked**
Can not of course promise but for those who tried everything and failed, give this a try. Also checked without a level converter and it fails at 115.2k but with 57.6k it uploaded.

MaxShcheglov, 2023-03-28:
> Hey all, I just created a github account for those facing the issue mentioned. It's definitely not a core issue, as someone mentioned, high quality cabling is required due to interference etc. I [...]

Thank you very much, it helps!

Note: if upload is stuck with the phrase "Hard reset via RTS pin..." - just reset using reset button.

gewmag, 2023-04-27:
To anyone reading this that has the same problem, 
Reset of Arduino connected to ground,
Rx arduino -> Rx esp
Tx arduino -> Tx esp
gnd arduino -> gnd esp
3.3v arduino -> vcc esp
3.3 v arduino ->chpd esp
gnd arduino -> GPIO 0

while upload the program from Arduino to Eps on UART, you need connect to 3.3 RESET esp for one second and GPIO 0 to ground, after that, disconnect the reset pin from 3.3 and wait the program finish upload. After that disconnect and reconnect the vcc.

And it's important to connect the usb cable of the board directly to the pc and not use a usb hub etc because it can't upload the code correctly

zero-one-soft, 2023-07-31:
I have the CH340E module to connect to the ESP828. 
had this issue that when I upload my  data it would give invalid head of packet error.

What did I do to fix it? Pull out some hair then I did the following.
On the ch340 you have a DTR pin ( Data Terminal Ready) and I just connected that to the Reset pin of the ESP and everything works fine now. I have my pin outs as follows 

CH340 -> ESP
GND -> GND
TXD -> RXD
RXD -> TXD
DTR -> RST

I think what happens is when you pull down IO0 to ground it probably pulls down the RST pin that gives the signal to the DTR pin and then everything works OK

TobiasKoller, 2023-08-06:
this was working for me:
just add
  board_upload.resetmethod = nodemcu
to the platformio.ini

https://community.platformio.org/t/my-esp8266-it-s-having-an-error-to-connect-when-i-write-his-code-a-fatal-error-occurred-failed-to-connect-to-esp8266-timed-out-waiting-for-packet-header/30034/11

vedantjoshi96, 2025-02-18:
> ### Basic Info
> * [ ]  This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).[ ]  I have read the documentation at [...]

Just disconnect all signal pins from NodeMCU. It will start working.