esp8266/Arduino #7126: Cannot make mdns work again on both interfaces with 2.6.3

mariusstaicu, 2020-03-02:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12
- Core Version:			2.6.3
- Development Env:		platformio
- Operating System:	        Ubuntu

### Settings in IDE

- Module:  			Wemos D1 mini
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   v2 Higher Bandwidth
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		OTA or serial
- Upload Speed:   115200

### Problem Description

I'm trying to restore mdns functionality in our firmware like it worked with prior arduino core 2.5.2 - ap working on both sta and ap interfaces.

I noticed there is no way to make mdns responder work on both interfaces correctly unless I create two separate responders, one for AP and one for STA. I start one at boot  and after that on every wifi mode change, I start the other and close the current one. This is working in my MCVE, however it in my production firmware it causes device to restart whenever the mdns reponder is started for the second time. The esp resets iself whenever reaching the _addService or enableArduino_ method calls.
I have tried everything and I am creating this issue to see if there is something I am doing wrong.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

I'm compiling the sketch with -DNO_GLOBAL_MDNS

```cpp

    #include <ESP8266WiFi.h>
    #include <ESP8266WebServer.h>
    #include <ArduinoOTA.h>
[... 315 lines cut ...]
3fffff10:  2f70612f 62616e65 8a00656c 0000002f  
3fffff20:  8000652f 00000000 401007de 40226bb1  
3fffff30:  00000001 00000001 00000000 3fff15c0  
3fffff40:  00000003 00000001 3fff0380 40214a84  
3fffff50:  3fff0180 3fff6568 3fff654c 3fff1f4c  
3fffff60:  00000001 3fff1534 3fff1510 4020a2cb  
3fffff70:  3fffdad0 00000000 3fff0380 3fff1f4c  
3fffff80:  3fffdad0 00000000 3fff02e8 40221a8a  
3fffff90:  3fffdad0 00000000 3fff1f0c 4020a398  
3fffffa0:  feefeffe feefeffe feefeffe 40226c5c  
3fffffb0:  feefeffe feefeffe 3ffe8554 40100349  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld

```

The main question remains: Am I doing something wrong in the MCVE ? How am I supposed tot make mdns work again on both interfaces on 2.6.3. I also tried to start responders only in the beginning, however, if I boot in AP mode, I cannot know the STA interface IP so I can start it.

mariusstaicu, 2020-03-02:
I found the problem from the production firmware, it was related to deviceName string which was an instantce variable and was passed by reference in the firmware. Because of that, it got deleted when I was doing MDNS.close() so when I did the next call of MDNS begin, the hostname was NULL. That cause things to blow up when I tried doing addService since that goes and does strcmp on null hostname string. 

However, the initial question still stands. Is it correct to use two different responders and initialize them every time the Wifi interfae changes ? Also, is it correct to close the other responder every time ?

d-a-v, 2020-03-02:
Thanks for the update.
It is correct to use two different responders, one for each interface.
One responder currently cannot work with more than one interface.
If only one interface is used at a time, the same responder *could* be used: `::close()` should be called first, then `::begin(...)` with the new address in parameters (the address of the chosen interface).

mariusstaicu, 2020-03-02:
Thanks for the response. So if I use two different mdns instances, do I still need to close and do ::begin every time interface changes ?

d-a-v, 2020-03-02:
If the IP address changes, then I believe so (close when interface is down, begin when it is up again).

mariusstaicu, 2020-03-02:
Thanks, it really helped me a lot.

closed 2020-03-02 by mariusstaicu

devyte, 2020-03-02:
If an IP changes, the MDNS instance should get notified and handle it automatically. See the wifi event handlers assigned in the begin() method.
Also, the wifi handler objects are class members, so it should work per instance without interfering with each other.