esp8266/Arduino #7130: Exception 29 ESP8266

AHakan, 2020-03-03:
- Hardware:					[ESP-01]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] 

### Problem Description
Exception 29

### Debug Messages

```
Exception (29):
epc1=0x40220657 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000019 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffecb0 end: 3fffffb0 offset: 01a0
3fffee50:  3fffee70 40220793 3ffecf6c 5c056537  
3fffee60:  4023b2ea 0000000f 00000001 00000006  
3fffee70:  00000018 3ffeeec4 00000004 3ffed6b0  
3fffee80:  3ffe87f4 3ffe87f4 3ffef2bf 4022275b  
3fffee90:  3ffed664 4022232c 5c056537 60000600  
3fffeea0:  40221ea8 40239ce8 3ffed038 4021b638  
3fffeeb0:  00000001 00000007 3ffed05c 40211a21  
3fffeec0:  40222367 3ffed758 3ffed758 60000600  
3fffeed0:  402242d4 3ffed758 4bc6a7f0 00000000  
3fffeee0:  3ffed038 40224298 3ffed67c 3ffedea0  
3fffeef0:  4022231c 402222f0 3ffed038 3ffedea0  
3fffef00:  402222bb 3ffed67c 3ffedea0 4021344c  
3fffef10:  40222829 1cc18ee6 3ffedea0 60000600  
3fffef20:  40233941 3ffed67c 3ffedea0 5c054014  
3fffef30:  40233986 3fffdab0 00000000 3fffdcb0  
3fffef40:  3ffedec0 3fffdad0 3ffee7b8 40208d73  
3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 40001878 00000002 00000000  
3fffef70:  3fffff10 aa55aa55 00000021 401049f1  
3fffef80:  401049f7 00000002 00000000 00000000  
3fffef90:  4010000d 00000000 42200000 43800000  
3fffefa0:  00000000 3fffef3c 00000000 3fffff68  
3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
[... 261 lines cut ...]
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40211574: new_linkoutput at glue-lwip/lwip-git.c line 220
0x402119cc: ethernet_output at netif/ethernet.c line 319
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40208da2: can_yield() at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 94
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40206070: WiFiServer::available(unsigned char*) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi/src/WiFiServer.cpp line 121
0x4020ab10: WiFiClient::connected() at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi/src/WiFiClient.cpp line 334
0x4020ab5c: WiFiClient::operator bool() at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi/src/WiFiClient.cpp line 349
0x4020ab10: WiFiClient::connected() at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi/src/WiFiClient.cpp line 334
0x40207d7c: String::copy(char const*, unsigned int) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/WString.cpp line 214
0x40208e1b: optimistic_yield(uint32_t) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 129
0x4020606a: WiFiServer::available(unsigned char*) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi/src/WiFiServer.cpp line 120
0x4020ab5c: WiFiClient::operator bool() at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi/src/WiFiClient.cpp line 349
0x40208dc7: __esp_yield() at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 107
0x40100220: ets_post(uint8, ETSSignal, ETSParam) at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40100241: esp_schedule() at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 113
0x40208ee5: loop_wrapper() at /home/ahc/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 182
```

TD-er, 2020-03-03:
What code to reproduce this?

AHakan, 2020-03-04:
My code very complicate. I have headers file. I can't share.

AHakan, 2020-03-04:
> What code to reproduce this?

My code very complicate. I have headers file. I can't share.

d-a-v, 2020-03-04:
@AHakan you must understand that anybody can write a one-line sketch producing this. 
We need to be able to reproduce what leaded to the exception in order to fix it: we need an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example).
You can also enable all debug options and check what you serial terminal displays just before the failure.

TD-er, 2020-03-04:
Just to give some hint on what may cause this exception 29:
Try to look at where in your code you use flash strings (F() or progmem) and see if it is somewhere used as a regular char pointer or String object.

These are most often the cause of this exception 29.

AHakan, 2020-03-04:
> Just to give some hint on what may cause this exception 29:
> Try to look at where in your code you use flash strings (F() or progmem) and see if it is somewhere used as a regular char pointer or String object.
> 
> These are most often the cause of this exception 29.

`
/*-------------------------Includes-------------------------*/
#include <Arduino.h>
#include <EEPROM.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include "hony.h"

/*-------------------------Defines-------------------------*/
#ifndef APSSID
#define APSSID "HONYcube"
#define APPSK  "cube1234"
#endif

const char *ssid = APSSID;
const char *password = APPSK;

/*-------------------------Objects-------------------------*/
ESP8266WebServer server(80);

/*-----------------------handleWrite-----------------------*/
void readWIFI(){
  String qsid = server.arg("ssid");
  String qpass = server.arg("pass");

  if (qsid.length() > 0 && qpass.length() > 0) {
    for (uint8_t i = 0; i < 64; ++i) { 
      EEPROM.write(i, 0); 
    }
    for (uint8_t i = 0; i < qsid.length(); ++i){
      EEPROM.write(i, qsid[i]);
    }
    for (uint8_t i = 0; i < qpass.length(); ++i){
      EEPROM.write(32+i, qpass[i]);
    } 
    if (EEPROM.commit()) {
      server.sendHeader("Access-Control-Allow-Origin", "*");
      server.send(200, "application/json", "{\"Success\":\"saved\"}");
      ESP.reset();
    } else {
      Serial.println("ERROR! EEPROM commit failed");
    }
  }
}

/*-------------------------Setup-------------------------*/
void setup() {
  Serial.begin(115200);
  EEPROM.begin(512);
  WiFi.softAP(ssid, password);
  IPAddress myIP = WiFi.softAPIP();
  server.on("/setting", readWIFI);
  server.begin();
    
  char esid[32]="";
  for (uint8_t i = 0; i < 32; ++i){
    esid[i]=char(EEPROM.read(i));
  }
  char epass[32]="";
  for (uint8_t i = 32; i < 64; ++i){
    epass[i-32]=char(EEPROM.read(i));
  }
  
  if(strlen(esid) < 49){ //= 49 WHY?
    WiFi.mode(WIFI_STA);
    WiFi.begin(esid,epass);
    for(uint8_t i=0; i<20; i++){ 
      if (WiFi.status() != WL_CONNECTED) {
        delay(500);
      }
      else if(WiFi.status() == WL_CONNECTED){
        Serial.println(".");
        WiFi.softAPdisconnect (true);
        server.close();
      }
    }
  }
  Hony.neopixelSETUP();
}

/*-------------------------Loop-------------------------*/
void loop() {
  server.handleClient();
  delay(1000);
  while(WiFi.status() == WL_CONNECTED){
    Hony.getDATA();
    Hony.neopixelCONTROL();
    delay(1250);  //Get data and delay 1500 ms.
  }
}`

This is part of the code. Other part (hony header) http get, arduino json and some library. I think the error is in this piece of code.

AHakan, 2020-03-04:
> @AHakan you must understand that anybody can write a one-line sketch producing this.
> We need to be able to reproduce what leaded to the exception in order to fix it: we need an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example).
> You can also enable all debug options and check what your serial terminal displays just before the failure.

I'll try debug options, I shared some of the code. I hope the code is understandable.

TD-er, 2020-03-04:
```c++
  char esid[32]="";
  for (uint8_t i = 0; i < 32; ++i){
    esid[i]=char(EEPROM.read(i));
  }
  char epass[32]="";
  for (uint8_t i = 32; i < 64; ++i){
    epass[i-32]=char(EEPROM.read(i));
  }
```
Better to change the declarations like this:
```c++
  char esid[33]={0};
  char epass[33]={0};
```

This guarantees there is a 0 after your strings.  (array length + 1 and set all to 0)
Otherwise functions to determine the  string length may access memory they're not supposed to access.

AHakan, 2020-03-04:
> ```c++
>   char esid[32]="";
>   for (uint8_t i = 0; i < 32; ++i){
>     esid[i]=char(EEPROM.read(i));
>   }
>   char epass[32]="";
>   for (uint8_t i = 32; i < 64; ++i){
>     epass[i-32]=char(EEPROM.read(i));
>   }
> ```
> 
> Better to change the declarations like this:
> 
> ```c++
>   char esid[33]={0};
>   char epass[33]={0};
> ```
> 
> This guarantees there is a 0 after your strings. (array length + 1 and set all to 0)
> Otherwise functions to determine the string length may access memory they're not supposed to access.
> 
> Also do yourself a favor and change the last for loop like this:
> 
> ```c++
>   for (uint8_t i = 0; i < 32; ++i){
>     epass[i]=char(EEPROM.read(i + 32));
>   }
> ```

I changed and tried. I got the same error.