esp8266/Arduino #7140: SoftwareSerialConfig not being the same type as SerialConfig makes it difficult to use template models to support both hardware and software serial with one template class

Makuna, 2020-03-08:
### Problem Description
At some point, software serial switched from using SerialConfig to SoftwareSerialConfig.

I have a library that consists of a template class that accepts a "serial" template definition class.  This allows the user to select if they want hardware or software serial implementation by definine my class with the type of serial.  Like...
```
// for hardware serial
Rfid134<HardwareSerial, RfidNotify> rfid(Serial1);
// for software serial
SoftwareSerial secondarySerial(10, 11); // RX, TX
Rfid134<SoftwareSerial, RfidNotify> rfid(secondarySerial);
```

The issue is that due to the software serial class changing away from using the same type for the config (from SerialConfig to SoftwareSerialConfig) has caused that you can no longer use it as a template class as the `begin()` method no longer has the same arguments and the compilation will fail.
```
error: no matching function for call to 'SoftwareSerial::begin(int, SerialConfig)'
```

The project and example can be found here
https://github.com/Makuna/Rfid134/

earlephilhower, 2020-03-08:
Calling @dok-net .  @Makuna, you will probably need to open something up on the SWSerial repo for this:  https://github.com/plerup/espsoftwareserial .  I'm not sure what kind of CI he's using, bout adding your use case may be worthwhile...

dok-net, 2020-03-08:
No chance. EspSoftwareSerial supports 9bit modes and parity modes that are not in the HW serial interfaces.
On a funny note, "Difficult" is Template Programming's middle name, please kindly solve this issue in your code base.
@earlephilhower Feel free to close this issue, I'm not going to accept it at EspSoftwareSerial, either.
@makuna No offense, but I honestly think this is much much MUCH better solved in your template wrapper.

Makuna, 2020-03-08:
@dok-net No offense taken.  I created this so we had an open dialog as to IF it could be solved in the SoftwareSerial code in manageable way as it was a change from the original implementation.

An example, instead of taking SoftwareSerialConfig, take an INT (I believe the original software serial; even for AVR has this).  Then the type doesn't get in the way.

dok-net, 2020-03-08:
@Makuna I'm really glad that the API has finally settled and is not generating issues, really, I will not reopen that can of worms. Sorry. Besides, in doing so, EspSoftwareSerial would introduce, untyped hence unsafe, values that someone might then pass into the HW serial APIs. Not good.
Also the specific arithmetic is different. You are asking for a real mess.
Now, let me make a hilarious suggestion to prove the point:
EspSoftwareSerial works verbatim on both ESP8266 and ESP32.
Please ask the HW serial API maintainers on both platforms to adapt their SerialConfig APIs to EspSoftwareSerial :-)

E.g.
EspSoftwareSerial: SoftwareSerialConfig::SWSERIAL_8N1 == 3.
ESP8266: SerialConfig::SERIAL_8N1 == 0x1c
ESP32: #define SERIAL_8N1 0x800001c

dok-net, 2020-03-08:
@Makuna btw, begin on EspSoftwareSerial has a lot of overloads and default arguments these days, I am almost certain you can work something out. Like common default use for begin is `Serial.begin(9600)` with EspSoftwareSerial too (again) now.

Makuna, 2020-03-09:
The best solution for me to work around this is a Modern API pattern that currently is uncommon in the Arduino world.

That is, if a primary class has API dependent types, those types are defined within the class.  The most common way to have a typedef inside the class such that the "specific" dependent type (defined outside the class) is defined to the "common" type; where the common (its just a name) is used by ALL classes of a similar implementation; like Serial being HardwareSerial and SoftwareSerial.
So inside SoftwareSerial, it should have a typedef like
```
typedef SoftwareSerialConfig SerialConfig;
```
and inside HardwareSerial
```
typedef HardwareSerialConfig SerialConfig;
```
and then this common (SerialConfig) is used in all public API methods.

This will allow me to work around the problem.  

And yes, templates are complex to work with, but a lot of people thought the same with classes.  They can and do solve a lot of problems; and often in a way that makes it easy to use them; but rarely easy to implement them.

devyte, 2020-03-10:
In an ideal world, the Arduino reference would have compatible apus for sw and hw serial. The last time I looked that wasn't the case. We could add types in the class, I don't mind, but I don't think that would help you.
In cases such as this, the usual solution is to implement either traits or adapter classes. I think traits would be more suitable for what you want. In this case, a trait is really an api wrapper around specific obkects that are similar but different. You would need to implement one trait each.

Makuna, 2020-03-14:
As identified above, this is a bigger design issue/inconsistency with the Arduino API and not something that can be addressed here or even per platform.

I punted the problem to the user and forced the serial begin to be called outside my class.  Several other platforms have changed their own software serial to even more drastic things, like dropping the argument for 8N2 completely as they originally didn't support it anyhow.

closed 2020-03-14 by Makuna