esp8266/Arduino #7141: former `configTime` method no longer works.

whyameye, 2020-03-08:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X ] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12F
- Core Version:			commit 252240326fc1c4161265731b8e091244ce4094f4
- Development Env:		1.8.9
- Operating System:		Ubuntu 18.04
### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

`configTime` appears to now have a new format for calls --- a welcome change. Unfortunately the previous format no longer works despite the documentation saying that it does.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

using example included in core `NTP-TZ-DST.ino` , line 223 reads `former configTime is still valid...` which I interpret to mean:
- line 218 can be commented out: `//configTime(MYTZ, "pool.ntp.org");`
- and line 225 can be uncommented: `//configTime(7 * 3600, 3600 / 2, "pool.ntp.org");`

and the example should still work. It crashes instead. Decoded stack dump below.

_I discovered this when my own code for getting time via NTP stopped working after I updated to latest commit on master branch._

### Debug Messages

```
Decoding stack results
0x40206e15: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 226
0x4020ad56: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4020b187: _svfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 667
0x40203d89: uart_tx_fifo_available(int) at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/uart.cpp line 463
0x40203db4: uart_do_write_char(int, char) at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/uart.cpp line 476
0x4020866d: _vsnprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 73
0x4020866d: _vsnprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 73
0x40203e79: uart_write(uart_t*, char const*, size_t) at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/uart.cpp line 509
0x40201bfc: ESP8266WiFiSTAClass::status() at /home/user/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 634
0x402086b0: vsnprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 42
0x40201ebc: HardwareSerial::write(unsigned char const*, unsigned int) at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h line 165
0x40201eb0: HardwareSerial::write(unsigned char const*, unsigned int) at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h line 164
0x40201eb0: HardwareSerial::write(unsigned char const*, unsigned int) at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h line 164
0x40202248: Print::printf(char const*, ...) at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 63
0x402021e9: Print::write(char const*) at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.h line 62
0x4020240d: Print::printNumber(unsigned long, unsigned char) at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 268
0x40203e79: uart_write(uart_t*, char const*, size_t) at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/uart.cpp line 509
0x40205d39: _findenv_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/getenv_r.c line 99
0x40201eb0: HardwareSerial::write(unsigned char const*, unsigned int) at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h line 164
0x4020126a: showTime() at /tmp/arduino_modified_sketch_870741/NTP-TZ-DST.ino line 140
0x402014b2: setup() at /tmp/arduino_modified_sketch_870741/NTP-TZ-DST.ino line 239
0x40202398: Print::println(char const*) at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 198
0x40202da8: loop_wrapper() at /home/user/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 182
```

d-a-v, 2020-03-09:
Thanks for reporting, this bug is in the example, not in the core.
It is fixed by #7144.

whyameye, 2020-03-09:
Perhaps, but there has been a regression because as I noted: "I discovered this when my own code for getting time via NTP stopped working after I updated to latest commit on master branch." So code that used to work with `configTime` no longer does. I suspect if we look at a previous version of the example we will discover it no longer works, but I have not confirmed. I can look into that when I have a chance.

d-a-v, 2020-03-09:
An MCVE would be great

d-a-v, 2020-03-16:
@whyameye Any news ?

dontsovcmc, 2020-05-16:
I confirm configTime error!

Good version **2.4.0**:
```
PLATFORM: Espressif 8266 2.4.0 #4a9e74d > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20603.200130 (2.6.3) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
```

Broken version **2.5.0**:
```
PLATFORM: Espressif 8266 2.5.0 #772ec50 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20700.200502 (2.7.0) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
```

```
#include <Arduino.h>
#include "time.h"
#include <ESP8266WiFi.h>

const char *ssid     = "YOUR_SSID";
const char *password = "YOUR_PASS";

void setup() 
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.begin(ssid, password);

  while ( WiFi.status() != WL_CONNECTED ) {
      delay ( 500 );
      Serial.print ( "." );
  }

  configTime(0, 0, "pool.ntp.org");

  Serial.println("Waiting for NTP time sync");
  unsigned long start = millis();

  time_t now = time(nullptr);

  while (now < 50000 && millis() - start < 3000UL) 
  {
    delay(100);
    now = time(nullptr);
  }

  if (millis() - start >= 3000UL)
  {
    Serial.print("NTP error");
  } 
  else
  {
    time_t now = time(nullptr);
    struct tm timeinfo;
    gmtime_r(&now, &timeinfo);
    Serial.print("Current time: ");
    Serial.print(asctime(&timeinfo));
  } 
}

void loop() 
{

}

```

d-a-v, 2020-05-16:
With your MCVE I get with current master (2.7.1+):
```
Leaving...
Hard resetting via RTS pin...
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
.......Waiting for NTP time sync
Current time: Sat May 16 13:37:32 2020
```
So it's working.

However, these loops look suspicious:
```
  while (now < 50000 && millis() - start < 3000UL) 
  {
    delay(100);
    now = time(nullptr);
  }

  if (millis() - start >= 3000UL)
  {
    Serial.print("NTP error");
  } 
```

If you need to wait until time is set, you can use a callback like [in the provided example](https://github.com/esp8266/Arduino/blob/c5f60e31cd1afa839a823c9f376ce3a0fdc2144c/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino#L211) in something like:

```
bool time_is_set = false;
...
void time_is_set ()
{
    time_is_set = true;
    // or do something
}
...
void setup ()
{
...
    // install callback - called when settimeofday is called (by SNTP or us)
    // once enabled (by DHCP), SNTP is updated every hour
    settimeofday_cb(time_is_set_scheduled);
...
    // one can wait here
    while (!time_is_set)
    {
        delay(100);
    }
}

void loop ()
{
   // one can do something here
    if (time_is_set)
    {
        // do something
    }
}
```

dontsovcmc, 2020-05-16:
Ok, I'll use callback. 
2.7.1 also doesn't work.
You think, till 2.7.0 it's spend more than 3 seconds to get time? I think no..

d-a-v, 2020-05-16:
Please try and check with the callback and report.
If it works, I'm interested in duration between start and when time is reported to be set.

d-a-v, 2020-11-09:
@whyameye Do you still have issues with the master branch ?

d-a-v, 2021-03-31:
Closing per fixed in current master.
Please open a new issue if there's still a problem with `configTime`.