esp8266/Arduino #7145: httpUpdateSigned crashes on verification when the digital signature does not match

JiriBilek, 2020-03-09:
#### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [commit from Feb-28] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [n/a] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12F]
- Core Version:			[Feb-28-2020]
- Development Env:		[Arduino IDE or Sloeber]
- Operating System:		[Windows 10]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB, FS:1MB, OTA:1019KB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[no dtr (aka ck)]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA]
- Upload Speed:   [n/a] (serial upload only)

### Problem Description

The OTA update with default signature verification (MANUAL_SIGNING = 0) work fine up to the moment when I tamper the signature in the bin file. 
Any byte changed in the 256 byte signature will cause wdt or various crashes. Changing the last 4 bytes (the signature length) does not crash the CPU, it correctly throws an error and resumes the sketch.
I traced the problem up to the file Updater.cpp. Changing the line https://github.com/esp8266/Arduino/blob/845241b72c03e18446a0d762e2c339f2a9f55842/cores/esp8266/Updater.cpp#L264 to e.g. `if (sig[0xf0] == 0) {` will allow playing with the signature and not crashing the CPU. Of course, it removes the digital signature verification :), it is only for tracing the bug.

I am asking for help as I don't understand the RSA stuff in https://github.com/esp8266/Arduino/blob/845241b72c03e18446a0d762e2c339f2a9f55842/libraries/ESP8266WiFi/src/BearSSLHelpers.cpp#L873 (the function used for verifying the signature).

Thanks for your time.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Standard example Arduino/libraries/ESP8266httpUpdate/examples/httpUpdateSecure/httpUpdateSecure.ino, no code changes, only RSA keys changed, generated by OpenSSL. I can upload them if it's worth doing it.

### Debug Messages

wdt or Fatal exception 2(InstructionRetchErrorCause) at various addresses (epc1)
I will decode the stack and update the post shortly.

JiriBilek, 2020-03-09:
There is also a memory leak in https://github.com/esp8266/Arduino/blob/845241b72c03e18446a0d762e2c339f2a9f55842/cores/esp8266/Updater.cpp#L264

The sig variable and the buffer needs to be freed. But it doesn't fix the problem.. 

```
    if (!_verify->verify(_hash, (void *)sig, sigLen)) {
      _setError(UPDATE_ERROR_SIGN);
      free(sig);
      _reset();
      return false;
    }
    free(sig);
```

JiriBilek, 2020-03-10:
I made a simple sketch to verify the signature on the running sketch (no uploading from the internet etc). The code is taken from https://github.com/esp8266/Arduino/blob/845241b72c03e18446a0d762e2c339f2a9f55842/cores/esp8266/Updater.cpp#L233
And it is running fine. So the problem won't be simply in the RSA signature, unfortunately.

[verifySig.zip](https://github.com/esp8266/Arduino/files/4310962/verifySig.zip)

If you want to play with the code, note that Arduino IDE uploads unsigned sketch. You need to upload the signed one using python script manually.

earlephilhower, 2020-03-10:
Good catch on the memory leak.

Because the PC has gone bad it feels like a stack corruption issue (as-in stack overflow).

We're now using the standard format signature and not just the SHA256 result, so a change of 1 byte in the signature may end up causing the packed format to go wacky.  BearSSL should still not crash things, in that case, though.

Can you give the exact pub/priv key and the exact offset from the end of the .bin.signed you changed, so I can rerun the same changes on my own builds to debug?

JiriBilek, 2020-03-10:
The keys are in verifySig.zip in the post above.

JiriBilek, 2020-03-11:
@earlephilhower:
I added `ESP.getFreeContStack();` reporting in the loop of the failing sketch (httpUpdateSecure.ino) and after verifying a bad signature I got this:

```
[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...
free cont stack: 3136
HTTP_UPDATE_FAILED Error (12): Update error: ERROR[12]: Signature verification failed
free cont stack: 0
```

Doesn't it mean stack overflow?

earlephilhower, 2020-03-11:
That looks like a smoking gun right there, @JiriBilek.  I may need to allocate/use the 2nd stack for the verification, or check that the Updater stack usage is minimized.

PR #7149 (2020-03-14): Use 2nd stack for update signature verification
Fixes #7145

When doing a signed update, the signature calculation can use a lot of
stack, so move it silently to the BearSSL second stack.

Also fix a memory leak of signature-bytes found by @JiriBilek
commit e2528732 (2020-03-14): Use 2nd stack for update signature verification (#7149)

* Use 2nd stack for update signature verification

Fixes #7145

When doing a signed update, the signature calculation can use a lot of
stack, so move it silently to the BearSSL second stack.

Also fix a memory leak of signature-bytes found by @JiriBilek

* Reset state on any error condition in Updater::end
closed 2020-03-14 by earlephilhower